evcc-io/evcc #10225: Zielladen nicht anzeigen wenn Kapazität oder Soc nicht bekannt

andig, 2023-10-07:
Das geht ja spätestens bei Offlinefahrzeugen nicht mehr. Besser wäre wohl, dass wir nur Laden nach Energiemenge anbieten wenn Kapazität oder Soc nciht bekannt sind.

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/10214#discussioncomment-7216513_

VolkerK62, 2023-10-07:
Also wird Zielladen nur angeboten wenn beides (Kapazität und Soc) bekannt ist.
Braucht es Soc unbedingt?
Bei einem `offline` Fahrzeug ist nur Kapazität vorhanden (wenn definiert) und es reicht doch, wenn für Zielladen eine Energiemenge ausgewählt wird.
Oder habe ich hier einen Gedankenfehler?

StefanSchoof, 2023-10-07:
Wenn Funktionen von der Kapazität abhängen, dann muss in der Dokumentation das anders als heute kommuniziert werden. Wenn dort optional steht, erwarten die Nutzer nicht, dass sie weniger Funktionen haben wenn sie den Wert nicht angeben. Oder es sollte in der UI deutlich werden, das man nicht Zielladen kann weil die Kapazität fehlt. Sonst kann es passieren, dass Nutzer die Funktion gar nicht entdecken und somit Teile von evcc nicht nutzen.

andig, 2023-10-07:
Es gibt zwei Sorten Zielladung: Kapa/Soc und Energiemenge. Für die erste Variante muss Beides vorhanden sein.

andig, 2023-10-10:
Also Fehlermeldung aus dem Backend?

naltatis, 2023-10-10:
Vtml macht beides Sinn. In der UI würd ich den Fall schon abfangen bevor der Nutzer probiert das abzuschicken. Wenn man als Gast oder Offlinefahrzeug eine Zielzeit setzt ohne ein Energielimit zu haben kommt heute schon ein ähnlicher Hinweis. Ich würde diesen hier analog dazu anzeigen.

[image]

Dem `POST /target/time/<>` Endpunkt sollten wir aber in der Tat auch eine Fehlerbehandlung verpassen damit man da nicht reinläuft, wenn man rein über die API kommt.

PR #10320 (2023-10-20): Planner: add message for soc vehicles without capacity
fixes #10225

[image]

TODO

- [ ] Validate api call `POST /target/time/<>` @andig
closed 2023-10-20 by andig