evcc-io/evcc #10233: PV Überschuss Laden nur noch mit 1,1kW

Tihmann, 2023-10-08:
### Describe the bug

Seit Update auf 0.120.3 funktioniert das Hochregeln der Ladeleistung nicht mehr. Die Ladeleistung bleibt bei 1P bei 1,1kW, obwohl Überschuss vorhanden ist. Hat sich etwas in der Konfiguration geändert. Muss man etwas neu einrichten?

### Steps to reproduce

Überschuss vorhanden, Laden

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xx

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

chargers:
  - type: template
    template: go-e-v3
    host: 192.168.178.xx
    name: wallbox2
  - type: template
    template: go-e-v3
    host: 192.168.178.xx
    name: wallbox3
vehicles:
  - type: template
    template: tesla
    title: Tesla Model 3
    accessToken: xx
    refreshToken: xx
[... 137 lines cut ...]
Okt 08 09:27:20 evcc evcc[195]: [lp-1  ] DEBUG 2023/10/08 09:27:20 detected active phases: 1p
Okt 08 09:27:20 evcc evcc[195]: [lp-1  ] DEBUG 2023/10/08 09:27:20 charge total import: 468.047kWh
Okt 08 09:27:20 evcc evcc[195]: [lp-1  ] DEBUG 2023/10/08 09:27:20 charger status: C
Okt 08 09:27:20 evcc evcc[195]: [lp-1  ] DEBUG 2023/10/08 09:27:20 soc estimated: 31.31% (vehicle: 31.00%)
Okt 08 09:27:20 evcc evcc[195]: [lp-1  ] DEBUG 2023/10/08 09:27:20 vehicle soc: 31%
Okt 08 09:27:20 evcc evcc[195]: [lp-1  ] DEBUG 2023/10/08 09:27:20 vehicle soc limit: 100%
Okt 08 09:27:20 evcc evcc[195]: [lp-1  ] DEBUG 2023/10/08 09:27:20 vehicle range: 162km
Okt 08 09:27:20 evcc evcc[195]: [lp-1  ] DEBUG 2023/10/08 09:27:20 pv charge current: 12.5A = 6.9A + 5.6A (-1287W @ 1p)
Okt 08 09:27:29 evcc evcc[195]: [site  ] DEBUG 2023/10/08 09:27:29 ----
Okt 08 09:27:29 evcc evcc[195]: [lp-1  ] DEBUG 2023/10/08 09:27:29 charge power: 1120W
Okt 08 09:27:29 evcc evcc[195]: [lp-2  ] DEBUG 2023/10/08 09:27:29 charge power: 0W
Okt 08 09:27:29 evcc evcc[195]: [site  ] DEBUG 2023/10/08 09:27:29 pv power: 2542W
Okt 08 09:27:30 evcc evcc[195]: [site  ] DEBUG 2023/10/08 09:27:30 grid power: -1287W
Okt 08 09:27:30 evcc evcc[195]: [site  ] DEBUG 2023/10/08 09:27:30 site power: -1287W
Okt 08 09:27:30 evcc evcc[195]: [lp-2  ] DEBUG 2023/10/08 09:27:30 charge currents: [0 0 0]A
Okt 08 09:27:30 evcc evcc[195]: [lp-2  ] DEBUG 2023/10/08 09:27:30 charge total import: 3482.978kWh
```

### What type of operating system are you running?

Linux

### Version

0.120.3

andig, 2023-10-08:
Im Log ist nur zu erkennen, dass die Box- ohne Einfluss von evcc- immer an und aus geht (5A, 0A, 5A, 0A). Bitte gleiches Log nochmal für 0.120.2.

VolkerK62, 2023-10-08:
evtl 5A als Limit im Auto gesetzt?

Tihmann, 2023-10-08:
Peinlich, das war die Lösung. Interessanterweise aber bei beiden Teslas nach dem letzten Tesla Update so voreingestellt bei der Home Adresse.

Kann geschlossen werden

closed 2023-10-08 by Tihmann

andig, 2023-10-08:
Ich lach mich scheckig. Beweist mal wieder, dass der Reflex „seit dieser Version…“ doch oft fehl leitet. Lesson learned Fehlersuche: immer Fehlerquellen einzeln ausschließen, in diesem Fall also Vergleich mit der letzten Version.