evcc-io/evcc #10241: Add dynamic pricing for dutch energy providers energyzero/anwb energy/mijn domein energie etc.

peebee01, 2023-10-09:
Can you extend the supported dynamic energy providers with https://www.energyzero.nl/?

they provide dynamic priced electricity to consumers as well as to various resellers.

there are already examples of homeassistant integration using their API

https://www.home-assistant.io/integrations/energyzero/
https://github.com/klaasnicolaas/python-energyzero

With the winter coming up, optimized charging based on dynamic rates will be more valuable to me than PV charging

andig, 2023-10-09:
Are the rates a 1:1 mapping of EntsoE? If yes you could already use that.

chrisnewblack, 2023-10-09:
Likely a mapping of EntsoE but with the ease of use of cost modifiers applied for the suppliers that use energyzero?
@peebee01 see https://github.com/evcc-io/evcc/issues/5291

peebee01, 2023-10-12:
Thanks I have tried to setup entsoe but couldnt find the settings parameters to use for evcc.yaml. Is there an example config somewhere? Entsoe is not in example yaml file

andig, 2023-10-12:
Added example in https://github.com/evcc-io/evcc/commit/9a27739a1dbf80ef4cd8c4e1f630e15b042eab04. Lets reopen if not sufficient.

closed 2023-10-12 by andig

peebee01, 2023-10-21:
Thanks, 
In the mean time I requested and received the token from entsoe (replaced by xxxxxx in the code below).
For others using these dutch energy providers you have to use the mentioned tax amounts to get the nett price in evcc
You can use the same for feedin to keep track of the money returned for the unused electricity (salderen)
 
```

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
     type: entsoe # Entso-E european market data
     domain: BZN|NL # https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html#_areas
     securitytoken: xxxxxx # api token
     charges: 0.1435 # energyzero additional charges per kWh
     tax: 0.21 # additional tax (21%)

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: entsoe # Entso-E european market data
    domain: BZN|NL # https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html#_areas
    securitytoken: xxxxxx # api token
     charges: 0.1435 # energyzero additional charges per kWh
     tax: 0.21 # additional tax (21%)

```