evcc-io/evcc #10248: phoenix-em-eth - Keine Zählwerte

Sami0178, 2023-10-09:
### Describe the bug

Seit evcc version 0.120.0 funktioniert kein Zählwert Start/Stop mehr.
Es kann an dieser Ladecontroller Version leider nur Registerweise abgefragt werden. (Messwerte werden über Modbus RTU vom Ladecontroller über Modbus TCP zur Verfügung gestellt)

### Steps to reproduce

1. install new version
2.
3.
...

### Configuration details

```yaml
- name: wallbox7
  type: phoenix-em-eth
  uri: 192.168.2.13:502
```

### Log details

```text
[lp-1  ] INFO 2023/10/09 11:55:22 loadpoint 1:
[lp-1  ] INFO 2023/10/09 11:55:22   mode:        minpv
[lp-1  ] INFO 2023/10/09 11:55:22   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/09 11:55:22   meters:      charge ✗
[lp-2  ] INFO 2023/10/09 11:55:22 loadpoint 2:
[lp-2  ] INFO 2023/10/09 11:55:22   mode:        pv
[lp-2  ] INFO 2023/10/09 11:55:22   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2023/10/09 11:55:22   meters:      charge ✗
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.0

andig, 2023-10-09:
> Seit evcc version 0.120.0 funktioniert kein Zählwert Start/Stop mehr.

Was bedeutet das?

> Es kann an dieser Ladecontroller Version leider nur Registerweise abgefragt werden. (Messwerte werden über Modbus RTU vom Ladecontroller über Modbus TCP zur Verfügung gestellt)

Was bedeutet das?

May be easier in English?

Sami0178, 2023-10-10:
Soll bedeuten, dass Anfangszählerstand und Endzählerstand leider nicht mehr in der Auswertung erscheinen. (siehe Tabelle)
Musste die Config anpassen, dass der Ladecontroller ab V0.120.0 noch funktioniert. 
"- name: wallbox7
  type: phoenix-em-eth
  uri: 192.168.2.13:502
  "
Eventuell hat sich die Modbus TCP Abfrage geändert? Es ist bei den alten Phoenix Ladecotrollern nur eine MB Registerweise Abfrage möglich. 

Ist meine Config unvollständig?
Danke für Euren Support!!  : )
 
<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/pbjm11/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/pbjm11/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/pbjm11/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/pbjm11/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Startzeit | Endzeit | Ladepunkt | Kennung | Fahrzeug | Kilometerstand (km) | Anfangszählerstand (kWh) | Endzählerstand (kWh) | Energie (kWh) | Charge Duration | Solar (%) | Price | Price/kWh | CO2/kWh (gCO2eq)
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
07.10.2023 13:10 | 07.10.2023 14:42 | Ladepunkt 2 |   | Fiat500CE | 2681 |   |   | 21,045 | 1h31m25s | 4,316 | 3,474 | 0,165 | 93,77

</body>

</html>

andig, 2023-10-10:
Zeig uns mal bitte

    evcc charger —log trace

Ich vermute such, dass du den angeschlossenen Zähler eigentlich konfigurieren müsstest?

premultiply, 2023-10-10:
Sprich: Das Bulk-Reading aller Phasenwerte funktioniert hier offensichtlich nicht.
Ganz ohne Meter funktioniert es dann, klar - aber halt dann auch ganz ohne Zählerwerte.

premultiply, 2023-10-10:
> hatte sich da was geändert?

Nein, da nicht, aber du hattest das Bulk-Reading der Phasenströme eingebaut (vereinheitlicht).
Sprich mit `currents: false` müsste eigentlich(tm) dann der Rest zumindest wieder funktionieren.

andig, 2023-10-10:
Es funktioniert doch alles- sonst würde es einen Fehler geben. Mir scheint hier eher die Config zu fehlen!

Sami0178, 2023-10-10:
evcc charger --log trace
[main  ] INFO 2023/10/10 12:02:08 evcc 0.120.3
[main  ] INFO 2023/10/10 12:02:08 using config file: /etc/evcc.yaml
[db    ] INFO 2023/10/10 12:02:09 using sqlite database: /home/pb/.evcc/evcc.db
[db    ] TRACE 2023/10/10 12:02:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT * FROM `settings` 11 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2023/10/10 12:02:09 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
wallbox6
--------
[em-eth] TRACE 2023/10/10 12:02:09 modbus: send 00 01 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2023/10/10 12:02:09 modbus: recv 00 01 00 00 00 05 b4 04 02 00 41
[em-eth] TRACE 2023/10/10 12:02:09 modbus: send 00 02 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2023/10/10 12:02:09 modbus: recv 00 02 00 00 00 04 b4 01 01 00
[em-eth] TRACE 2023/10/10 12:02:09 modbus: send 00 03 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2023/10/10 12:02:09 modbus: recv 00 03 00 00 00 07 b4 04 04 3c 51 00 00
Charge status: A
Enabled:       false
Duration:      4h17m21s

wallbox7
--------
[em-eth] TRACE 2023/10/10 12:02:09 modbus: send 00 01 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2023/10/10 12:02:09 modbus: recv 00 01 00 00 00 05 b4 04 02 00 41
[em-eth] TRACE 2023/10/10 12:02:09 modbus: send 00 02 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2023/10/10 12:02:09 modbus: recv 00 02 00 00 00 04 b4 01 01 00
[em-eth] TRACE 2023/10/10 12:02:09 modbus: send 00 03 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2023/10/10 12:02:09 modbus: recv 00 03 00 00 00 07 b4 04 04 00 6f 00 00
Charge status: A
Enabled:       false
Duration:      1m51s

WP_SG
-----
[fritzdect] TRACE 2023/10/10 12:02:09 GET https://fritz.box/login_sid.lua
[fritzdect] TRACE 2023/10/10 12:02:09 <?xml version="1.0" encoding="utf-8"?><SessionInfo><SID>0000000000000000</SID><Challenge>c01dd8f3</Challenge><BlockTime>0</BlockTime><Rights></Rights><Users><User last="1">fritz7405</User></Users></SessionInfo>
[fritzdect] TRACE 2023/10/10 12:02:09 GET https://fritz.box/login_sid.lua?response=c01dd8f3-235fd30611d95bba0a7dba3b86d666c4&username=fritz1111
[fritzdect] TRACE 2023/10/10 12:02:12 <?xml version="1.0" encoding="utf-8"?><SessionInfo><SID>3682504c80981620</SID><Challenge>b836adc4</Challenge><BlockTime>0</BlockTime><Rights><Name>Dial</Name><Access>2</Access><Name>App</Name><Access>2</Access><Name>HomeAuto</Name><Access>2</Access><Name>BoxAdmin</Name><Access>2</Access><Name>Phone</Name><Access>2</Access><Name>NAS</Name><Access>2</Access></Rights><Users><User last="1">fritz1111</User></Users></SessionInfo>
[fritzdect] TRACE 2023/10/10 12:02:12 GET https://fritz.box/webservices/homeautoswitch.lua?ain=116560620573&sid=3682504c80981620&switchcmd=getswitchpower
[fritzdect] TRACE 2023/10/10 12:02:12 0
[fritzdect] TRACE 2023/10/10 12:02:12 GET https://fritz.box/webservices/homeautoswitch.lua?ain=116560620573&sid=3682504c80981620&switchcmd=getswitchenergy
[fritzdect] TRACE 2023/10/10 12:02:13 904747
[... cut ...]
[fritzdect] TRACE 2023/10/10 12:02:14 1
[fritzdect] TRACE 2023/10/10 12:02:14 GET https://fritz.box/webservices/homeautoswitch.lua?ain=116560620573&sid=3682504c80981620&switchcmd=getswitchpower
[fritzdect] TRACE 2023/10/10 12:02:15 0
[fritzdect] TRACE 2023/10/10 12:02:15 GET https://fritz.box/webservices/homeautoswitch.lua?ain=116560620573&sid=3682504c80981620&switchcmd=getswitchstate
[fritzdect] TRACE 2023/10/10 12:02:15 0
Power:         0W
Energy:        904.7kWh
Charge status: B
Enabled:       false
Features:      []

Sami0178, 2023-10-10:
Meine Config:
[main  ] INFO 2023/10/10 12:08:50 evcc 0.120.3
[main  ] INFO 2023/10/10 12:08:50 using config file: /etc/evcc.yaml
[db    ] INFO 2023/10/10 12:08:50 using sqlite database: /home/pb/.evcc/evcc.db
[site  ] WARN 2023/10/10 12:08:56 deprecated: use 'pv' instead of 'pvs'
[site  ] WARN 2023/10/10 12:08:56 deprecated: use 'battery' instead of 'batteries'
[site  ] WARN 2023/10/10 12:08:56 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://https://docs.evcc.io/docs/reference/configuration/site#residualpower

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: *****
#plant: *****
sponsortoken: *****
interval: 15s # control cycle interval

meters:
- type: template
  template: lg-ess-home-8-10 
  usage: grid  
  host: 192.168.2.31  
  registration: ********* 
  name: grid1

#- name: grid1
#  type: template
#  template: tibber-pulse
#  usage: grid
#  token: *****

- type: template
  template: lg-ess-home-8-10 
  usage: pv  
  host: 192.168.2.31  
  registration: DE1911BKRE0046AE  
  name: pv2

- type: template
  template: lg-ess-home-8-10 
  usage: battery  
  host: 192.168.2.31  
  registration: *********  
  name: battery3

- type: template
  name: WP_Pwr
  template: fritzdect
  usage: charge
  uri: https://fritz.box # HTTP(S) Adresse (Optional)
  user: *****
  password: *****
  ain: *****
  

chargers:
- name: WP_SG
  type: template
  template: fritzdect
  uri: https://fritz.box # HTTP(S) Adresse (Optional)
  user: *****
  password: *****
  ain: *****
  standbypower: 15 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (Optional) 

- name: wallbox6
  type: phoenix-em-eth
  uri: 192.168.2.12:502
#  meter:
#  power: true
#  energy: true
#  currents: true

- name: wallbox7
  type: phoenix-em-eth
  uri: 192.168.2.13:502
#  meter:
#  power: true
#  energy: true
#  currents: true

#- type: template
#  template: phoenix-em-eth 
#  host: 192.168.2.12  
#  port: 502  
#  name: wallbox6

#- type: template
#  template: phoenix-em-eth 
#  host: 192.168.2.13  
#  port: 502  
#  name: wallbox7

vehicles:
- name: WP
  template: offline
  type: template
  title: Wärmepumpe
  icon: waterheater
  phases: 1

- type: template
  template: id 
  title: ID4 Pro
  user: *****
  password: *****
  vin: *****
  capacity: 77  
  name: id

- type: template
  template: tesla 
  title: Model3 
  accessToken: *****
  refreshToken: *****
  vin: *****
  capacity: 60  
  name: m3

- type: template
  template: fiat
  title: Fiat500CE 
  user: *****
  password: *****
  vin: *****
  capacity: 42 # Akkukapazität in kWh (Optional)
  pin: *****
  name: fiat

loadpoints:
- title: Ladepunkt 1
  charger: wallbox6
  vehicle: m3
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 20
  resetOnDisconnect: false
  guardduration: 10m     # Schaltpause 10m

- title: Ladepunkt 2
  charger: wallbox7
  vehicle: id
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 20
  resetOnDisconnect: false
  guardduration: 5m     # Schaltpause 5m

- title: Poolheizung
  charger: WP_SG
  meter: WP_Pwr
  vehicle: WP
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 9
  enable:
    threshold: 0
    delay: 5m
  guardduration: 15m     # Schaltpause 15m

site:
  title: PV Manager
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3

tariffs:
  #currency: EUR
  grid:
    type: tibber
    #smartCostLimit: 0.15 # [currency]/kWh
    token: *****
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.18 # [currency]/kWh
  co2:
    type: grünstromindex
    zip: "*****"

Version: `0.120.3`

andig, 2023-10-10:
> evcc charger --log trace

Könntest Du das bitte formatieren?

premultiply, 2023-10-10:
Gemeint ist aber die ursprüngliche Konfig-Variante mit `meters:`

andig, 2023-10-10:
Ich verstehe nur noch Bahnhof. Die Config steht oben und enthält das nicht.

Sami0178, 2023-10-10:
sorry, hier nochmal meine config als txt
[config.txt](https://github.com/evcc-io/evcc/files/12857478/config.txt)

premultiply, 2023-10-10:
> Musste die Config anpassen, dass der Ladecontroller ab V0.120.0 noch funktioniert.

Probierst du bitte nochmal mit
```yaml
- name: wallbox7
  type: phoenix-em-eth
  uri: 192.168.2.13:502
  meter:
    power: true
    energy: true
    currents: true
```
und dann
```yaml
- name: wallbox7
  type: phoenix-em-eth
  uri: 192.168.2.13:502
  meter:
    power: true
    energy: true
    currents: false
```
jeweils `evcc charger --log trace` aus.

Sami0178, 2023-10-10:
Mit  "currents: true" funtioniert es nun. Bei "false" kommen keine Zählwerte.
War zu doof für die yaml Config. 
Hatte nach "meter:" nicht eingerückt...
Sorry für den Aufwand!
Dankeschön!

Sami0178, 2023-10-10:
currents: false
 --------
 wallbox7
--------
[em-eth] TRACE 2023/10/10 15:21:15 modbus: send 00 01 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2023/10/10 15:21:15 modbus: recv 00 01 00 00 00 07 b4 04 04 02 2d 00 00
[em-eth] TRACE 2023/10/10 15:21:15 modbus: send 00 02 00 00 00 06 b4 04 00 80 00 02
[em-eth] TRACE 2023/10/10 15:21:15 modbus: recv 00 02 00 00 00 07 b4 04 04 d7 02 00 17
[em-eth] TRACE 2023/10/10 15:21:15 modbus: send 00 03 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2023/10/10 15:21:15 modbus: recv 00 03 00 00 00 05 b4 04 02 00 43
[em-eth] TRACE 2023/10/10 15:21:15 modbus: send 00 04 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2023/10/10 15:21:15 modbus: recv 00 04 00 00 00 04 b4 01 01 01
[em-eth] TRACE 2023/10/10 15:21:15 modbus: send 00 05 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2023/10/10 15:21:15 modbus: recv 00 05 00 00 00 07 b4 04 04 00 c9 00 00
Power:         5570W
Energy:        15623.7kWh
Charge status: C
Enabled:       true
Duration:      3m21s

currents: true
--------
wallbox7
--------
[em-eth] TRACE 2023/10/10 15:22:59 modbus: send 00 01 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2023/10/10 15:22:59 modbus: recv 00 01 00 00 00 07 b4 04 04 02 2a 00 00
[em-eth] TRACE 2023/10/10 15:22:59 modbus: send 00 02 00 00 00 06 b4 04 00 80 00 02
[em-eth] TRACE 2023/10/10 15:22:59 modbus: recv 00 02 00 00 00 07 b4 04 04 d7 16 00 17
[em-eth] TRACE 2023/10/10 15:22:59 modbus: send 00 03 00 00 00 06 b4 04 00 72 00 02
[em-eth] TRACE 2023/10/10 15:22:59 modbus: recv 00 03 00 00 00 07 b4 04 04 1f 34 00 00
[em-eth] TRACE 2023/10/10 15:22:59 modbus: send 00 04 00 00 00 06 b4 04 00 74 00 02
[em-eth] TRACE 2023/10/10 15:22:59 modbus: recv 00 04 00 00 00 07 b4 04 04 1f 4c 00 00
[em-eth] TRACE 2023/10/10 15:22:59 modbus: send 00 05 00 00 00 06 b4 04 00 76 00 02
[em-eth] TRACE 2023/10/10 15:22:59 modbus: recv 00 05 00 00 00 07 b4 04 04 1e e2 00 00
[em-eth] TRACE 2023/10/10 15:22:59 modbus: send 00 06 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2023/10/10 15:22:59 modbus: recv 00 06 00 00 00 05 b4 04 02 00 43
[em-eth] TRACE 2023/10/10 15:22:59 modbus: send 00 07 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2023/10/10 15:22:59 modbus: recv 00 07 00 00 00 04 b4 01 01 01
[em-eth] TRACE 2023/10/10 15:22:59 modbus: send 00 08 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2023/10/10 15:22:59 modbus: recv 00 08 00 00 00 07 b4 04 04 01 31 00 00
Power:          5540W
Energy:         15623.9kWh
Current L1..L3: 7.99A 8.01A 7.91A
Charge status:  C
Enabled:        true
Duration:       5m5s

premultiply, 2023-10-10:
Huch, also funktioniert es doch alles?

Sami0178, 2023-10-10:
Ja, Fehler war 50cm vor dem Bildschirm.
yaml Formatierung ...
sorry & vielen Dank!

closed 2023-10-10 by Sami0178

andig, 2023-10-10:
Danke, sag ich doch…

andig, 2023-10-10:
Lesson learned: vollständige Issues mit kompletten Logs (trace!)