evcc-io/evcc #10254: Started charging in pv mode during night

timostr, 2023-10-10:
### Describe the bug

EVCC was configured in PV mode. House battery was empty (10%) and no sun at all (20:37:59 O'clock).
Target SoC was configured to 80% in EVCC and in the Car.
At 20:37:59 O'clock the car charging was started by evcc and could only be switched off, by lowering the target SoC in the car. I.e. the car has stopped the charging.

### Steps to reproduce

1.Set Car SoC above current charging level
2.Configure EVCC to use PV mode
3.Wait that charging starts, without necessary criterias are met
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: https
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: casastroh.ml
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval (default 30s)

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: "xxx"

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true
plant: yyyy

# log settings
log: debug
levels:
  site: debug
  lp-1: trace
  lp-2: debug
  cache: debug
  db: debug

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
#  - port: 502
#    uri: 192.168.188.56:1502 # Vom Wechselrichter
#    # rtu: true
#    # readonly: true

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: grid
[... 739 lines cut ...]
[lp-1  ] DEBUG 2023/10/09 21:08:55 charge total import: 2403.032kWh
[lp-1  ] DEBUG 2023/10/09 21:08:55 charger status: B
[lp-1  ] DEBUG 2023/10/09 21:08:55 pv charge current: 0A = 0A + -0.703A (485W @ 3p)
[site  ] DEBUG 2023/10/09 21:09:25 ----
[lp-1  ] DEBUG 2023/10/09 21:09:25 charge power: 0W
[site  ] DEBUG 2023/10/09 21:09:25 pv power: 0W
[site  ] DEBUG 2023/10/09 21:09:25 battery soc: 10%
[site  ] DEBUG 2023/10/09 21:09:25 battery power: 0W
[site  ] DEBUG 2023/10/09 21:09:25 grid power: 242W
[site  ] DEBUG 2023/10/09 21:09:25 site power: 492W
[lp-1  ] DEBUG 2023/10/09 21:09:25 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/09 21:09:25 charge total import: 2403.032kWh
[lp-1  ] DEBUG 2023/10/09 21:09:25 charger status: B
[lp-1  ] DEBUG 2023/10/09 21:09:25 pv charge current: 0A = 0A + -0.713A (492W @ 3p)
[site  ] DEBUG 2023/10/09 21:09:55 ----
.....
```

### What type of operating system are you running?

Linux

### Version

0.120.3

timostr, 2023-10-10:
> Bist du sicher dass das SE-System und die Box keinerlei Zugriffsmöglichkeiten mehr aufeinander haben?
> 
Was ist ein SE-System? EVCC läuft in einem Docker-Container innerhalb von HomeAssistant. EVCC ist direkt (Ethernet mit TCP) mit der Wallbox verbunden. Ich würde erwarten, dass evcc in diesem Fall einen Verbindungsfehler zur Wallbox meldet, oder?
> Das Laden beginnt in dem Moment wo der WR einen (kaputten) NaN-Wert für den Batterie-SoC liefert ohne dass evcc irgendetwas dazu beigetragen hätte. Ebenso müht sich evcc ab den Charger zu deaktivieren was ebenfalls nicht gelingt.
Was für Traces würden helfen, um dieses Problem weiter analysieren zu können?

premultiply, 2023-10-10:
SE = SolarEdge

Klingt mit erstmal nicht so als ob du die Komponenten, insbesondere die Wallbox entsprechend konfiguriert hast so dass sie nichts mehr voneinander wissen. Hier gibt es dann wohl zwei steuernde Instanzen: SE via OCPP+X und dann noch evcc via Modbus.

evcc kann hier in dem Fall nur zuschauen was SE der Wallbox sagt.