evcc-io/evcc #10255: Prozentualer Anteil Sonnenenergie wird in GUI nicht mehr angezeigt

Peter1984Git, 2023-10-10:
### Describe the bug

Anzeige "Sonnenenergie" ist 0%, kein Update mehr.

### Steps to reproduce

Anzeige auf 0%. Wert wurde schon mal korrekt angezeigt.

### Configuration details

```yaml
GNU nano 5.4                                                                                                       evcc.yaml                                                                                                                
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc-pi.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 1234

interval: 30s # control cycle interval

sponsortoken: 1234

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: fronius-gen24
  usage: grid
  host: 192.168.178.39
  port: 502
  name: grid1
- type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.178.39
  port: 502
  name: pv2
- type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.178.39
  port: 502
  name: battery3
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.178.38
  name: pv4

chargers:
- type: template
  template: alfen
  id: 1
  host: 192.168.178.110
  port: 502
  modbus: tcpip
  name: wallbox6

vehicles:
- type: template
  template: id
  title: ID.4 GTX
  user: asdf
  password: 1234
  vin: asdf
  capacity: 77
  name: ev5
  targetSoc: 80

loadpoints:
- title: Carport
  charger: wallbox6
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  enable:
   threshold: 0
   delay: 2m
  disable:
   threshold: 200
   delay: 5m

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    - pv4
    battery:
    - battery3
  #bufferSoC: 100
  #prioritySoC: 50
  residualPower: 100

mqtt:
  broker: 192.168.178.82:1883
  topic: evcc
```

### Log details

```text
evcc --log debug
[main  ] INFO 2023/10/10 09:10:22 evcc 0.120.3
[main  ] INFO 2023/10/10 09:10:22 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/10 09:10:22 starting ui and api at :7070
[db    ] INFO 2023/10/10 09:10:22 using sqlite database: /home/peterreinhardt/.evcc/evcc.db
[mqtt  ] INFO 2023/10/10 09:10:22 connecting evcc-838411122 at tcp://192.168.178.82:1883
[mqtt  ] DEBUG 2023/10/10 09:10:22 tcp://192.168.178.82:1883 connected
[alfen ] DEBUG 2023/10/10 09:10:22 detected 3p alfen
[site  ] WARN 2023/10/10 09:10:25 bufferSoc must be larger than prioritySoc
[site  ] INFO 2023/10/10 09:10:25 site config:
[site  ] INFO 2023/10/10 09:10:25   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/10 09:10:25     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/10/10 09:10:25     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/10 09:10:25     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/10 09:10:25     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/10/10 09:10:25   vehicles:
[site  ] INFO 2023/10/10 09:10:25     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2023/10/10 09:10:25 loadpoint 1:
[lp-1  ] INFO 2023/10/10 09:10:25   mode:        pv
[lp-1  ] INFO 2023/10/10 09:10:25   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/10/10 09:10:25   meters:      charge ✓
[lp-1  ] INFO 2023/10/10 09:10:25     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/10/10 09:10:25 phase timer inactive
[lp-1  ] DEBUG 2023/10/10 09:10:25 pv timer inactive
[lp-1  ] DEBUG 2023/10/10 09:10:25 guard timer inactive
[site  ] DEBUG 2023/10/10 09:10:25 ----
[lp-1  ] DEBUG 2023/10/10 09:10:25 charge power: 0W
[site  ] DEBUG 2023/10/10 09:10:25 pv power: 992W
[site  ] DEBUG 2023/10/10 09:10:25 battery soc: 6%
[site  ] DEBUG 2023/10/10 09:10:25 battery power: -603W
[site  ] DEBUG 2023/10/10 09:10:25 grid power: -9W
[site  ] DEBUG 2023/10/10 09:10:25 grid currents: [0.913 1.04 -0.557]A
[site  ] DEBUG 2023/10/10 09:10:25 site power: -512W
[lp-1  ] DEBUG 2023/10/10 09:10:25 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/10 09:10:25 charge total import: 2185.147kWh
[lp-1  ] DEBUG 2023/10/10 09:10:25 charger status: A
[lp-1  ] INFO 2023/10/10 09:10:25 car disconnected
[lp-1  ] DEBUG 2023/10/10 09:10:25 set charge mode: pv
[lp-1  ] DEBUG 2023/10/10 09:10:25 set min current: 6
[lp-1  ] DEBUG 2023/10/10 09:10:25 set max current: 16
[lp-1  ] DEBUG 2023/10/10 09:10:25 set priority: 0
[site  ] DEBUG 2023/10/10 09:10:55 ----
[lp-1  ] DEBUG 2023/10/10 09:10:55 charge power: 0W
[site  ] DEBUG 2023/10/10 09:10:55 pv power: 1006W
[site  ] DEBUG 2023/10/10 09:10:55 battery soc: 6%
[site  ] DEBUG 2023/10/10 09:10:55 battery power: -615W
[site  ] DEBUG 2023/10/10 09:10:55 grid power: 2W
[site  ] DEBUG 2023/10/10 09:10:55 grid currents: [0.927 1.06 -0.546]A
[site  ] DEBUG 2023/10/10 09:10:55 site power: -513W
[lp-1  ] DEBUG 2023/10/10 09:10:55 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/10 09:10:55 charge total import: 2185.147kWh
[lp-1  ] DEBUG 2023/10/10 09:10:55 charger status: A
[site  ] DEBUG 2023/10/10 09:11:25 ----
[lp-1  ] DEBUG 2023/10/10 09:11:25 charge power: 0W
[site  ] DEBUG 2023/10/10 09:11:25 pv power: 932W
[site  ] DEBUG 2023/10/10 09:11:25 battery soc: 6%
[site  ] DEBUG 2023/10/10 09:11:25 battery power: -608W
[site  ] DEBUG 2023/10/10 09:11:25 grid power: -4W
[site  ] DEBUG 2023/10/10 09:11:25 grid currents: [0.932 1.05 -0.558]A
[site  ] DEBUG 2023/10/10 09:11:25 site power: -513W
[lp-1  ] DEBUG 2023/10/10 09:11:25 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/10 09:11:25 charge total import: 2185.147kWh
[lp-1  ] DEBUG 2023/10/10 09:11:25 charger status: A
```

### What type of operating system are you running?

Linux

naltatis, 2023-10-10:
Meinst du die Anzeige unten rechts im Footer der Anwendung? Oder gehts um die Anzeige am Ladepunkt.

> Anzeige auf 0%. Wert wurde schon mal korrekt angezeigt.

Kannst du das genauer ausführen? Was ist passiert, dass das nicht mehr angezeigt wird? Nach einem Update?

Peter1984Git, 2023-10-10:
> Meinst du die Anzeige unten rechts im Footer der Anwendung? Oder gehts um die Anzeige am Ladepunkt.
> 
> > Anzeige auf 0%. Wert wurde schon mal korrekt angezeigt.
> 
> Kannst du das genauer ausführen? Was ist passiert, dass das nicht mehr angezeigt wird? Nach einem Update?

Ja, es geht um die Anzeige unten rechts im Footer. An der Anzeige am Ladepunkt sehe ich beim aktuellen PV-Überschuss-Laden 99,9%.

Ich weiß leider nicht genau, was passiert, sonst hätte ich es rückgängig gemacht. Ist mir wahrscheinlich nicht gleich aufgefallen. Ich vermute ein Update - oder liegt es an den auskommentierten Werten #bufferSoC: 100 und #prioritySoC: 50?

Peter1984Git, 2023-10-10:
{"result":{"auth":{"vehicles":{}},"battery":[{"power":-3.1,"energy":0,"soc":100,"capacity":0}],"batteryCapacity":0,"batteryConfigured":true,"batteryEnergy":0,"batteryPower":-3.1,"batterySoc":100,"bufferSoc":95,"bufferStartSoc":0,"currency":"EUR","greenShareHome":1,"greenShareLoadpoints":1,"gridConfigured":true,"gridCurrents":[-4.082,-3.101,-4.671],"gridEnergy":2390.387,"gridPower":-2757.7000000000003,"homePower":794.8999999999995,"loadpoints":[{"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":0,"chargePower":0,"chargeTotalImport":2193.986,"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"charging":false,"connected":false,"connectedDuration":9223372036854775807,"enabled":false,"guardAction":"inactive","guardRemaining":0,"maxCurrent":16,"minCurrent":6,"minSoc":0,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":3,"phasesConfigured":0,"phasesEnabled":0,"planActive":false,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"targetSoc":100,"targetTime":"0001-01-01T00:00:00Z","title":"Carport","vehicleDetectionActive":false}],"prioritySoc":5,"pv":[{"power":1426.7,"energy":13134.485200000001},{"power":2129,"energy":0}],"pvConfigured":true,"pvEnergy":13134.485200000001,"pvPower":3555.7,"residualPower":100,"savingsAmount":null,"savingsEffectivePrice":null,"savingsGridCharged":null,"savingsSelfConsumptionCharged":null,"savingsSelfConsumptionPercent":null,"savingsSince":"2023-03-29T21:47:17+02:00","savingsTotalCharged":null,"siteTitle":"Zuhause","smartCostLimit":0,"smartCostType":null,"sponsor":"Peter1984Git","sponsorTokenExpires":0,"vehicles":["ID.4 GTX"],"version":"0.120.3"}}

naltatis, 2023-10-10:
@Peter1984Git Das ist die aktuelle Release-Version 0.120.3 und kein Nightly, richtig?

naltatis, 2023-10-10:
@andig ich steh hier etwas auf dem Schlauch. Unter welchen Umständen kann denn dieser Code hier (woanders publishen wird die Werte nicht) ein `null` in der API erzeugen?

https://github.com/evcc-io/evcc/blob/master/core/savings.go#L137-L142

naltatis, 2023-10-10:
@Peter1984 Hast du seit dem Start von evcc schon mal wieder einen Ladevorgang gehabt? Wenn nicht dann starte mal testweise einen. Diese Werte werden beim Hochfahren einmal veröffentlicht und dann erst wieder während ein Fahrzeug geladen wird. Vielleicht ist nur das erste Publishing schief gegangen und es zieht sich danach wieder gerade.

Das `savingsSince` Datum ist auch korrekt vorhanden. Daher gehe ich davon aus, dass das kein Problem in der Datenbank ist und die Daten weiterhin da sind.

Peter1984Git, 2023-10-10:
@naltatis Ich hatte nicht geladen, aber aktuell läuft ein Ladevorgang aus dem Haus-Akku.

Die Ausgabe ist meiner Ansicht nach gleich. Build ist kein nightly.

{"result":{"auth":{"vehicles":{}},"battery":[{"power":1960.6000000000001,"energy":0,"soc":48.1,"capacity":0}],"batteryCapacity":0,"batteryConfigured":true,"batteryEnergy":0,"batteryPower":1960.6000000000001,"batterySoc":48,"bufferSoc":95,"bufferStartSoc":0,"currency":"EUR","greenShareHome":1,"greenShareLoadpoints":0.9960767530464244,"gridConfigured":true,"gridCurrents":[4.399,-1.9020000000000001,-2.407],"gridEnergy":2390.576,"gridPower":5.5,"homePower":566.1999755859376,"loadpoints":[{"chargeCurrent":6,"chargeCurrents":[6.01800012588501,0,0],"chargeDuration":4357000000000,"chargePower":1401.9000244140625,"chargeRemainingDuration":5929000000000,"chargeRemainingEnergy":2308.666666666399,"chargeTotalImport":2195.704,"chargedEnergy":1716.0000000003492,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"charging":true,"connected":true,"connectedDuration":0,"enabled":true,"guardAction":"inactive","guardRemaining":0,"maxCurrent":16,"minCurrent":6,"minSoc":0,"mode":"minpv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":0,"phasesEnabled":0,"planActive":false,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":1716.0000000003492,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":93.72523770709208,"targetEnergy":0,"targetSoc":55,"targetTime":"0001-01-01T00:00:00Z","title":"Carport","vehicleCapacity":77,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"car","vehicleOdometer":31180,"vehiclePresent":true,"vehicleRange":236,"vehicleSoc":52.301558441558754,"vehicleTargetSoc":80,"vehicleTitle":"ID.4 GTX"}],"prioritySoc":5,"pv":[{"power":2,"energy":13140.46981},{"power":0,"energy":0}],"pvConfigured":true,"pvEnergy":13140.46981,"pvPower":2,"residualPower":100,"savingsAmount":null,"savingsEffectivePrice":null,"savingsGridCharged":null,"savingsSelfConsumptionCharged":null,"savingsSelfConsumptionPercent":null,"savingsSince":"2023-03-29T21:47:17+02:00","savingsTotalCharged":null,"siteTitle":"Zuhause","smartCostLimit":0,"smartCostType":null,"sponsor":"Peter1984Git","sponsorTokenExpires":0,"vehicles":["ID.4 GTX"],"version":"0.120.3"}}

Peter1984Git, 2023-10-10:
[image]
Anbei noch ein Screenshot.

naltatis, 2023-10-10:
Ok danke. Dann liegts nicht am Laden.
Kannst du mal die Savings Werte aus der Tabelle dumpen in:
```
sqlite3 ~/.evcc/evcc.db ".dump 'settings'" | grep "savings"
```

Peter1984Git, 2023-10-10:
Nachdem ich sqlite3 nachinstalliert habe, hier:

INSERT INTO settings VALUES('savings.started','2023-03-30T16:04:58+02:00');
INSERT INTO settings VALUES('savings.gridCharged','0.09658750591475534');
INSERT INTO settings VALUES('savings.gridCost','0.028976251774426608');
INSERT INTO settings VALUES('savings.gridSavedCost','0.11706165703677718');
INSERT INTO settings VALUES('savings.selfConsumptionCharged','0.5320984410762599');
INSERT INTO settings VALUES('savings.selfConsumptionCost','0.0425678752861008');

naltatis, 2023-10-10:
Hm, das sind überraschend kleine Werte. Dass du seit März nur `0,09 kWh` aus dem Netz und  `0,53 kWh` aus Eigenproduktion geladen hast, klingt unwahrscheinlich. Richtig erschließt sich mir aber nicht wie das zustande kommen kann. Auch erklärt das immer noch nicht wieso du in der API `null` Werte bekommst und nicht die aus der Datenbank.

Peter1984Git, 2023-10-10:
Ja, das stimmt auch absolut nicht - ich lade ca. 300 kwh im Monat über PV und kaum was aus dem Netz.

Peter1984Git, 2023-10-11:
Naja, laut Log wird diese verwendet: [db    ] INFO 2023/10/10 09:10:22 using sqlite database: /home/peter/.evcc/evcc.db

Peter1984Git, 2023-10-11:
Ja, mache ich. Heute Mittag wird wieder geladen.

Peter1984Git, 2023-10-11:
So, knapp 17 kwh ins Auto geladen - keine Veränderung der Werte in der DB, auch nicht nach Neustart von evcc. Wie kann ich sicherstellen, dass das die richtige DB ist die wir hier prüfen?

andig, 2023-10-11:
Und was sagt das trace log? Werden Statements gegen die DB ausgeführt? Welche? Erscheinen die Daten wenn Du die statements direkt in SQlite ausführst?

Peter1984Git, 2023-10-11:
Wie kann ich das direkt in SQlite ausführen? Davon habe ich keine Ahnung. Tracelog kommt später.

Peter1984Git, 2023-10-11:
Ich schicke dir das Tracelog direkt, ich weiß nicht ob da vertrauliche Daten enthalten sind.

Peter1984Git, 2023-10-12:
Kam das Tracelog bei dir per email an?

TMoersch, 2023-10-15:
Hallo,
ich habe ein ähnliches Thema, evtl. von der Ursache her gleichgelagert. Meine evcc Version ziegt dort konstant viel zu kleine Werte an, 44% und jetzt kleiner werdend, obwohl ich zu 97,5% aus dem PV Akku lade. Mir erscheinen hier teilweise Werte bei denen "." und "," falsch angezeigt werden, das kann aber evtl nur im GUI sein. Die Gesamtstrommenge wird auch mit utopischen >60.000 kWh angebeben, das können weder 60kWh noch 60000 sein. War für mich bisher aber nur Kosmetik, man konnte damit nicht posen, sondern musste erst in die Ladehistorie einsteigen. Diese Hostorie war aber Anfang des Jahres mit einem Update auch schon einmal durch den Wind (unvollständig / falsche Werte), aktuell stimmt sie aber wieder.
Vielleicht hilftes weiter. 
btw.
Seit dem Update gestern startet evcc in HomeAssistent super langsam, aktuell scheint es gar nicht mehr zu starten bei mir.

naltatis, 2023-10-15:
@TMoersch Bist du auf dem Nightly? Dass das nicht startet, könnte hier ran liegen: https://github.com/evcc-io/evcc/issues/10321 Das sollte morgen wieder laufen.

> Mir erscheinen hier teilweise Werte bei denen "." und "," falsch angezeigt werden, das kann aber evtl nur im GUI sein.

Hast du hier was konkretes? In der UI formatieren wir die Werte abhängig von der gewählten Sprache. Sollte allerdings ein reines Anzeigethema sein und keine Auswirkung auf die Daten in der Datenbank haben.

TMoersch, 2023-10-15:
Hallo naltatis,
nein, ich habe nur die mir offiziellen zum Update angebotene Version 0.121.0.
evcc hat sich bei mir auch wieder eingekriegt, der Start dauerte aber gefühlt ewig. Hatte es im debug mode gestartet aber ab einem bestimmten Zeitpunkt kam halt minutenlang rein gar nix mehr im log.... 
Bei mir steht in der Summenanzeige 31% obwohl tatsächlich über die Aufzeichnung der Ladevorgänge irgendwas >95% stehen sollte. Ist aber wie gesagt für mich Kosmetik, wäre halt gut wenn man einfach zeigen könnte "...hier, schau mal 95% mit Sonne getankt". Sonst bin ich vollkommen zufrieden.

Hier die Aussage mal konkret
Sonnenenergie
31,5%
44.597 kWh Sonne
97.110 kWh Netz
Ich habe keine Ahnung woraus sich diese Zahlen zusammensetzen.
Laut Ladeaufzeichnung im September 198,5 kWh geladen und davon 97,7 Sonne, was korrekt ist.
Ist es möglich, dass die Summeanzeige nicht mit der gleichen Berechnungsmethode erzeugt wird oder das einfach an einer Stelle mal aufgezeichneter schrott in der Datenbasis steht? 
Durch das AddOn in HA habe ich quasi keinen Zugang zu den tieferliegenden Teilen.

Alternativ habe ich schon darüber nach gedacht den ganzen Raspi neu zu installieren. 
Ist nur blöd das man dann die ganzen Werte aus der Vergangeheit verliert.

Von mir aus muss mein Teil nicht weiter verfolgt werden, wollte ja auch keinen eigenen Thread dazu eröffenen.

Vielen Dank für Eure Arbeit.
Thomas

closed 2023-10-20 by andig

Peter1984Git, 2023-10-20:
Ist es gefixt?

Schnup89, 2024-01-31:
Hallo zusammen,

ich schließe mich den Problem an.
Das System läuft seit Version 0.105.2 und wurde bisher immer über die .deb Datei aktualisiert.

Die "Sonnenergie" wir überall mit 0% angezeigt, im Footer wie auch in den Auswertungen unter "Ladevorgänge".

/api/state zeigt keine savings Werte:
`{"result":{"auth":{"vehicles":{}},"batteryConfigured":false,"batteryDischargeControl":false,"batteryMode":"unknown","bufferSoc":0,"bufferStartSoc":0,"currency":"EUR","greenShareHome":0,"greenShareLoadpoints":0,"gridConfigured":true,"gridPower":1369,"homePower":1369,"loadpoints":[{"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":0,"chargePower":0,"chargeTotalImport":1547.752,"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":"","charging":false,"connected":false,"connectedDuration":9223372036854775807,"disableThreshold":0,"effectiveLimitSoc":100,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanSoc":0,"effectivePlanTime":"0001-01-01T00:00:00Z","effectivePriority":0,"enableThreshold":0,"enabled":false,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":6,"mode":"minpv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":3,"phasesConfigured":null,"phasesEnabled":3,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"title":"Garage","vehicleCapacity":40,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"","vehicleName":"Megane","vehicleOdometer":12922.650390625,"vehicleRange":0,"vehicleSoc":0,"vehicleTargetSoc":0}],"prioritySoc":0,"pv":[{"power":0}],"pvConfigured":true,"pvEnergy":0,"pvPower":0,"residualPower":0,"siteTitle":"Haus","smartCostActive":false,"smartCostLimit":0,"smartCostType":null,"statistics":{"30d":{"avgCo2":0,"avgPrice":0,"chargedKWh":169.692,"solarPercentage":0},"365d":{"avgCo2":0,"avgPrice":0,"chargedKWh":1138.823999999999,"solarPercentage":33.68713296744939},"total":{"avgCo2":0,"avgPrice":0,"chargedKWh":1138.823999999999,"solarPercentage":33.68713296744939}},"vehicles":{"Megane":{"title":"Megane E-Tech"}},"version":"0.123.9"}}`

Genauso die sql-Abfrage --  sqlite3 ~/.evcc/evcc.db ".dump 'settings'"  --
```
PRAGMA foreign_keys=OFF;
BEGIN TRANSACTION;
CREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`));
INSERT INTO "settings" VALUES('vehicle.Megane.planTime','0001-01-01T00:00:00Z');
INSERT INTO "settings" VALUES('vehicle.Megane.planSoc','0');
COMMIT;
```

Hier noch die Config:

```
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 8888

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

mqtt:
  broker: 192.168.0.35:1883
  user: xxxxxxxxxxxxxx
  password: xxxxxxxxxxxxxxxxx

meters:
- name: grid
  type: custom ##Leistungswert aus solarview EVU-Zähler
  power:
    source: mqtt
    topic: evcc/Bezug
- name: pv
  type: custom
  power:    ##Leistungswert aus solarview Umrichter
    source: mqtt
    topic: evcc/Erzeugung
- name: wbec-heidelberg-ec-meter
  type: custom
  power:
    source: mqtt
    topic: wbec/lp/1/power
  energy:
    source: mqtt
    topic: wbec/lp/1/energy
  currents:
    - source: mqtt
      topic: wbec/lp/1/currL1
    - source: mqtt
      topic: wbec/lp/1/currL2
    - source: mqtt
      topic: wbec/lp/1/currL3

chargers:
- name: wbec-heidelberg-ec
  type: custom
  status:
    source: mqtt
    topic: wbec/lp/1/status
  enabled:
    source: mqtt
    topic: wbec/lp/1/enabled
  enable:
    source: mqtt
    topic: wbec/lp/1/enable
  maxcurrent:
    source: mqtt
    topic: wbec/lp/1/maxcurrent

loadpoints:
- title: Garage
  charger: wbec-heidelberg-ec
  meter: wbec-heidelberg-ec-meter
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  #resetOnDisconnect: true
  vehicle: Megane

interval: 3s # control cycle interval

vehicles:
- type: renault
  title: Megane E-Tech
  user: xxxxxxxxxx
  password: xxxxxxxxxxxx
  vin: xxxxxxxxxxxxxx
  capacity: 40
  name: Megane
#  phases: 1

site:
  title: Haus
  meters:
    grid: grid
    pv: pv

```

Bilder:
[image]

[image]