evcc-io/evcc #10260: Ford Integration only getting 403

appcoders, 2023-10-10:
### Describe the bug

Since about 2 days the integration of ford does not work: Always getting a 403 for SOC. User/Password is working on FordPass App.

### Steps to reproduce

Starting evcc

### Configuration details

```yaml
:
:
vehicles:
- type: template
  template: ford 
  title: Mach-E  
  user: ****** 
  password: ******  
  vin: ******  
  capacity: 90  
  name: ev4
:
:
```

### Log details

```text
evcc --log trace | grep ford
[ford  ] TRACE 2023/10/10 12:26:21 GET https://sso.ci.ford.com/v1.0/endpoint/default/authorize?client_id=9fb503e0-715b-47e8-adfd-ad4b7770f73b&code_challenge=ukJUIIALozSb2itppUlkAX0OQjSNDXRCUoxX8e0w_5w&code_challenge_method=S256&max_age=3600&redirect_uri=fordapp%3A%2F%2Fuserauthorized&response_type=code&scope=openid&state=GSLufCbx2rCZjN5P
[ford  ] TRACE 2023/10/10 12:26:22 GET https://sso.ci.ford.com/idaas/mtfim/sps/idaas/login?client_id=9fb503e0-715b-47e8-adfd-ad4b7770f73b&identity_source_ids=75d08ad1-510f-468a-b69b-5ebc34f773e3&max_age=3600&Target=https%3A%2F%2Fsso.ci.ford.com%2Foidc%2Fendpoint%2Fdefault%2Fauthorize%3FqsId%3D8602807b-1954-4712-b7ae-230e47552888%26client_id%3D9fb503e0-715b-47e8-adfd-ad4b7770f73b
[ford  ] TRACE 2023/10/10 12:26:22 GET https://sso.ci.ford.com/authsvc/mtfim/sps/authsvc?PolicyId=urn:ibm:security:authentication:asf:basicldapuser&identity_source_id=75d08ad1-510f-468a-b69b-5ebc34f773e3&Target=https%3A%2F%2Fsso.ci.ford.com%2Foidc%2Fendpoint%2Fdefault%2Fauthorize%3FqsId%3D8602807b-1954-4712-b7ae-230e47552888%26client_id%3D9fb503e0-715b-47e8-adfd-ad4b7770f73b
[ford  ] TRACE 2023/10/10 12:26:22 <!doctype html>
  <meta id="fma-version" version="fma-5.32.0" version-url="https://www.account.ford.com/etc/fd/fma/manifest.json">
  <link id="favicon" rel="shortcut icon" type="image/x-icon" href="https://www.account.ford.com/favicon.ico">
  <link id="completeStyle" href="https://www.account.ford.com/static/css/main.d9f26451.chunk.css" rel="stylesheet">
  <script type="text/javascript" id="SCRIPT_RUNTIME" src="https://www.account.ford.com/static/js/runtime-main.f4de8767.js"></script>
  <script type="text/javascript" id="SCRIPT_CHUNK" src="https://www.account.ford.com/static/js/2.cc3aef49.chunk.js"></script>
  <script type="text/javascript" id="SCRIPT_MAIN" src="https://www.account.ford.com/static/js/main.0229317c.chunk.js"></script>
                                                window.location.hostname.indexOf('sso.ci.ford.com') > -1 ||
[ford  ] TRACE 2023/10/10 12:26:22 POST https://sso.ci.ford.com/authsvc/mtfim/sps/authsvc?identity_source_id=75d08ad1-510f-468a-b69b-5ebc34f773e3&StateId=28c11110-f46f-4e73-8f7c-109bc9e75848
[ford  ] TRACE 2023/10/10 12:26:22 login-form-type=pwd&operation=verify&password=***&username=***
<P><A HREF="https://sso.ci.ford.com/oidc/endpoint/default/authorize?qsId=8602807b-1954-4712-b7ae-230e47552888&client_id=9fb503e0-715b-47e8-adfd-ad4b7770f73b&identity_source_id=75d08ad1-510f-468a-b69b-5ebc34f773e3">Click here</A> to fetch the resource.
[ford  ] TRACE 2023/10/10 12:26:22 GET https://sso.ci.ford.com/oidc/endpoint/default/authorize?qsId=8602807b-1954-4712-b7ae-230e47552888&client_id=9fb503e0-715b-47e8-adfd-ad4b7770f73b&identity_source_id=75d08ad1-510f-468a-b69b-5ebc34f773e3
[ford  ] TRACE 2023/10/10 12:26:23 POST https://sso.ci.ford.com/oidc/endpoint/default/token
[ford  ] TRACE 2023/10/10 12:26:23 code=Vt4NrpGsjPRfFblnClkrD9waW3i_RGpsMObiJ0MRtso.SkwwDtuX5fAbXPK6nvK2aFvbaNekyHO-863vgjPkfeqSfR3eeBbU16Vc_3JjPEEia5A8-LCqZXNar3xRh3nZTQ&code_verifier=MTF3WXRvMDhTQkdkR0NyRVNYdlIxME80VTFPdmdwcjA&grant_type=authorization_code&redirect_uri=fordapp%3A%2F%2Fuserauthorized
[ford  ] TRACE 2023/10/10 12:26:23 POST https://api.mps.ford.com/api/token/v2/cat-with-ci-access-token
[ford  ] TRACE 2023/10/10 12:26:24 {"ciToken":"eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6InNlcnZlciJ9.eyJhY3IiOiJ1cm46aWJtOnNlY3VyaXR5OnBvbGljeTppZDoxIiwiZ3JhbnRfdHlwZSI6ImF1dGhvcml6YXRpb25fY29kZSIsInJlYWxtTmFtZSI6ImNsb3VkSWRlbnRpdHlSZWFsbSIsIkFNUiI6IiIsInVuaXF1ZVNlY3VyaXR5TmFtZSI6IjY2NDAwMlNZTEkiLCJncmFudF9pZCI6IjgxNTQ1ODI3LTMyNWUtNDU2Ni05YzVlLTc0NTgxNDY4MDhjNiIsInVzZXJUeXBlIjoicmVndWxhciIsInByZWZlcnJlZF91c2VybmFtZSI6Impvc2VmLmxlaXRsQGdteC5kZSIsImFwcF9pZCI6IjE5OTgwODc4NTM1OTA4NzI3NzkiLCJhdWQiOiI5ZmI1MDNlMC03MTViLTQ3ZTgtYWRmZC1hZDRiNzc3MGY3M2IiLCJhdXRoX3RpbWUiOjE2OTY5MzM1ODIsInNjb3BlIjoib3BlbmlkIiwianRpIjoiUEVpaEQ5T0tBbWhWbllsVjQ0eEl1Qm9Qd2FVNlRLIiwiY2xpZW50X2lkIjoiOWZiNTAzZTAtNzE1Yi00N2U4LWFkZmQtYWQ0Yjc3NzBmNzNiIiwiaXNzIjoiaHR0cHM6Ly9mY2lzLmljZS5pYm1jbG91ZC5jb20vb2lkYy9lbmRwb2ludC9kZWZhdWx0Iiwic3ViIjoiNjY0MDAyU1lMSSIsImlhdCI6MTY5NjkzMzU4MywiZXhwIjoxNjk2OTMzODgzfQ.N0edyjdCgVUUUfBhBj2PqtfhsNLot1kh62HZcjNWSBIuxmPp_WMb0AREt2ZwJD_T6ACSfb1NUhmax_wpe6Qm_dbjXdsiw0ux7UvW5_zxk16yC_sdPkgUr7PjYLXIcEpCqxeEGrPNQZDbugzMPk-FYP19cXxpAw3XLZXT47id02CdEOi0sUzJR6Hn86hoFYH_uHeqBKH9L_7Zg1lKSnuJtYaqpT-s6QhHdAhwcyceCzWSrHmmyEaMKiKZAMfnW8cXgmPbNg3PEWMqTQNTNbsY-qFcBYtGPjyk6XoAGnTa34CbIhJkLh2UBDSVpoe8ENIGKPaZVEvxGBquyBwJijzJVg"}
{"access_token":"eyJ0eXAiOiJKV1QiLCJ0b2tlbl90eXBlIjoiQSIsImFsZyI6IlJTMjU2Iiwia2lkIjoiY3NkbmtleTEifQ.eyJzdWIiOiJlZGJhOGQ1Yy1hMDE3LTQxZGEtODZhMi03NWNhNDQxOGY4ZjMiLCJhdWQiOiIxZThjNzc5NC1mZjVmLTQ5YmMtOTU5Ni1hMWUwYzg2YzViMTkiLCJpc3MiOiI1RDQ3MjAyNi1FMjZBLTQ0NzItODVGMS03QkJDRDUzMEU3OEMiLCJ0eXBlIjoic3NvIiwiZXhwIjoxNjk2OTM1Mzg0LCJpYXQiOjE2OTY5MzM1ODQsImp0aSI6ImVkYmE4ZDVjLWEwMTctNDFkYS04NmEyLTc1Y2E0NDE4ZjhmM3wxNjk2OTMzNTg0NTkxNjk2MDAwfGEiLCJ1c2VybmFtZSI6Impvc2VmLmxlaXRsQGdteC5kZSJ9.HQJ9vAnxSYMuOcKeVzPUJrHN58A2iCyYoWoKPTYJ6nC-xhLVgTFpbBwwXhoZEOhb-d_iSx4ZFrqQHL9D1cMFs3NvmL5te9mGS7M3_emCMvALpNPcsanzjJQEc78PfrPw-8DhA4PddmoPk0HrYYr8DZFLooofTWYo4R-_4Z9PqhqT8E3Sb9vVgP6hcfmHgavNdL_Q1QRX7Nn_zn6ym8CeRf3BiOrYGeFJHiujUH1juebII8gYaKZ8nWsLXwngVtwgSdsxVDuS1akV07V5H7aTYvpUp57GLe2VkrX8MM1WY3L8Q5EK2X0GpkMo6SyfteeB2fTAWTPzu0u36IsX-S1ZXw","refresh_token":"eyJ0eXAiOiJKV1QiLCJ0b2tlbl90eXBlIjoiUiIsImFsZyI6IlJTMjU2Iiwia2lkIjoiY3NkbmtleTEifQ.eyJzdWIiOiJlZGJhOGQ1Yy1hMDE3LTQxZGEtODZhMi03NWNhNDQxOGY4ZjMiLCJhdWQiOiIxZThjNzc5NC1mZjVmLTQ5YmMtOTU5Ni1hMWUwYzg2YzViMTkiLCJkZXZpY2VfaWQiOm51bGwsImlzcyI6IjVENDcyMDI2LUUyNkEtNDQ3Mi04NUYxLTdCQkNENTMwRTc4QyIsInR5cGUiOiJzc28iLCJleHAiOjE3Mjg1NTU5ODQsImlhdCI6MTY5NjkzMzU4NCwianRpIjoiZWRiYThkNWMtYTAxNy00MWRhLTg2YTItNzVjYTQ0MThmOGYzfDE2OTY5MzM1ODQ1OTMzODcwMDAiLCJ1c2VybmFtZSI6Impvc2VmLmxlaXRsQGdteC5kZSJ9.K1wusGnKlgDmawxqP3xSoi2w5AhdEyRCxtxrIIAnwC87je-ZC8jWWuDJsEcQPmff6aKzyaIh0qfQA6D6x__Cvr8x5XTvpnazYRwEB-97I69RktC_XDYb_8W6qK7L3Nx7mfTutidpz-906EZlEXvW5_WMfnVLoo585GtWSvjRUZ0ddihjrvNta4YchCQYgoztFbkNrBOy4mdCG4vTBUdyaK7EZAhaLImNMf2148lFS_QJYNrHsUHcZCwzd34t9XFVfTcPAhpj6TF_IP325Xot9sW7GoSssM7K5uuJ1IjXs4_FJNA0kRKa9kZLvKMht-ZC_y7lEKWQtbvxJyT2X7iPUQ","ford_consumer_id":"edba8d5c-a017-41da-86a2-75ca4418f8f3","expires_in":1800,"refresh_expires_in":31556952}
[ford  ] TRACE 2023/10/10 12:26:24 POST https://api.mps.ford.com/api/expdashboard/v1/details
[ford  ] TRACE 2023/10/10 12:26:26 {"dashboardRefreshRequest":"All"}
{"userVehicles":{"vehicleDetails":[{"VIN":"***","nickName":"Ford Mustang Mach E ","tcuEnabled":true,"isASDN":false}],"status":{"cache-control":"max-age=86400","last_modified":"Tue, 10 Oct 2023 10:26:25 GMT","statusCode":"200"}},"vehicleProfile":[{"VIN":"***","showEVBatteryLevel":true,"showFuelLevel":false,"cabinCargoDoubleUnlock":"False","lastMileNavigated":true,"alarmFunctionality":"True","asBuiltCountry":"DEU","year":2022,"heatedSteeringWheel":false,"numberOfLightingZones":"0","cabinCargoUnlock":"False","activationType":"HMI","transmissionIndicator":"A","bodyCompletionState":"Unknown","plugAndChargeToggleOnStatus":false,"warrantyStartDate":"2022-06-21","recallCount":0,"model":"Mustang Mach-E","zoneUnlock":"False","fuelLevelApplink":false,"proPowerZones":"0","oilLifeApplink":false,"usageBasedInsuranceLevel":"Enhanced","healthDataViaApplink":false,"rearCargoArea":"Power","blackLabelTrim":false,"doubleLocking":"False","driverHeatedSeat":"None","usageBasedInsurance":"Available","nonRecallCount":0,"otaSettingsScheduler":true,"proPowerApplink":false,"remoteClimateControl":false,"sendNavigationToCar":false,"sdn":"TMC","displayRecommendedTirePressure":false,"vehicleImage":"https://www.gpas-cache.ford.com/composite?VIN=***&angleNo=0&viewType=exterior","numberOfTires":"Four","remoteParkAssistLevel":"None","engineType":"BEV","frontCargoArea":"eLatch","lockStatus":"Inferred","make":"Ford","displaySmartCharging":"Unavailable","productType":"C","grossVehicleWeightRatingPounds":0,"paintDescription":"ABSOLUTE BLACK ","modemDeactivated":"None","proPowerWattage":"None","onboardScaleSmartHitch":"None","vehicleImageLabel":"2022 Fo [... cut ...]
[ford  ] TRACE 2023/10/10 12:26:26 { "statusCode": 403, "message": "Forbidden" }
[ford  ] TRACE 2023/10/10 12:26:38 GET https://usapi.cv.ford.com/api/vehicles/v5/***/status
[ford  ] TRACE 2023/10/10 12:26:38 { "statusCode": 403, "message": "Forbidden" }
[ford  ] TRACE 2023/10/10 12:26:56 GET https://usapi.cv.ford.com/api/vehicles/v5/***/status
[ford  ] TRACE 2023/10/10 12:26:56 { "statusCode": 403, "message": "Forbidden" }
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.3

tobox, 2023-10-10:
In Home-Assistent, the problem has already been solved. Ford switched to a completly new API (again...):

https://github.com/itchannel/fordpass-ha/issues/303

EWG6EH, 2023-10-10:
@andig denke da kommt ein EVCC Update ?

naltatis, 2023-10-11:
Siehe auch https://github.com/evcc-io/evcc/discussions/10257

Fs3107, 2023-10-11:
Same problem since a few days, no changes made

andy-pv-21, 2023-10-11:
Same problem with IO-Broker 
https://usapi.cv.formd.com/api/vehicles/v4/FINXXXXXXXXXXXXX/status

rotfussl, 2023-10-11:
I do have exactly the same with my Mach-E in Germany.

andy-pv-21, 2023-10-11:
> > Same problem with IO-Broker
> 
> Did you try current version?

Sure. No Update for the Ford adapter available.

closed 2023-10-12 by andig

andy-pv-21, 2023-10-12:
0.0.15 biete er mir nicht an. Sobald ich sie habe, teste ich asap

rotfussl, 2023-10-13:
Bei mir sieht es nach erstem Test gut aus!!
Vielen Dank für den Einsatz!!!

Antwort an: evcc-io/evcc
An: evcc-io/evcc ***@***.***>
Kopie: rotfussl ***@***.***>, Comment
#10260)
Datum: Thu, 12 Oct 2023 10:00:33 -0700

Aktuell ist 0.1.5, die geht bei mir.
Ein erster Lösungsansatz auf der Basis ist drin. Falls veraltete Daten
kommen muss ich den Refresh noch nachrüsten, dafür bitte erstmal
testen.

appcoders, 2023-10-17:
Works again. Thanks