evcc-io/evcc #10265: Ladevorgang stoppt bei Ladeplaner->Ankunft 50% nach kurzer Zeit obwohl mit 11kW weiter geladen wird

jheinitz, 2023-10-10:
### Describe the bug

Moin!

Jetzt wo das Wetter schlechter wird und der Überschuss kaum ausreicht, um überhaupt 10% in das Auto zu laden, habe ich mal den Ladeplaner für die Ankunft ausprobiert. Ich habe den Wert auf 50% gestellt und der Ladevorgang startet auch mit 11kW. Allerdings endet der Ladevorgang nach 21s und startet auch nicht wieder. Allerdings zieht der Ladepunkt weiter knapp 11kW und das Auto wird auch geladen. Da der Ladezustand des Autos aufgrund des beendeten Ladevorgangs von evcc nicht mehr aktualisiert wird, wird der Ladevorgang auch nicht bei 50% beendet. Im evcc GUI wird weiter 32% angezeigt. Über die Skoda App konnte ich sehen, dass das Auto bereits auf 59% geladen worden ist. Ich beende den Vorgang in evcc durch Wechsel des Modus von **PV** auf **Off**. 

Hat jemand eine Idee? Es ist bis jetzt nur eine Vermutung, dass es an dem Ankunfts-Ladeplaner liegt. Am Donnerstag könnte ich detailierte Traces liefern. Meine Konfiguration und Auszüge aus dem Log habe ich hier angehängt.

Danke und beste Grüße

Jens

### Steps to reproduce

1. Ladeplaner für Ankunft auf 50% stellen
2. Fahrzeug mit SOC <50% anschließen (ChargerStatus: A)
3. Ladevorgang startet (ChargerStatus: C)
4. Ladevorgang wird nach kurzer Zeit unterbrochen (ChargerStatus: B)
5. Wallbox lädt das Fahrzeug weiter über 50% hinaus
6. evcc zeigt keinen aktualisierten SoC des Autos an

### Configuration details

```yaml
---
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  # site: trace
  # lp-1: debug
  # messaging: debug
  # easee: trace
  # enyaq: debug
  # cache: error
  cache: debug

# unique installation id
plant: <my plant id>

interval: 20s # control cycle interval

sponsortoken: <my sponsor token>

# database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /db/evcc.db

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

mqtt:
  broker: server2.example.com:1883
  topic: evcc
  clientid: evcc
  user: evcc
  password: <my mqtt password>

meters:
  - name: grid1
    type: custom
    power:
      source: mqtt
      topic: modbus/power/sun2000/meter
[... 434 lines cut ...]
# removed a bunch of lines with no change
# .......
[site  ] DEBUG 2023/10/10 20:49:11 ----
[lp-1  ] DEBUG 2023/10/10 20:49:11 charge power: 10854W
[site  ] DEBUG 2023/10/10 20:49:11 pv power: 0W
[site  ] DEBUG 2023/10/10 20:49:11 battery soc: 5%
[site  ] DEBUG 2023/10/10 20:49:11 battery power: 0W
[site  ] DEBUG 2023/10/10 20:49:11 grid power: 9416W
[site  ] DEBUG 2023/10/10 20:49:11 grid currents: [16.1 16.5 16.3]A
[site  ] DEBUG 2023/10/10 20:49:11 site power: 9666W
[lp-1  ] DEBUG 2023/10/10 20:49:11 charge currents: [15.5 15.5 15.5]A
[lp-1  ] DEBUG 2023/10/10 20:49:11 charge total import: 5264.875kWh
[lp-1  ] DEBUG 2023/10/10 20:49:11 charger status: B
[lp-1  ] DEBUG 2023/10/10 20:49:11 !! scalePhasesIfAvailable: 3p -> 3p
[lp-1  ] DEBUG 2023/10/10 20:49:11 !! scalePhases: GetPhases 3p <> 3p wanted
jens@server2:~$
```

### What type of operating system are you running?

Docker container

### Version

0.120.3

andig, 2023-10-10:
>  Ich habe den Wert auf 50% gestellt und der Ladevorgang startet auch mit 11kW. Allerdings endet der Ladevorgang nach 21s und startet auch nicht wieder. 

Im Log ist nichts von einem Ladeplan zu sehen. Woran machst Du fest dass das hier irgendwas mit Ladeplanung zu tun hat?

andig, 2023-10-10:
Hier ist eher zu sehen, dass die Easee von sich aus mit laden beginnt was sich leider nicht verhindern lässt. Mit Ladeplanung hat das nichts zu tun.

andig, 2023-10-10:
@GrimmiMeloni das (Easee) ist allerdings maximal unschön:

    [lp-1 ] DEBUG 2023/10/10 18:59:31 charge currents: [15.4 15.5 15.4]A
    [lp-1 ] DEBUG 2023/10/10 18:59:31 charge total import: 5253.678kWh
    [lp-1 ] DEBUG 2023/10/10 18:59:31 charger status: B

jheinitz, 2023-10-11:
> >  Ich habe den Wert auf 50% gestellt und der Ladevorgang startet auch mit 11kW. Allerdings endet der Ladevorgang nach 21s und startet auch nicht wieder. 
> 
> Im Log ist nichts von einem Ladeplan zu sehen. Woran machst Du fest dass das hier irgendwas mit Ladeplanung zu tun hat?

Im UI ist der Balken rot und da steht auch etwas von "Minimalladen bis 50%" oder so ähnlich.

andig, 2023-10-11:
Oder so ähnlich?

jheinitz, 2023-10-11:
> Oder so ähnlich?

Ich hab's nicht auswendig gelernt... 😉

jheinitz, 2023-10-11:
> > @GrimmiMeloni das (Easee) ist allerdings maximal unschön:
> > ```
> > [lp-1 ] DEBUG 2023/10/10 18:59:31 charge currents: [15.4 15.5 15.4]A
> > [lp-1 ] DEBUG 2023/10/10 18:59:31 charge total import: 5253.678kWh
> > [lp-1 ] DEBUG 2023/10/10 18:59:31 charger status: B
> > ```
> 
> Ja, da ist was merkwürdiges passiert. Vielleicht ein Easee Schluckauf? @jheinitz Falls Du es reproduzieren kannst - ein Trace Log wäre (wie immer ;) ) super. Spannend wäre zusätzlich auch, was in so einem Moment in der Easee App angezeigt wird.

Moin @GrimmiMeloni 
In der Easee App konnte ich sehen, dass das Auto mit 10,9kW geladen wurde. Und auch die Skoda app zeigte an, dass geladen wurde und auh der Soc des Enyaq stieg langsam aber stetig.

Werde es am Donnerstag nochmal versuchen zu reproduzieren.

jheinitz, 2023-10-11:
> > Ich habe den Wert auf 50% gestellt und der Ladevorgang startet auch mit 11kW. Allerdings endet der Ladevorgang nach 21s und startet auch nicht wieder.
> 
> Im Log ist nichts von einem Ladeplan zu sehen. Woran machst Du fest dass das hier irgendwas mit Ladeplanung zu tun hat?

Moin @andig 
was für Meldungen müssten denn Deiner Meinung nach im Log auftauchen, wenn der Ladeplaner aktiv ist? Muss ich da noch etwas mehr DEBUG aktivieren?

andig, 2023-10-11:
Der rote Balken wäre minSoc. Da müsste IMMER geladen werden was zum Verhalten, aber nicht zum Log passt. Ich verstehe grad nur Bahnhof. Neuer Anlauf mit trace log wäre gut.

jheinitz, 2023-10-11:
> Der rote Balken wäre minSoc. Da müsste IMMER geladen werden was zum Verhalten, aber nicht zum Log passt. Ich verstehe grad nur Bahnhof. 

Hi! 
Ich beziehe mich diesen [Blogeintrag ](https://docs.evcc.io/blog/2023/10/05/feature-highlights-10-2023/?s=09#mindestladung-%C3%BCber-ui-einstellbar) auf den Abschnitt "Mindestladung über UI einstellbar). 
Was du mit **minSoc** meinst, ist das was man in der YAML Datei am `vehicle` einstellen kann, oder?

> Neuer Anlauf mit trace log wäre gut.
Mach ich am Donnerstag. 

Danke und bis dann.

jheinitz, 2023-10-12:
Moin!

Ich habe das Ganze heute nochmal probiert und konnte das Verhalten vom ersten Mal nicht reproduzieren. Das Auto fing an zu laden und nachdem der Skoda Enyaq erkannt worden ist, wurde auch das Minimal Laden aktiviert.

Nach Erreichen von 50% wurde auf normalen PV Modus gewechselt. Dieser wiederum wurde dann ausgesetzt, weil kein PV Überschuss mehr vorhanden war. 
Im Log des Docker Containers war nichts vom Minimalen Laden zu sehen. Deshalb habe ich dafür mal das Logging eingebaut. Der PR dazu ist #10282 

Ich mach das Ticket hier erstmal zu. Danke für die Hilfe.

Viele Grüße

Jens

closed 2023-10-12 by jheinitz