evcc-io/evcc #10267: evcc can't connect to Kia API

flexguse, 2023-10-11:
### Describe the bug

In the last days, let's say 10, my evcc is not able to connect to Kia to read the charge state of my EV. There are no entries in the logs.
I use the current evcc version 0.120.3, this behavior was also in 0.120.1 and 0.120.2. In the Kia App I'm able to connect and see / change values.

I checkt the docs if the configuration for the vehicles changed and I removed the `targetSoc: 90` setting which did not help.

Maybe Kia again changed the login?

### Steps to reproduce

1. run `evcc vehicle` in command line
2. see logs

### Configuration details

```yaml
vehicles:
  - name: niro
    type: template
    template: kia
    title: Niro EV
    user: xxx
    password: xxx
    capacity: 64
```

### Log details

```text
/app # evcc vehicle
[main  ] INFO 2023/10/11 12:20:49 evcc 0.120.3
[main  ] INFO 2023/10/11 12:20:49 using config file: /etc/evcc.yaml
[db    ] INFO 2023/10/11 12:20:49 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/10/11 12:20:49 connecting evcc-123211189 at tcp://xx.xxx.xx:1883
[mqtt  ] DEBUG 2023/10/11 12:20:49 tcp://xx.xxx.xx:1883 connected
[main  ] ERROR 2023/10/11 12:20:50 creating vehicle niro failed: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
Capacity: 64.0kWh
Features: [Offline]
```

### What type of operating system are you running?

Docker container

### Version

0.120.3

andig, 2023-10-11:
Geht es mit ioBroker?

andig, 2023-10-12:
Anscheinend temporäre Probleme mit Bluelink. Konnte es gerade erfolgreich testen.

closed 2023-10-12 by andig

flexguse, 2023-10-14:
Mittlerweile funktioniert bei mir der Ladestatus auch wieder.