evcc-io/evcc #10269: Änderung bei maxcurrent wird "versetzt" ausgeführt

VolkerK62, 2023-10-11:
### Describe the bug

Wenn ich per API oder im UI `maxcurrent` ändere, dann hat das beim ersten Mal keine Auswirkung. Wenn ich danach nochmal eine Änderung schicke, wird die ausgeführt die vorher geschickt wurde.
Siehe Log:
13:43:11 Start Mode now mit 20A
13:44:19 set maxcurrent auf 10A ; keine Reaktion
13:45:53 set maxcurrent auf 6A ; es wird auf 10A reduziert
13:47:15 set maxcurrent auf 20A ; es wird auf 6A reduziert

### Steps to reproduce

siehe oben

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

network:
  schema: http
  host: evcc.local
  port: 7070

interval: 10s

sponsortoken: *****

telemetry: true

log: debug
levels:
  #site: trace
  #lp-1: info
  lp-2: info
  lp-3: info
  lp-4: info
  #daheimladen-mb: trace
  #hyundai: trace
  #main: trace
  #cache: trace
  #db: trace
  #http: trace
  #influx: trace
  #mqtt: trace
  #proxy-5200: trace

########################################################################
meters:
- name: grid
  type: template
  template: shelly-3em
  usage: grid
  host: 192.168.178.28

- name: pv
  type: template
  template: solarlog
  usage: pv
  host: 192.168.178.23
  port: 502

- name: battery
  type: custom
  ##usage: battery
  power:
    ##Shelly EM Stromzange 2
    source: http
    uri: http://192.168.178.21/emeter/1
    jq: if .pf==0 then .power else .power / (.pf | fabs) end
  soc:
[... 835 lines cut ...]
Okt 11 13:47:41 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:41 giving priority to battery charging at soc: 61%
Okt 11 13:47:41 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:41 aux power 1: 1W
Okt 11 13:47:41 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:41 aux power: 1W
Okt 11 13:47:41 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:41 site power: -1815W
Okt 11 13:47:50 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:50 ----
Okt 11 13:47:50 raspberrypi evcc[2239]: [lp-1  ] DEBUG 2023/10/11 13:47:50 charge power: 1200W
Okt 11 13:47:50 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:50 pv power: 4633W
Okt 11 13:47:50 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:50 battery soc: 61%
Okt 11 13:47:50 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:50 battery power: -1184W
Okt 11 13:47:50 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:50 grid power: -1920W
Okt 11 13:47:50 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:50 grid powers: [-325 -1464 -132]W
Okt 11 13:47:51 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:51 grid currents: [-1.88 -6.36 -1.69]A
Okt 11 13:47:51 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:51 giving priority to battery charging at soc: 61%
Okt 11 13:47:51 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:51 aux power 1: 1W
Okt 11 13:47:51 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:51 aux power: 1W
Okt 11 13:47:51 raspberrypi evcc[2239]: [site  ] DEBUG 2023/10/11 13:47:51 site power: -1821W
```

### What type of operating system are you running?

Linux

### Version

0.120.3 (7626c45a)

VolkerK62, 2023-10-11:
Aufgefallen ist es mir, weil im PV-Modus nicht mehr richtig geregelt wurde. Dann habe ich per `maxcurrent` nachgeholfen und dabei diese Verhalten festgestellt.

andig, 2023-10-11:
@VolkerK62 eigentlich ziemlich unmöglich (tm). Kannst Du das Log bitte nochmal erzeugen mit GoE auf `trace`? Dann sehen wir, was an den Charger geht. Was zeigt die Go-E App jeweils bei Deinen Änderungen an?

VolkerK62, 2023-10-12:
In der DaheimLaden-App bekommen ich nur die power, nicht die currents angezeigt. Die power stimmt mit evcc überein.

VolkerK62, 2023-10-12:
Wenn du zum Vergleich ein Log mit 0.120.3 Release brauchst (wo der Fehler nciht auftritt). Hab ich erstellt und gespeichert.

andig, 2023-10-12:
Ja bitte- im Log ist erstmal kein Fehler zu erkennen.

VolkerK62, 2023-10-12:
Log mit 0.120.3 Release

18:30:05 NOW mit 20A
18:31:57 set 10A; Reduzierung auf 10A
18:32:58 set 6A; Reduzierung auf 6A
18:34:08 set 16A; Erhöhung auf 16A
18:34:49 set 20A; Erhöhung auf 20A
```
Okt 12 18:30:04 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:04 ----
Okt 12 18:30:04 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:04 modbus: send 00 68 00 00 00 06 ff 03 00 0c 00 02
Okt 12 18:30:04 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:04 modbus: recv 00 68 00 00 00 07 ff 03 04 00 00 00 00
Okt 12 18:30:04 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:30:04 charge power: 0W
Okt 12 18:30:04 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:04 pv power: 0W
Okt 12 18:30:04 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:04 battery soc: 20%
Okt 12 18:30:04 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:04 battery power: -46W
Okt 12 18:30:04 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:04 grid power: 582W
Okt 12 18:30:04 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:04 grid powers: [173 194 216]W
Okt 12 18:30:04 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:04 grid currents: [0.93 1.26 1.17]A
Okt 12 18:30:04 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:04 aux power 1: 0W
Okt 12 18:30:04 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:04 aux power: 0W
Okt 12 18:30:04 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:04 site power: 635W
Okt 12 18:30:05 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:30:05 set charge mode: now
Okt 12 18:30:05 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:05 ----
Okt 12 18:30:05 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:05 modbus: send 00 69 00 00 00 06 ff 03 00 0c 00 02
Okt 12 18:30:05 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:05 modbus: recv 00 69 00 00 00 07 ff 03 04 00 00 00 00
Okt 12 18:30:05 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:30:05 charge power: 0W
Okt 12 18:30:05 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:05 pv power: 0W
Okt 12 18:30:05 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:05 battery soc: 20%
Okt 12 18:30:05 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:05 battery power: -45W
Okt 12 18:30:05 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:05 grid power: 584W
Okt 12 18:30:05 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:05 grid powers: [174 196 215]W
Okt 12 18:30:05 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:05 grid currents: [0.93 1.27 1.16]A
Okt 12 18:30:06 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:06 aux power 1: 0W
Okt 12 18:30:06 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:06 aux power: 0W
Okt 12 18:30:06 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:30:06 site power: 639W
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: send 00 6a 00 00 00 06 ff 03 00 6c 00 06
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: recv 00 6a 00 00 00 0f ff 03 0c 00 00 09 13 00 00 00 00 00 00 00 00
Okt 12 18:30:06 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:30:06 charge voltages: [232 0 0]V
Okt 12 18:30:06 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:30:06 detected connected phases: 1p
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: send 00 6b 00 00 00 06 ff 03 00 05 00 06
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: recv 00 6b 00 00 00 0f ff 03 0c 00 00 00 00 00 00 00 00 00 00 00 00
Okt 12 18:30:06 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:30:06 charge currents: [0 0 0]A
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: send 00 6c 00 00 00 06 ff 03 00 1c 00 02
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: recv 00 6c 00 00 00 07 ff 03 04 00 00 29 4a
Okt 12 18:30:06 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:30:06 charge total import: 1057.000kWh
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: send 00 6d 00 00 00 06 ff 03 00 00 00 01
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: recv 00 6d 00 00 00 05 ff 03 02 00 02
Okt 12 18:30:06 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:30:06 charger status: B
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: send 00 6e 00 00 00 06 ff 03 00 36 00 10
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: recv 00 6e 00 00 00 23 ff 03 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Okt 12 18:30:06 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:30:06 next soc poll remaining time: 4h56m17s
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: send 00 6f 00 00 00 06 ff 03 00 5b 00 01
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: recv 00 6f 00 00 00 05 ff 03 02 00 00
Okt 12 18:30:06 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:30:06 modbus: send 00 70 00 00 00 09 ff 10 00 5b 00 01 02 00 c8
[... cut ...]
Okt 12 18:35:51 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:35:51 vehicle soc limit: 0%
Okt 12 18:35:51 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:35:51 vehicle range: 326km
Okt 12 18:35:51 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:35:51 modbus: send 01 00 00 00 00 06 ff 03 00 5b 00 01
Okt 12 18:35:51 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:35:51 modbus: recv 01 00 00 00 00 05 ff 03 02 00 00
Okt 12 18:35:51 raspberrypi evcc[17317]: [site  ] DEBUG 2023/10/12 18:35:51 ----
```

VolkerK62, 2023-10-12:
Ich habe mir mal die letzten commits angesehen.
Hier gab es bei DaheimLaden eine Änderung
https://github.com/evcc-io/evcc/commit/03da1f129f37648ac8ccf9ac008fbc143f345274

andig, 2023-10-12:
Also. 0120.3:

```
Okt 12 18:31:57 raspberrypi evcc[17317]: [lp-1  ] DEBUG 2023/10/12 18:31:57 set max current: 10
Okt 12 18:32:02 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:32:02 modbus: send 00 9c 00 00 00 06 ff 03 00 57 00 01
Okt 12 18:32:02 raspberrypi evcc[17317]: [daheimladen-mb] TRACE 2023/10/12 18:32:02 modbus: recv 00 9c 00 00 00 05 ff 03 02 00 c8
```

master:

```
Okt 12 17:30:30 raspberrypi evcc[16623]: [lp-1  ] DEBUG 2023/10/12 17:30:30 set max current: 10
Okt 12 17:30:38 raspberrypi evcc[16623]: [site  ] DEBUG 2023/10/12 17:30:38 ----
Okt 12 17:30:38 raspberrypi evcc[16623]: [daheimladen-mb] TRACE 2023/10/12 17:30:38 modbus: send 00 c2 00 00 00 06 ff 03 00 0c 00 02
Okt 12 17:30:38 raspberrypi evcc[16623]: [daheimladen-mb] TRACE 2023/10/12 17:30:38 modbus: recv 00 c2 00 00 00 07 ff 03 04 00 00 10 68-
```

Closed in https://github.com/evcc-io/evcc/commit/e3583399fdd68d2a8a6b963ebe9afb2268d6656c.

closed 2023-10-12 by andig

VolkerK62, 2023-10-13:
top. Funktioniert wieder wie gewohnt. Danke