evcc-io/evcc #10276: SOC of heating system is not updated when not charging

MikeTheTux, 2023-10-11:
### Describe the bug

I've setup a heating system like described in https://docs.evcc.io/docs/guides/faq

EVCC reads and displays the SOC of the heating system during program startup.
EVCC updates the SOC while charging.
Once the target SOC is reached, EVCC is no longer reading the SOC.
SOC remains on the previous target level.
Charging is not initiated again, even if SOC falls under target SOC - workaround: restart of program.

[image]

Could this be an implementation behavior from Electronic cars? When a certain target SOC is reached, it is not decreasing until the car is dis-connected/re-connected.

### Steps to reproduce

1. PV charge heating system until target SOC is reached
2. Reduce current SOC (e.g. via MQTT)
3. SOC shown/used by EVCC is not updated. Charging is not initiated again, even if SOC falls under target SOC.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: nas
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

# log settings
log: debug
levels:
  cache: info
  audi: info
  site: info
  lp-1: info
  script: info
  push: info
  lp-2: debug

interval: 10s # control cycle interval

chargers:
  - name: keba
    type: template
    template: keba-modbus
    host: 192.168.69.108
#    serial: xxx
    id: 255
    port: 502
  - name: heatpump_switch
    type: template
    template: shelly
    host: 192.168.69.152
    standbypower: -1
    integrateddevice: true
    heating: true
    icon: waterheater

loadpoints:
- title: Wallbox
  charger: keba
  mode: off                 # set default charge mode, use "off" to disable by default if charger is publicly available, now, off, minpv, pv
[... 432 lines cut ...]
[lp-2  ] DEBUG 2023/10/11 20:39:55 targetSoc reached: 56.7% > 40%
[lp-2  ] DEBUG 2023/10/11 20:40:05 charge power: 1506W
[lp-2  ] DEBUG 2023/10/11 20:40:15 charge power: 1553W
[lp-2  ] DEBUG 2023/10/11 20:40:15 charger status: B
[lp-2  ] DEBUG 2023/10/11 20:40:15 targetSoc reached: 56.7% > 40%
[lp-2  ] DEBUG 2023/10/11 20:40:25 charge power: 1352W
[lp-2  ] DEBUG 2023/10/11 20:40:35 charge power: 1542W
[lp-2  ] DEBUG 2023/10/11 20:40:35 charger status: B
[lp-2  ] DEBUG 2023/10/11 20:40:35 targetSoc reached: 56.7% > 40%
[lp-2  ] DEBUG 2023/10/11 20:40:45 charge power: 1435W
[lp-2  ] DEBUG 2023/10/11 20:40:55 charge power: 1407W
[lp-2  ] DEBUG 2023/10/11 20:40:55 charger status: B
[lp-2  ] DEBUG 2023/10/11 20:40:55 targetSoc reached: 56.7% > 40%
[lp-2  ] DEBUG 2023/10/11 20:41:05 charge power: 1537W

SOC changed from 56.7% to 56.5% which is not shown.
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.120.3

StefanSchoof, 2023-10-11:
The default `poll.mode` is `charging`. You need it to set on `always`, see https://docs.evcc.io/docs/reference/configuration/loadpoints/#pollmode

MikeTheTux, 2023-10-11:
@StefanSchoof thanks, that's the trick! In parallel to our post I found the same lines in the config shared via #10269.

I extended #10220 with that info in order to improve the sample configuration / documentation.

andig, 2023-10-12:
I would like to reopen this. I would argue that specifying a poll mode for an integrated device (i.e. one that does not have an attached vehicle with separate power supply) is unexpected.

I would suggest that integrated devices ignore the poll mode and allow polling every time (if possible even disregarding the cache setting).

Any objections?

/cc @VolkerK62

andig, 2023-10-12:
Looking at the config this feels kinda bad: you have an integrated device (i.e. one that does NOT have a vehicle attached) but still have a vehicle associated at the loadpoint to have a device to provide soc.

If- instead- the charger (Shelly) had the ability to provide soc, we would already and unconditionally read and publish that.

I see two options:
1. integrated devices ignore the poll mode and allow polling every time (as proposed above) OR
2. implement the ability to decorate existing templates with additional capability (probably nothing short-term).

/cc @premultiply @naltatis

VolkerK62, 2023-10-12:
>Any objections? 

I agree. But, depending on the type of query, can this lead to any problems?

andig, 2023-10-12:
Under which scenario would an integrated device- that is always queried anyways- have a problem with one more query?

closed 2023-10-12 by andig