evcc-io/evcc #10287: Cheapest-Funktion mit Batterie: zwingend Netzladen

WisManue, 2023-10-12:
Ich nutze sehr gerne die Funktion "cheapest" also entweder co2 basiert oder auf Basis von tibber für günstiges Laden. 
Da mein Setting eine PV-Anlage mit Batteriespeicher beinhaltet, habe ich jetzt beim Laden in der "cheapest-Zeit" das Phänomen, dass 3,7KW (Die Ausgangsleistung von SMA) immer von der Batterie geladen werden. 
Da die Batterie durch den hohen Konsum dann nicht mehr die ganze Nacht durchhält, nutzt der Haushalt dann den "teuren/dreckigen" Strom. 

Ich würde mir als Feature-Request wünschen, dass ich über evcc einstellen kann (Oberfläche oder in der yaml-Datei), dass ein Laden nur aus dem Netz erfolgt und der Batteriespeicher dafür nicht genutzt wird. 

Ich weiß, dass hierfür ein Eingriff in die eigene Ladesteuerung nötig ist. Da zumindest SMA nach meinem Wissensstand hier über das Sunny-Portal eine Schnittstelle bietet, wäre das jedenfalls eine Möglichkeit.

PS: Dies ist mein erster Request auf Git überhaupt, da ich evcc super genial finde und auch meine Gedanke hier beitragen möchte, auch wenn ich selbst nicht programmieren kann. 😀 ich hoffe die Beschreibung passt so, ansonsten gerne ein Feedback geben ;)

StefanSchoof, 2023-10-12:
Schaue mal unter https://github.com/evcc-io/evcc/discussions/10288#discussioncomment-7266878

andig, 2023-10-12:
Äaktuell nicht möglich, da wir keine Apis dafür haben. Gerne weiter in verlinkter Diskussion

closed 2023-10-12 by andig