evcc-io/evcc #10294: Ford Kuga loading not started after 5min. off

gereic, 2023-10-13:
### Describe the bug

Vehicle: Ford Kuga
Wallbox: Go-E-Charger V3

### Steps to reproduce

1. Vehicle unplugged and locked
2. EVCC set to off
3. plugged in cable to vehicle
4. wait 8min. 
5. set evcc to charge immediately
6. Go-E goes to green, but loading the vehicle is not starting.
7. When Go-E -charger is unplugge at vehicle or power, and plugged in again, charging is starting.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: e67e6b4569d06c59a533a861c2e68b8aeac6bf9d5381941d947d84556013e555

interval: 10s # control cycle interval

sponsortoken:

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.108  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.108  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.108  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.109  
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
vehicles:
- name: Ford Kuga
  type: template
  template: ford
  title: # Wird in der Benutzeroberfläche angezeigt (Optional)
  user: user@email.cpm
  password: userpw
  vin: vim # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (Optional)
  capacity: 14 # Akkukapazität in kWh (Optional) 
  phases: 1

site:
  title: kunde
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
no log
```

### What type of operating system are you running?

Linux

### Version

0.120.3

andig, 2023-10-13:
Missing log

closed 2023-10-13 by andig

gereic, 2023-10-13:
Hi, for all who have the same problem.
I found the problem already.
The problem was, that you have to set a option on the go-e-charger,  "simulate unplug and plug again" must be set on for the Ford Kuga.
This setting is in vehicle-settings directly on the go-e app avaiable.
So it has nothing todo with the evcc-software.

andig, 2023-10-13:
Das könnte einen Hinweis im GoE Template wert sein. 

/cc @VolkerK62

VolkerK62, 2023-10-14:
@gereic  see https://github.com/evcc-io/evcc/issues/10301
 I do not have go-e and the app, would you please add a screenshot off the vehicle-setting page to this issue