evcc-io/evcc #10303: Alfen Eve charger does not switch back to 3P

ivonoppen, 2023-10-14:
### Describe the bug

When plugging in my EV to the Alfen Eve, 3-phase charging works if it can immediately charge. When enabling the planner, (I only charge on dynamic tariff) EVCC puts the charger back in 1-phase mode. When starting the charging later on, the charger is not (always) switched back in 3-phase mode anymore, even though the logs indicate that 3p is wanted. 

Specifically, every 30 seconds you can see in the log the following lines:
```
evcc  | [lp-1  ] DEBUG 2023/10/14 07:35:57 !! scalePhasesIfAvailable: 3p -> 3p
evcc  | [lp-1  ] DEBUG 2023/10/14 07:35:57 !! scalePhases: GetPhases 3p <> 3p wanted
```

In the example log attached:
- Around 01.00, everything goes well (switches to 3P)
- Around 04.00, it switches to 3P, but fails to do so
- From there, the charger remains stuck in 1P, even though EVCC can still start and stop the charging and adjusts the charge planning accordingly

### Steps to reproduce

1. Start charging
2. Set planner based on tariff

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ***

interval: 30s # control cycle interval

meters:
- type: template
  template: homewizard
  usage: grid
  host: 192.168.2.22
  name: homewizard

vehicles:
- type: template
  template: ford
  title: Car
  user: ***
  password: ***
  vin: ***
  capacity: 98 #kWh
  name: car
  mode: pv
  targetSoc: 90

loadpoints:
  - title: Garage # display name for UI
    charger: alfen # Wallbox Referenz
    mode: pv # charge mode (off, now, minpv, pv)

tariffs:
  grid:
    type: tibber
    token: ***
  feedin:
    type: tibber
    token: ***

[... 143 lines cut ...]
evcc  | [site  ] DEBUG 2023/10/14 04:01:56 ----
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 charge power: 3594W
evcc  | [site  ] DEBUG 2023/10/14 04:01:57 grid power: 3846W
evcc  | [site  ] DEBUG 2023/10/14 04:01:57 pv power: 0W
evcc  | [site  ] DEBUG 2023/10/14 04:01:57 site power: 3846W
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 charge currents: [15.9 0 0]A
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 detected active phases: 1p
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 charge total import: 2477.813kWh
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 charger status: C
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 soc estimated: 47.66% (vehicle: 47.50%)
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 vehicle soc: 48%
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 vehicle range: 280km
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 plan: charge 12h31m45s starting at 2023-10-14 04:01:58 +0200 CEST until 2023-10-14 23:30:00 +0200 CEST (power: 3680W, avg cost: 0.179)
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 !! scalePhasesIfAvailable: 3p -> 3p
evcc  | [lp-1  ] DEBUG 2023/10/14 04:01:57 !! scalePhases: GetPhases 3p <> 3p wanted
evcc  | [site  ] DEBUG 2023/10/14 04:02:26 ----
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.121.0

andig, 2023-10-14:
Please update

ivonoppen, 2023-10-14:
> Please update

I seem to be running the latest version according to the releases page? Not sure how to interpret your message?

andig, 2023-10-14:
DEBUG 2023/10/14 04:00:27 switched phases: 3p
    DEBUG 2023/10/14 04:01:27 charge currents: [15.8 0 0]A

Seems with has been done but car continues on 1p. No idea how that is possible...

ivonoppen, 2023-10-14:
A (for me) potential temporary solution would be to have an option to disable the phase switching in the configuration and lock it to 3 phases. As I have no PV currently, 1P also has no real use for me. Just 3P with load balancing is enough.

I couldn't directly find such option in the documentation. Does that option exist to fixate to 3P and thus only use the planner / load balancing features?

andig, 2023-10-14:
phases: 3

is enough. This does not explain though why your charger does not follow the command.

closed 2023-10-14 by andig

ivonoppen, 2023-10-14:
I'll do some digging tomorrow in the logs of the charger itself to see if it is indeed the charger or the car. Hopefully it contains enough information to debug. Will keep you posted.

DieterC1984, 2024-03-04:
Any news on this perhaps? I just stumbled on this thread and having the same problem. I want to be able to charge on 1phase during sunny days, and from sunset switch back to 3 phase automatically. But the Alfen is refusing it for some reason...
I can't figure out what is causing this...