evcc-io/evcc #10318: evcc docker v.0.121.0 doesn't work after update - ui frozen

fpeghub, 2023-10-14:
### Describe the bug

Looks like there is not detection of current values anymore. All numbers are frozen, no change. 
Connected car wasn't detected as well.
I'm using evcc since some month ago with no problems after container updates. No config change from my site.

### Steps to reproduce

1. Restart via ui help menu
2. Container restart
3.
...

### Configuration details

```yaml
/app # evcc --log debug
[main  ] INFO 2023/10/14 17:36:44 evcc 0.121.0
[main  ] INFO 2023/10/14 17:36:44 missing config file - switching into demo mode
[main  ] INFO 2023/10/14 17:36:44 starting ui and api at :7070
[db    ] INFO 2023/10/14 17:36:44 using sqlite database: /root/.evcc/evcc.db
[site  ] WARN 2023/10/14 17:36:44 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] WARN 2023/10/14 17:36:44 vehicle 'blauer e-Golf' does not support automatic detection
[site  ] WARN 2023/10/14 17:36:44 vehicle 'grüner Honda e' does not support automatic detection
[site  ] WARN 2023/10/14 17:36:44 vehicle 'schwarzes VanMoof' does not support automatic detection
[site  ] WARN 2023/10/14 17:36:44 vehicle 'Wärmepumpe' does not support automatic detection
[site  ] INFO 2023/10/14 17:36:44 site config:
[site  ] INFO 2023/10/14 17:36:44   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/14 17:36:44     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/14 17:36:44     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/14 17:36:44     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/10/14 17:36:44   vehicles:
[site  ] INFO 2023/10/14 17:36:44     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2023/10/14 17:36:44     vehicle 2: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
[site  ] INFO 2023/10/14 17:36:44     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2023/10/14 17:36:44     vehicle 4: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2023/10/14 17:36:44     vehicle 5: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/10/14 17:36:44 loadpoint 1:
[lp-1  ] INFO 2023/10/14 17:36:44   mode:        pv
[lp-1  ] INFO 2023/10/14 17:36:44   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/14 17:36:44   meters:      charge ✓
[lp-1  ] INFO 2023/10/14 17:36:44     charge:    power ✓ energy ✗ currents ✗
[lp-2  ] INFO 2023/10/14 17:36:44 loadpoint 2:
[lp-2  ] INFO 2023/10/14 17:36:44   mode:        off
[lp-2  ] INFO 2023/10/14 17:36:44   charger:     power ✗ energy ✗ currents ✗ phases ✓ wakeup ✗
[lp-2  ] INFO 2023/10/14 17:36:44   meters:      charge ✓
[lp-2  ] INFO 2023/10/14 17:36:44     charge:    power ✓ energy ✗ currents ✗
[lp-1  ] DEBUG 2023/10/14 17:36:44 phase timer inactive
[lp-1  ] DEBUG 2023/10/14 17:36:44 pv timer inactive
[lp-1  ] DEBUG 2023/10/14 17:36:44 guard timer inactive
[lp-1  ] INFO 2023/10/14 17:36:44 vehicle updated: unknown -> blauer e-Golf
[lp-1  ] DEBUG 2023/10/14 17:36:44 set target soc: 90
[js-shared] DEBUG 2023/10/14 17:36:44 state: {"batterySoc":55,"batterypower":-750,"gridpower":-1000,"loadpoints":[{"chargepower":0,"enabled":true,"maxcurrent":6,"phases":1,"vehicleSoc":62},{"chargepower":0,"enabled":false,"maxcurrent":0,"phases":3,"vehicleSoc":22}],"pvpower":5000,"residualpower":500}
[lp-1  ] DEBUG 2023/10/14 17:36:44 max charge current: 6A
[lp-2  ] DEBUG 2023/10/14 17:36:44 phase timer inactive
[lp-2  ] DEBUG 2023/10/14 17:36:44 pv timer inactive
[lp-2  ] DEBUG 2023/10/14 17:36:44 guard timer inactive
[lp-2  ] INFO 2023/10/14 17:36:44 vehicle updated: unknown -> weißes Model 3
[lp-2  ] DEBUG 2023/10/14 17:36:44 set target soc: 75
[main  ] FATAL 2023/10/14 17:36:44 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
[site  ] WARN 2023/10/14 17:36:44 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
```

### Log details

```text
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: 192.168.155.18 # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 983af8346171763d3616c809d127e6cd8a1d216dea12736a7af4cabccc5f8922

# interval: 45s # control cycle interval

sponsortoken: trallala

meters:
- type: template
  template: solarwatt 
  usage: grid  
  host: 192.168.155.11  
  name: grid1
- type: template
  template: solarwatt 
  usage: pv  
  host: 192.168.155.11  
  name: pv2
- type: template
  template: solarwatt 
  usage: battery  
  host: 192.168.155.11  
  name: battery3

chargers:
- type: template
  template: pulsarplus 
  name: wallbox4
  stationid: PulsarPlus  
  connector: 1  
  connecttimeout: 5m  
  timeout: 2m  

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  vehicle: my_car
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

#vehicles:
#  - name: my_car
#    type: mercedes
#    title: Mercedes GLC
#    phases: 2
#    capacity: 11
#    clientId: trallala
#    clientSecret: trallala
#    vin: trallala

vehicles:
  - name: my_car
    type: template
    template: offline
    title: GLC
    capacity: 11

site:
  title: Frank-Home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
# residualPower: 100
```

### What type of operating system are you running?

Docker container

### Version

0.121.0

VolkerK62, 2023-10-14:
`[main  ] INFO 2023/10/14 17:36:44 missing config file - switching into demo mode`

closed 2023-10-14 by andig

fpeghub, 2023-10-14:
evcc uses the correct config file, see startup log:

[main  ] INFO 2023/10/14 18:04:45 evcc 0.121.0
[main  ] INFO 2023/10/14 18:04:45 using config file: /evcccfg/evcc.yaml
[main  ] INFO 2023/10/14 18:04:45 starting ui and api at :7070
[db    ] INFO 2023/10/14 18:04:46 using sqlite database: /root/.evcc/evcc.db
[PulsarPlus-1] DEBUG 2023/10/14 18:04:46 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2023/10/14 18:05:17 charge point connected: PulsarPlus
[site  ] WARN 2023/10/14 18:05:21 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2023/10/14 18:05:21 site config:
[site  ] INFO 2023/10/14 18:05:21   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/14 18:05:21     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/14 18:05:21     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/14 18:05:21     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/10/14 18:05:21   vehicles:
[site  ] INFO 2023/10/14 18:05:21     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/10/14 18:05:21 loadpoint 1:
[lp-1  ] INFO 2023/10/14 18:05:21   mode:        pv
[lp-1  ] INFO 2023/10/14 18:05:21   charger:     power ✗ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/14 18:05:21   meters:      charge ✗
[lp-1  ] DEBUG 2023/10/14 18:05:21 phase timer inactive
[lp-1  ] DEBUG 2023/10/14 18:05:21 pv timer inactive
[lp-1  ] DEBUG 2023/10/14 18:05:21 guard timer inactive
[lp-1  ] INFO 2023/10/14 18:05:21 vehicle updated: unknown -> GLC
[site  ] WARN 2023/10/14 18:05:21 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/10/14 18:05:21 ----
[lp-1  ] DEBUG 2023/10/14 18:05:21 charge power: 0W
[site  ] DEBUG 2023/10/14 18:05:22 pv power: 0W
[site  ] DEBUG 2023/10/14 18:05:22 battery soc: 95%
[site  ] DEBUG 2023/10/14 18:05:22 battery power: 557W
[site  ] DEBUG 2023/10/14 18:05:22 grid power: 0W
[site  ] DEBUG 2023/10/14 18:05:23 site power: 557W

andig, 2023-10-14:
This is not what your issue shows. Check your config, clarify issue, open clean report.

fpeghub, 2023-10-14:
<div></div><div>Closed #10318 as completed.
<br/>
<br/>-- 
<br/>
<br/>https://github.com/evcc-io/evcc/issues/10318#event-10654612476
<br/>