evcc-io/evcc #10325: Add home battery charging capability from grid when using variable tariff (e.g. Tibber)

kLOsk, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
I guess this is old news and has probably been discussed before, but with winter coming it would be great to have the option in evcc to charge the household battery from grid, when tariff prices are low.

**Describe the solution you'd like**
I'm aware this needs thought but for now I believe it would be sufficient to simply charge the household battery when the tariff is the lowest for the day, similar to how car battery target charging works. In the future it would of course be great to also add PV production forecasts, assumptions about consumption, etc.

ThinkEV, 2023-10-15:
How shall evcc enable/disable charging the byttery from grid?

Maybe this could be combined with disabling the battery while charging above a certain threshhold.
See https://github.com/evcc-io/evcc/discussions/5826 and https://github.com/evcc-io/evcc/discussions/9090
While charging at night at a very low tariff, I don't want to drain my battery. I rather like to use it afterwards, when the tariff prices are high again.

kLOsk, 2023-10-15:
I'm using a Victron inverter along with Pylontech batteries and it can be easily turned on and off via REST API and MQTT. I think most inverters can be controlled this way?

andig, 2023-10-16:
This has been discussed multiple times. We do not have enough critical mass of battery apis to support this case. See related discussion. Closing as duplicate for time being.

closed 2023-10-16 by andig