evcc-io/evcc #10330: Refactor session settings apis

andig, 2023-10-15:
We're planning to refactor handling of the various settings and their associated APIs. Here is the proposal:

| Value      | Loadpoint | Charger | Vehicle | Session | Comment                           | Api               |
| ---------- | --------- | ------- | ------- | ------- | --------------------------------- | ----------------- |
| soc        | -         | x       | x       | -       | charger has precedence            |                   |
| capacity   | -         | -       | x       | -       |                                   |                   |
| minsoc     | -         | -       | x       | -       |                                   | vehicle           |
| limitsoc   | -         | -       | x       | x       |                                   | vehicle           |
| plansoc    | -         | -       | x       | -       | requires capacity, maybe deducted | vehicle           |
| planenergy | -         | -       | -       | x       | not available with soc            | loadpoint         |
| mincurrent | x         | -       | x       | -       | vehicle has precendence           | loadpoint+vehicle |
| maxcurrent | x         | -       | x       | -       | vehicle has precendence           | loadpoint+vehicle |

Open points:

- [ ] define charger handling for EEbus

naltatis, 2023-10-15:
`limitenergy` is missing (guest, offline vehicles, no plan)

MarkusGH, 2023-10-15:
Question regarding mincurrent/maxcurrent.
In the table is a note about precendence.
Does't it make more sense to use the Min() respective Max() of the two values?

premultiply, 2023-10-15:
For many chargers it would be possible to read their internal current limit setup.

Would that be an improvement or simplification for standard setup scenarios?

andig, 2023-10-16:
> Would that be an improvement or simplification for standard setup scenarios?

OT für diesen PR.

andig, 2023-10-16:
> Does't it make more sense to use the Min() respective Max() of the two values?

These are the use cases I'm aware of:
- eebus wants to use _lower_ currents than defined 6A PWM min
- Zoe wants higher 1p current than defined 6A PWM min
- user wants higher 1p max current than defined 16A PWM max

Min/max won't be able to cover at least the first case.

DerAndereAndi, 2023-10-24:
Some thoughts:

- It is not EEBUS that wants to use lower currents than defined in PWM, it is the charger using ISO15118 with a compatible EV that wants that. EEBUS is just the only charger implementation that supports this in here right now. Others will follow.
- The min current is NOT defined by a loadpoint, but always by the communication standard used between the charger and the EV. So the charger knows the minCurrent. If the charger supports only PWM, then for all these it is always 6A.
- An EV might report lower minCurrents than it actually is able to work with, e.g. MEB series reports 1.5A but below 3A it only actively used the 3rd phase. So it is helpful to define a protocol specific higher (!!) minCurrent on the vehicle.
- The max current is first of all a physical electrical limitation that usually is setup in the charger using a DIP switch, so the charger knows it may never exceed this limit. So that can be argued to be defined on the loadpoint as being a electrical limit
- Neither the charger nor the EV may ever use a higher max Current. And further an EV may never use a higher limit than what the charger supports (depending on the communication protocol used).
- Therefor I would suggest the following rule:
  - the loadpoints minCurrent is effectively: 0
  - `minCurrent`: Use the max of charger and vehicle `minCurrent`
  - `maxCurrent`: Use the min of loadpoint, charger and vehicle maxCurrent
- ISO15118-20 will support providing the `soc` and `minSoC` via the charger. Supporting this already might be good idea
- ISO15118-2 supports providing plannedEnergy and a datetime for this via the charger. Supporting might be a good idea. It could be already leveraged by EEBUS charger and compatible EVs (Porsche, MEB based EVs, Mercedes, some BMW, ...)
- ISO15118-2 supports providing the energy to reach the car defined minSoC. Supporting this already might be good idea

andig, 2023-10-30:
Most points agreed.

> The max current is first of all a physical electrical limitation that usually is setup in the charger using a DIP switch, so the charger knows it may never exceed this limit. So that can be argued to be defined on the loadpoint as being a electrical limit

To be frank and clear: evcc is not designed to ensure electrical limits. This MUST be done in hardware.

> Neither the charger nor the EV may ever use a higher max Current. And further an EV may never use a higher limit than what the charger supports (depending on the communication protocol used).

Ackknowledged- but a hardware task, too.

> `minCurrent`: Use the max of charger and vehicle minCurrent

Good idea. However, this would require us to add `minCurrent` (or a communication standard as proxy for going so) to all chargers. As we would *still* need `maxCurrent` on `loadpoint` level that is also not entirely obvious.

> `maxCurrent`  Use the min of loadpoint, charger and vehicle maxCurrent

This does not cover the Zoe 1p3p use case mentioned on top unfortunately

DerAndereAndi, 2023-10-31:
> To be frank and clear: evcc is not designed to ensure electrical limits. This MUST be done in hardware.

Did not say that evcc should enforce that. But evcc should know what  the limits are setup to and enforced by the wallbox.

> Ackknowledged- but a hardware task, too.

The hardware will enforce it, sure. So what is the benefit of allowing limits outside of this?

> This does not cover the Zoe 1p3p use case mentioned on top unfortunately

The Zoe doesn't have a problem with `maxCurrent`, but with `minCurrent`. So this is not related.

DerAndereAndi, 2023-11-01:
One addition:

The maxCurrent at the loadpoint can be seen as the electrical connection, but also a possibility to lower that limit further. E.g. for the case of not wanting to use the maximum possible that the EVSE is connected to, because of general power stability reasons or if a CEE32 cable is used with a CEE16 plug via an adapter or ...

Example:
- EVSE is connected via 32A
- loadpoint is set to maxCurrent 24A -> range is 0A - 24A
- charger reports minCurrent 2A and maxCurrent 32A -> range is 2A - 24A
- vehicle is setup with maxCurrent of 20A -> range is 2A - 20A

DerAndereAndi, 2023-11-01:
One more side addition, not directly related to this, but conceptually: loadpoint phases should have the same meaning as for `maxCurrent` (when going with my suggestion), and this state how it is electrically connected. The option of phase switching should be used (or even is possible), should then but moved into the charger

premultiply, 2023-12-10:
We can extend the API  the query the set hardware current limits from most chargers.
So limits could be calucated by `min/max(loadpoint(user), charger(hardware), vehicle)`.

andig, 2023-12-10:
Könnte man, ist aktuell aber nicht so ;) Es sollte zu https://github.com/evcc-io/evcc/issues/10330#issuecomment-1765203043 passen.

premultiply, 2023-12-10:
Denke schon, denn im Fall von ISO geht der minCurrent des chargers dann auf 2 runter.

andig, 2023-12-10:
Dann erklär bitte nochmal, wie du min/max auf welchen Strom anwenden willst? min auf min und max auf max?

premultiply, 2023-12-10:
Andersrum: Die max()-Funktion auf die minCurrents und die min()-Funktion auf die maxCurrents.

premultiply, 2023-12-11:
1 & 2 sind damit gelöst.
Für 3 braucht man nach 1p und 3p unterschiedene Settings an Loadpoint und Vehicle.

VolkerK62, 2023-12-12:
Da schließt sich dann der Kreis ;-)
https://github.com/evcc-io/evcc/issues/3528#issuecomment-1763382387