evcc-io/evcc #10338: Flobz psa controller: add vehicle wake up

deadrabbit87, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
Sometimes the car is not waking up for charging. 

**Describe the solution you'd like**
integrate in the flobz template the WakeUp-function. This can be done with 
`http://localhost:5000/charge_now/YOURVIN/1`
[documentation](https://github.com/flobz/psa_car_controller)

**Describe alternatives you've considered**
-

**Additional context**
-

andig, 2023-10-15:
PR welcome!

mdkeil, 2023-10-20:
`http://localhost:5000/wakeup/YOURVIN` is the right call for wakeup.

mdkeil, 2023-10-20:
#10411

deadrabbit87, 2023-10-20:
Danke.

Für was ist dann `charge_now`?

closed 2023-10-21 by andig

deadrabbit87, 2023-10-24:
> `http://<IP>:<PORT>/wakeup/YOURVIN` is the right call for wakeup.

In meinem Fall hat das leider nichts gebracht. Fahrzeug wacht trotzdem sporadisch nicht auf.

Setze ich in der webui den Schalter charge, beginnt er zu laden.

Daher nochmal: Warum wäre mein Vorschlag mit `charge_now` falsch?

mdkeil, 2023-10-24:
Von welchem Fahrzeug reden wir denn? Mein e208 schläft z.B. nie soweit ein, dass er keine Ladung startet. Ich kann es daher auch nicht vernünftig testen.. wenn das Fahrzeug an den PSA Controller keine Daten mehr sendet (weil es vermutlich in Standby ist) reicht bei mir immer das aktuell implementierte Kommando (/wakeup) und ich erhalte wieder Aktualisierungen.. funktioniert dies nicht, gibt es meistens wie so häufig Probleme beim PSA Server oder der 2G Empfang ist zu schlecht..

deadrabbit87, 2023-10-24:
Es ist auch ein e208.

Wenn es der Empfang wäre, dann würde es doch auch nicht starten wenn ich den Schalter im psa Controller setze?

mdkeil, 2023-10-24:
Kann es vielleicht sein, dass bei dir zeitversetztes Laden aktiv ist? Was steht denn in der PSA-App unter Laden? Dort muss "Sofortiges Aufladen" aktiviert sein.

deadrabbit87, 2023-10-24:
Nein, ist nicht der Fall: [image]

mdkeil, 2023-10-24:
Ich habe auch mal eben mein e208 via "charge_now" aufgeweckt.. der e208 scheint wohl unterschiedliche "Schlafmodi" zu haben.. Ich fahre meinen seit 10 Monate (> 20.000 km) und ich habe noch nie die Zustand gehabt, dass sich das Auto soweit runtergefahren hat, dass es nicht geladen hat, sobald der Loadpoint via evcc freigegeben wurde.. via "charge_now" ist das Fahrzeug auf jeden Fall derart wach, dass ich auch einen aktuellen Zeitstempel bekomme.. das "normale" wakeup erzeugt dieses Verhalten nicht.. wie gesagt, das hat für mich zum Laden und zur Aktivierung der Klimatisierung immer ausgereicht.

Es ist natürlich einfach umgesetzt den wakeup-code gegen "charge_now" zu ersetzen, ob das aber für die Masse notwendig ist, dass bezweifle ich. Darüber hinaus kann ich eventuelle negative Seiteneffekte nicht ausschließen. Was ich mir vorstellen könnte einen z.B. zusätzlichen Parameter "wakeup_extended" einzufügen, der dann den "charge_now" code als wakeup nutzt, wenn auf true gesetzt.

deadrabbit87, 2023-10-24:
Ich weiß auch nicht, warum es bei manchen Fahrzeugen problemlos klappt und bei anderen wieder nicht.

Ich kenne zwei weitere Fahrzeuge im Bekanntenkreis: Einer ist komplett problemlos, der andere stand deswegen schon Wochen in der Werkstatt.

Bei meinem ist es nicht so extrem, aber von Zeit zu Zeit funktioniert es eben nicht.

Was für Nebeneffekte meinst du? Entladung der 12-Volt-Batterie?

mdkeil, 2023-10-24:
Genau die Nebeneffekte kann ich nicht überblicken, wenn es denn welche gibt. Darüber hinaus fehlt immer noch die Rückmeldung wann und wie oft wakeup durch evcc überhaupt getriggert wird. Das konnte ich im Code auf die schnelle nicht finden.

deadrabbit87, 2023-10-24:
Ich hätte da auch schon versucht durchzublicken. Aber irgendwie fehlt mir dazu schon das wissen, wie evcc da aufgebaut ist. 

Ich verstehe auch nicht, dass es bei manchen Fahrzeugen implementiert ist und bei anderen wieder nicht. 

So ist es echt nervig, vor allem wenn man mit Tibber Zielladen will und dann am nächsten Morgen das Auto nicht geladen ist.

Kannst du sagen, welches Baujahr deiner ist und welchen OBC er hat? In meinem Fall wurde der schon erneuert. Siehe Foto vom Typenschild[image]

mdkeil, 2023-10-24:
Gebaut in Q4 2022 und hat von Hause aus noch den 11kW Charger.

mdkeil, 2023-10-25:
PR with alternative "charge_now" wakeup-code; der neue Parameter wakeup_alt muss dafür auf true gesetzt werden. 
#10481

mdkeil, 2023-10-25:
PR #10483 habe ich mal erstellt; sollte er ins nightly gemerged werden, könntest Du es mal testen. es muss dann nur der zusätzliche Parameter "wakeup_alt: true" mit in die evcc.yaml eingefügt werden, wo das Fahrzeug angelegt wird.

deadrabbit87, 2023-10-25:
Vielen Dank, hab es schon gesehen und werde in testen wenn es im nightly ist.

mdkeil, 2023-10-26:
@deadrabbit87 erhöhe mal das Loglevel auf debug (müsste reichen), um zu sehen, wann genau wakeup getriggert wird. Dies dann bitte hier mal posten..

deadrabbit87, 2023-10-26:
Ich teste das mal und melde mich dann im PR