evcc-io/evcc #10339: OCPP GetConfiguration läuft auf Timeout - wie debuggen?

tvand, 2023-10-15:
### Discussed in https://github.com/evcc-io/evcc/discussions/10184

<div type='discussions-op-text'>

<sup>Originally posted by **tvand** October  4, 2023</sup>
Ich hatte mein ursprüngliches Problem (https://github.com/evcc-io/evcc/discussions/9678) vertagt, weil ich es temporär mit `getconfiguration: false` umgehen konnte. Trotzdem würde ich gern wissen, warum das Lesen der Konfiguration nicht funktioniert.

Im Log steht 
```
[ocpp  ] TRACE 2023/09/02 12:49:19 sent JSON message to Wall-E: [2,"4010434209","GetConfiguration",{}]
[ocpp  ] TRACE 2023/09/02 12:49:19 received JSON message from Wall-E: [3,"4010434209",{"configurationKey":[{"key":"AuthorizeRemoteTxRequests","readonly":"true","value":"false"},{"key":"ChargeProfileMaxStackLevel","readonly":"true","value":5},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":"true","value":"W"},{"key":"ChargingScheduleMaxPeriods","readonly":"true","value":20},{"key":"ClockAlignedDataInterval","readonly":"false","value":900},{"key":"ConnectionTimeOut","readonly":"false","value":300},{"key":"ConnectorPhaseRotation","readonly":"false","value":"unknown"},{"key":"GetConfigurationMaxKeys","readonly":"true","value":10},{"key":"HeartbeatInterval","readonly":"false","value":60},{"key":"LocalAuthListEnabled","readonly":"false","value":"false"},{"key":"LocalAuthListMaxLength","readonly":"true","value":20},{"key":"LocalAuthorizeOffline","readonly":"false","value":"true"},{"key":"LocalPreAuthorize","readonly":"false","value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":"true","value":15},{"key":"MeterValueSampleInterval","readonly":"false","value":300},{"key":"MeterValuesAlignedData","readonly":"false","value":"Energy.Active.Import.Register,Power.Active.Import"},{"key":"MeterValuesSampledData","readonly":"false","value":"Energy.Active.Import.Register,Power.Active.Import"},{"key":"NumberOfConnectors","readonly":"true","value":1},{"key":"NumberOfResetRetries","readonly":"false","value":0},{"key":"SendLocalListMaxLength","readonly":"true","value":20},{"key":"StopTransactionOnEVSideDisconnect","readonly":"false","value":"true"},{"key":"StopTransactionOnInvalidId","readonly":"false","value":"false"},{"key":"StopTxnAlignedData","readonly":"false","value":"Energy.Active.Import.Register,Power.Active.Import"},{"key":"StopTxnSampledData","readonly":"false","value":"Energy.Active.Import.Register,Power.Active.Import"},{"key":"SupportedFeatureProfiles","readonly":"true","value":"Core,LocalAuthListManagement,SmartCharging"},{"key":"TransactionMessageAttempts","readonly":"false","value":3},{"key":"TransactionMessageRetryInterval","readonly":"false","value":180},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":"false","value":"false"},{"key":"ckchargepowermax","readonly":"true","value":11000}]}]
[ocpp  ] TRACE 2023/09/02 12:49:20 received JSON message from Wall-E: [2,"26","BootNotification",{"chargePointModel":"HomeChargePrototype","chargePointSerialNumber":"000000000727","chargePointVendor":"HomeCharge","firmwareVersion":"01.08"}]
[ocpp  ] TRACE 2023/09/02 12:49:20 sent JSON message to Wall-E: [3,"26",{"currentTime":"2023-09-02T10:49:20Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2023/09/02 12:49:20 received JSON message from Wall-E: [2,"27","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available"}]
[ocpp  ] TRACE 2023/09/02 12:49:20 sent JSON message to Wall-E: [3,"27",{}]
[main  ] FATAL 2023/09/02 12:50:19 cannot create charger 'walle': cannot create charger 'template': cannot create charger 'ocpp': ocpp message (4010434209): GenericError - Request timed out
```

Das JSON der Response wie oben geloggt sieht für mich gut aus. Ich schließe aus den fehlenden Log-Ausgaben der Callback-Funktion in ocpp.go
```
c.log.TRACE.Printf("%s (%s): %s", opt.Key, rw[opt.Readonly], *opt.Value)
``` 
dass der Aufruf `ocpp.Instance().GetConfiguration(...)` schon aus der OCPP-Bibliothek mit einem Fehler zurückkommt. 

Beim tieferen Eingraben kriege ich einen Fehler im JSON Parser (ocppj.go:428):
```
error(*encoding/json.UnmarshalTypeError) *{Value: "string", Type: reflect.Type(*reflect.rtype) *{t: (*"internal/abi.Type")(0x409f640)}, Offset: 73, Struct: "ConfigurationKey", Field: "configurationKey.readonly"}
```
Kann es sein, dass der Parser statt `"readonly":"true"` sowas wie `"readonly":true` haben will? Wie reden wir ihm das aus?

andig, 2023-10-15:
Kann gut sein, muss ich in Ruhe anschauen. Zu prüfen wäre auch, was die OCPP Spec dazu sagt. Wenn es ein Fehler des Chargers ist dann sollte der hier allerdings ordentlich zurück gegeben und nicht irgendwo verschluckt werden. Mich irritiert etwas, dass der im Logfile nicht auftaucht- hast Du eine Idee warum? Ignorieren wir den?

premultiply, 2023-10-15:
Nach JSON-Spec ist "true" ein String und kein Boolean.
Insofern ist die Implementierung in der Box ggf. einfach falsch falls an der Stelle seitens OCPP-Spec ein Booelan gefordert wird - andernfalls wäre nur die Lib zu streng. 🧐

andig, 2023-10-15:
Welche Json spec? Hier gehts um ocpp!

andig, 2023-10-15:
Readonly der lib ist jdfls ein bool: https://github.com/lorenzodonini/ocpp-go/blob/master/ocpp1.6/core/get_configuration.go#L14. Da das mit allen anderen Chargern funktioniert tippe ich auf Bug der Wallbox. Schön wäre trotzdem, wenn wir den Fehler sehen können 😩

premultiply, 2023-10-15:
Eben. OCPP 1.6j basiert ja auf JSON über Websocket und bei JSON ist `"true"` ein String und `true` ein Boolean.

Ich tippe daher bislang auch eher auf einen Bug in der Wallbox-Software, habe aber dazu bislang auch noch nichts Eindeutiges in der OCPP Spec gefunden ausser im Schema dass `readonly` ein Boolean ist...

tvand, 2023-10-16:
Ihr habt damit sicher recht. Ich verweise aber hier mal auf [Postels Law](https://en.wikipedia.org/wiki/Robustness_principle). SteVe parst die Response jedenfalls korrekt. Wahrscheinlich haben die Kollegen von EFR das nur mit dem Java-Parser getestet.

andig, 2023-10-16:
Das mag sein. Dennoch sind Spezifikationen dazu da eingehalten zu werden. Ob oder dass andere Systeme und Programmiersprachen entspannter sind spielt keine Rolle. Aus meiner Sicht geht es hier nur noch darum, wie wir den Fehler früher hätten identifizieren können. Beheben muss ihn der Hersteller. Die Box entspricht nicht dem Ocpp Standard!

andig, 2023-10-16:
@tvand könntest Du bitte den aktuellen master nochmal probieren (oder morgen das nightly)? Da sollte zumindest der Fehler im Logfile sichtbar werden. Upstream Issue warum diese Fehler verschluckt werden ist auch eröffnet. 

Hast Du schon ein Ticket beim Hersteller aufgemacht?

andig, 2023-10-16:
> Das ist nicht nett.

Was ist an Standards nicht nett? Hast Du dem Hersteller schon gesagt, dass es nicht nett ist, seine Produkte nicht Kompatibilitätstests zu unterziehen?

> Für mich war jetzt nicht eindeutig ersichtlich, dass hier der Standard verletzt wird. 

Ist es. Die Nachrichten sind falsch.

> Das hier dürfte auf eine halbe Zeile mehr Code an der richtigen Stelle hinauslaufen. 

Das ist eine gewagte Behauptung auf welcher Grundlage? Ich habe jetzt in 2h meiner Freizeit keine Lösung dafür gefunden.

Nachdem Du uns jetzt erklärst wie einfach das Problem zu lösen sei (was es nicht ist) würde ich nochmal darum bitten, Dir die Mühe zu machen den Hersteller auch nur über den Fehler zu informieren. Davon unabhängig aber ich Dir auch bereits den Workaround beschrieben. Die Diskussion ist an der Stelle für mich beendet.

Hier geht es nur noch um Sichtbarkeit des Fehlers- denn das _ist_ ein Problem von evcc.

andig, 2023-10-18:
@tvand sehen wir mit aktuellem master jetzt eine Fehlermeldung im Log? Vielen Dank.

tvand, 2023-10-19:
Komme erst am Wochenende dazu.

tvand, 2023-10-19:
Hersteller informiert. Bisher keine Rückmeldung.

andig, 2023-10-20:
> Komme erst am Wochenende dazu.

Ich mache hier zu. Falls der Fehler nicht geloggt werden sollte bitte nochmal melden.

closed 2023-10-20 by andig

tvand, 2023-10-22:
```
[ocpp  ] ERROR 2023/10/22 18:02:00 ocpp message (1067087506): FormationViolation - json: cannot unmarshal string into Go struct field ConfigurationKey.configurationKey.readonly of type bool ([3,"1067087506",{"configurationKey":[{"key":"AuthorizeRemoteTxRequests","readonly":"true","value":"false"},{"key":"ChargeProfileMaxStackLevel","readonly":"true","value":5},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":"true","value":"W"},{"key":"ChargingScheduleMaxPeriods","readonly":"true","value":20},{"key":"ClockAlignedDataInterval","readonly":"false","value":900},{"key":"ConnectionTimeOut","readonly":"false","value":300},{"key":"ConnectorPhaseRotation","readonly":"false","value":"unknown"},{"key":"GetConfigurationMaxKeys","readonly":"true","value":10},{"key":"HeartbeatInterval","readonly":"false","value":60},{"key":"LocalAuthListEnabled","readonly":"false","value":"false"},{"key":"LocalAuthListMaxLength","readonly":"true","value":20},{"key":"LocalAuthorizeOffline","readonly":"false","value":"true"},{"key":"LocalPreAuthorize","readonly":"false","value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":"true","value":15},{"key":"MeterValueSampleInterval","readonly":"false","value":300},{"key":"MeterValuesAlignedData","readonly":"false","value":"Energy.Active.Import.Register,Power.Active.Import"},{"key":"MeterValuesSampledData","readonly":"false","value":"Energy.Active.Import.Register,Power.Active.Import"},{"key":"NumberOfConnectors","readonly":"true","value":1},{"key":"NumberOfResetRetries","readonly":"false","value":0},{"key":"SendLocalListMaxLength","readonly":"true","value":20},{"key":"StopTransactionOnEVSideDisconnect","readonly":"false","value":"true"},{"key":"StopTransactionOnInvalidId","readonly":"false","value":"false"},{"key":"StopTxnAlignedData","readonly":"false","value":"Energy.Active.Import.Register,Power.Active.Import"},{"key":"StopTxnSampledData","readonly":"false","value":"Energy.Active.Import.Register,Power.Active.Import"},{"key":"SupportedFeatureProfiles","readonly":"true","value":"Core,LocalAuthListManagement,SmartCharging"},{"key":"TransactionMessageAttempts","readonly":"false","value":3},{"key":"TransactionMessageRetryInterval","readonly":"false","value":180},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":"false","value":"false"},{"key":"ckchargepowermax","readonly":"true","value":11000}]}])
```

premultiply, 2023-10-22:
Da kommt ein `"false"` (String) statt `false` (Bool).

andig, 2023-10-22:
Da ist nix zu machen. Spec ist Spec. Siehe https://github.com/lorenzodonini/ocpp-go/issues/190