evcc-io/evcc #10342: Zeitzonen Preis wird mit custom tariff nicht im UI angezeigt

VolkerK62, 2023-10-16:
### Describe the bug

Ich habe den configurable custom tariff https://github.com/evcc-io/evcc/pull/10306 ausprobiert.
Funktoniert sowohl bei grid, als auch bei feedin.
Wenn man aber einen Zeitzonen Tarif definiert, wird im UI bei Netzbezug kein Preis angezeigt.

### Steps to reproduce

.

### Configuration details

```yaml
tariffs:
  currency: EUR
  grid:
    type: custom 
    price: 
      source: mqtt
      topic: Haus/Preis_Bezug
      timeout: 25h
    zones:
      - days: Mo-So
        hours: 2-5
        price:
          source: mqtt
          topic: Haus/Preis_Bezug
          timeout: 25h
  feedin:
    type: custom 
    price: 
      source: mqtt
      topic: Haus/Preis_Einspeisung
      timeout: 25h
```

### Log details

```text
bitte Beschied geben, was evtl an Log nötig ist
```

### What type of operating system are you running?

Linux

### Version

0.121.0 (40a391da)

VolkerK62, 2023-10-16:
Gleiches Verhalten, auch wenn der Zonenpreis als fester Wert eingetragen wird
```
tariffs:
  currency: EUR
  grid:
    type: custom 
    price: 
      source: mqtt
      topic: Haus/Preis_Bezug
      timeout: 25h
    zones:
      - days: Mo-So
        hours: 2-5
        price: 0.22
  feedin:
    type: custom 
    price: 
      source: mqtt
      topic: Haus/Preis_Einspeisung
      timeout: 25h
```

andig, 2023-10-16:
> Wenn man aber einen Zeitzonen Tarif definiert, wird im UI bei Netzbezug kein Preis angezeigt.

Das geht ja auch nicht. Die Config sollte eigentlich einen Fehler liefern und abbrechen!

closed 2023-10-16 by andig

VolkerK62, 2023-10-17:
Ich habe das ausprobiert, um die Doku zu aktualisieren.
>Die Config sollte eigentlich einen Fehler liefern und abbrechen:

Es gab keinen Fehler. Deshlab dachte ich, es sollte möglich sein.

andig, 2023-10-17:
Sehr seltsam. Das *muss* einen Fehler geben!