evcc-io/evcc #10348: "Vorübergehendes Schnellladen im PV-Modus" aktiv bei 140 g/kWh Grünstromindex, obwohl in GUI < 140 g/kWh ausgewählt ist

All-Ex-U, 2023-10-16:
### Describe the bug

Bei Nutzung des Grünstromindex ist "Vorübergehendes Schnellladen im PV-Modus" aktiv, wenn der CO2-Wert von Grünstromindex **gleich** dem ausgewählten Wert bei "Smartes Netzladen" ist.

Erwartetes Ergebnis: Schnellladen darf nur aktiviert werden, wenn der CO2-Wert **kleiner** als der ausgewählte Wert ist.

**Fehler:**
- g/kWh gemäß Grünstromindex: 140 g
- eingestellte CO2-Grenze bei Smartes Netzladen: < 140 g/kWh
- in der GUI ist alles konsistent, d.h. dort wird "Saubere Energie" **nicht** angezeigt und der linke Balken im Dialog "Smartes Netzladen" ist **nicht** grün.
[image]
[image]
- Ergebnis: Lädt mit 11 kW (statt mit 4,2 kW)

**Alles OK:**
- g/kWh gemäß Grünstromindex: 140 g
- eingestellte CO2-Grenze bei Smartes Netzladen: < 145 g/kWh
- in der GUI wird "Saubere Energie" angezeigt und der linke Balken im Dialog "Smartes Netzladen" ist grün:
[image]
[image]
- Ergebnis: Lädt mit 11 kW

**Alles OK:**
- g/kWh gemäß Grünstromindex: 140 g
- eingestellte CO2-Grenze bei Smartes Netzladen: < 135 g/kWh
- in der GUI wird "Saubere Energie" nicht angezeigt
- Ergebnis: Lädt mit 4,2 kW

Evt. handelt es sich um ein Rundungs-Problem?

### Steps to reproduce

1. Grünstromindex aktivieren
2. Modus PV wählen
3. Smartes Netzladen auf < 140 g/kWh stellen
4. aktueller Grünstromindex beträgt laut GUI 140 g/kWh
5. Bei "Smartes Netzladen" ist die aktuelle Stunde nicht grün, d.h. vorübergehendes Schnellladen dürfte nicht aktiv sein
[image]
6. Die Meldung "Saubere Energie verfügbar." erscheint nicht.
7. Problem: Trotzdem wird mit 11 kW geladen.
[image]

Erwartetes Ergebnis: bei 140 g/kWh darf vorübergehendes Schnellladen nicht aktiv sein, wenn <140 g/kWh ausgewählt ist.

Ausgabe von ...evcc/api/state
```json
{
   "result":{
      "auth":{
         "vehicles":{            
         }
      },
      "batteryConfigured":false,
      "bufferSoc":0,
      "bufferStartSoc":0,
      "currency":"EUR",
      "greenShareHome":1,
      "greenShareLoadpoints":0.0058049163520389226,
      "gridConfigured":true,
      "gridEnergy":3247995.9,
      "gridPower":10807,
      "homePower":328.8999786376953,
      "loadpoints":[
         {
            "chargeCurrent":16,
            "chargeCurrents":[
               15.623000144958496,
               15.625,
               15.541999816894531
            ],
[... 360 lines cut ...]
Okt 16 14:34:36 linux evcc[178239]: [site  ] DEBUG 2023/10/16 14:34:36 ----
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 charge power: 10751W
Okt 16 14:34:36 linux evcc[178239]: [site  ] DEBUG 2023/10/16 14:34:36 pv power: 486W
Okt 16 14:34:36 linux evcc[178239]: [site  ] DEBUG 2023/10/16 14:34:36 grid power: 10800W
Okt 16 14:34:36 linux evcc[178239]: [site  ] DEBUG 2023/10/16 14:34:36 site power: 10800W
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 charge voltages: [231 232 231]V
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 detected connected phases: 3p
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 charge currents: [15.6 15.6 15.6]A
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 detected active phases: 3p
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 charge total import: 183.431kWh
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 charger status: C
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 soc estimated: 76.28% (vehicle: 76.00%)
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 vehicle soc: 76%
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 vehicle soc limit: 100%
Okt 16 14:34:36 linux evcc[178239]: [lp-1  ] DEBUG 2023/10/16 14:34:36 vehicle range: 352km
Okt 16 14:34:51 linux evcc[178239]: [site  ] DEBUG 2023/10/16 14:34:51 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.0

andig, 2023-10-16:
@naltatis Gui oder Code korrigieren? Bei Preis 30ct würde ich schon erwarten, dass bei 30ct geladen würde?

naltatis, 2023-10-18:
Oh, das ist wirklich inkonsistent. Ich hätte geschworen, dass wir in der Regellogik `<` und nicht `<=` verwenden, aber das scheint nicht der Fall zu sein. Ich passe die UI so an, dass wir `≤` verwenden. Glaube auch, dass das eher der Erwartung des Nutzers entspricht.

naltatis, 2023-10-18:
In 0.121.1 ist ein Teil hiervon bereits durch https://github.com/evcc-io/evcc/commit/51fa4905492d1809344157760562fde5a3ec453b behoben. `Saubere Energie verfügbar.` wird damit in der UI nur noch dann angezeigt, wenn die Regelung das auch umsetzt.

Ausstehend ist nur noch, dass wir das ≤ auch sauber bei der Auswahl kommunizieren.

PR #10381 (2023-10-18): Smart Grid Charging: use ≤ instead of < for limits
Options, highlighting and select is updated to use lower then or equals logic.

[image]

fixes #10348
closed 2023-10-18 by andig