evcc-io/evcc #10360: ocpp nil exception bei disconnect

eas-ol, 2023-10-17:
### Describe the bug

Disconnected wirft einen nil exception wenn ein CP nicht in der configuration angelegt ist und sich wieder abmeldet.

In err ist eine Nachricht wie cp nicht konfiguriert

Somit ist err nicht nil und cp.connect(false) wird aufgerufen aber der cp ist nil
Es fehlt die Abfrage für "cp !=nil"

// ChargePointDisconnected implements ocpp16.ChargePointConnectionHandler
func (cs *CS) ChargePointDisconnected(chargePoint ocpp16.ChargePointConnection) {
	cs.log.DEBUG.Printf("charge point disconnected: %s", chargePoint.ID())

	if cp, err := cs.ChargepointByID(chargePoint.ID()); err == nil {
		cp.connect(false)
	}
}

### Steps to reproduce

Beispiel:
1. 2 WB mit OCPP
2. Eine WB wurde in der Yaml angelegt
3. Beider Wallboxen melden sich am CS an.
4. Nach einer gewissen Zeit meldet sich die nicht konfigurierte WB ab.

### Configuration details

```yaml
nicht relevant:

chargers:
- type: ocpp 
  stationid: ABL_10283305
  connector: 1
  name: wallbox1
```

### Log details

```text
wird nachgereicht wenn nötig
```

### What type of operating system are you running?

Linux

andig, 2023-10-17:
Das ist leider mit der Library schwer zu testen. Was steht denn im Log?

> Somit ist err nicht nil und cp.connect(false) wird aufgerufen aber der cp ist nil

Wie soll das passieren?

```
func (cs *CS) ChargepointByID(id string) (*CP, error) {
	cs.mu.Lock()
	defer cs.mu.Unlock()

	cp, ok := cs.cps[id]
	if !ok {
		return nil, fmt.Errorf("unknown charge point: %s", id)
	}
	if cp == nil {
		return nil, fmt.Errorf("charge point not configured: %s", id)
	}
	return cp, nil
}
```

Leider fehlt Log und v.a. auch Deine Version!

eas-ol, 2023-10-17:
`
if cp == nil {
		return nil, fmt.Errorf("charge point not configured: %s", id)
}
`
Das ist der Fall den ich meine.
Wenn diese Abfrage true ist wird cp==nil und err!=nil zurückgegeben

in der ChargePointDisconnected funktion wird daraufhin
auf dem cp der nil ist cp.connect(false) ausgerufen.
'if cp, err := cs.ChargepointByID(chargePoint.ID()); err == nil {
	cp.connect(false)
}'

ich denke das würde bei
'
if !ok {
		return nil, fmt.Errorf("unknown charge point: %s", id)
}
'
auch passieren.

andig, 2023-10-17:
> in der ChargePointDisconnected funktion wird daraufhin auf dem cp der nil ist cp.connect(false) aufgerufen.

Glaubst Du das oder ist das so???

    if cp, err := cs.ChargepointByID(chargePoint.ID()); err == nil {

Hier wird auf err == nil geprüft. Wir Du eben geschrieben hast ist aber err != nil:

> Wenn diese Abfrage true ist wird cp==nil und err!=nil zurückgegeben

Lass uns bitte nicht raten- was passiert steht im Logfile!

eas-ol, 2023-10-17:
Du hast natürlich recht, raten bringt nichts.

Werde sehen ob ich das Problem noch einmal habe und dann die logs hochladen.

closed 2023-10-17 by eas-ol

andig, 2023-10-17:
So machen wir das 👍🏻