evcc-io/evcc #10366: OCPP WB StartTransaction "status":"Expired"

eas-ol, 2023-10-17:
### Describe the bug

Auto startet das Laden nicht.

Auto ist angeschlossen, WB ist an, EVCC ist gestartet

Wenn ich von Off auf Schnell umstelle wird versucht die WB zu starten (RemoteStartTransaction) und wenn die WB starten (StartTransaction) meldet erscheint:

`sent JSON message to SN10052306091364: [3,"7",{"idTagInfo":{"status":"Expired"},"transactionId":0}]`

Wenn ich in StartTransaction von < auf > ändere, wie in StopTransaction dann geht es.

```
// expired request
	if request.Timestamp != nil && conn.clock.Since(request.Timestamp.Time) < transactionExpiry {
		res := &core.StartTransactionConfirmation{
		IdTagInfo: &types.IdTagInfo{
			Status: types.AuthorizationStatusExpired,
		},
	}

	return res, nil
	}
```

Hoffe das hilft.

### Steps to reproduce

1. WB einschalten
2. Auto anstecken
3. EVCC Starten
4. Max Charge einstellen bei evcc weil charger auf 10A begrenzt ist
5. EVCC von Off auf Schnell stellen
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  ocpp: trace
  cache: error

# unique installation id
plant: ************************

interval: 10s # control cycle interval

meters:
- type: template
  template: powerfox-poweropti
  usage: grid
  user: ***********
  password: ***********
  name: grid1

chargers:
- type: template
  template: ocpp
  connector: 1
  name: wallbox1
  stationid: SN10052306091364

[... 84 lines cut ...]
[lp-1  ] DEBUG 2023/10/17 11:59:23 charge power: 6027W
[site  ] DEBUG 2023/10/17 11:59:23 grid power: -1351W
[site  ] DEBUG 2023/10/17 11:59:23 pv power: 7378W
[site  ] DEBUG 2023/10/17 11:59:23 site power: -1351W
[lp-1  ] DEBUG 2023/10/17 11:59:23 charge total import: 0.985kWh
[lp-1  ] DEBUG 2023/10/17 11:59:23 charger status: B
[ocpp  ] TRACE 2023/10/17 11:59:24 received JSON message from SN10052306091364: [2,"7","StartTransaction",{"connectorId":1,"idTag":"evcc","meterStart":3,"timestamp":"2023-10-17T09:59:20Z"}]
[ocpp  ] TRACE 2023/10/17 11:59:24 sent JSON message to SN10052306091364: [3,"7",{"idTagInfo":{"status":"Expired"},"transactionId":0}]
[site  ] DEBUG 2023/10/17 11:59:28 ----
[lp-1  ] DEBUG 2023/10/17 11:59:28 charge power: 6027W
[site  ] DEBUG 2023/10/17 11:59:30 grid power: -1324W
[site  ] DEBUG 2023/10/17 11:59:30 pv power: 7351W
[site  ] DEBUG 2023/10/17 11:59:30 site power: -1324W
[lp-1  ] DEBUG 2023/10/17 11:59:30 charge total import: 0.985kWh
[lp-1  ] DEBUG 2023/10/17 11:59:30 charger status: B
[site  ] DEBUG 2023/10/17 11:59:38 ----
```

### What type of operating system are you running?

Linux

### Version

master: 4b5e8029

andig, 2023-10-18:
Yikes- böses refactor!

andig, 2023-10-18:
Closed in a346c104e92f32f20e7d8cb15d104e6e8bba39aa

closed 2023-10-18 by andig