evcc-io/evcc #10370: battery: energy direction maybe wrong?

Jambola77, 2023-10-17:
### Describe the bug

I have a battery attached to my pv installation (AC with inverter) and the Victron Cerbo shows me positive power values when the battery is charged and negative value when energy from the battery is used.
In evcc it seems that positive values are on the generator side (like pv meter) and so the display is wrong.

[image]

In the picture the battery is charging (positive values in mqtt).

### Steps to reproduce

Start evcc.

### Configuration details

```yaml
- name: grid
  type: custom
  power:
    source: mqtt
    broker: localhost:1883
    topic: cerbo/ACPOWER

site:
- title: xyz
  residualPower: 200
  meters:
          pv:
            - pv
            - mikrowr
          grid: grid
          battery: mybat
```

### Log details

```text
Oct 17 16:01:37 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:37 battery soc: 53%
Oct 17 16:01:37 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:37 battery power: 2855W
Oct 17 16:01:37 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:37 grid power: -3W
Oct 17 16:01:37 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:37 site power: 3052W
Oct 17 16:01:47 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:47 ----
Oct 17 16:01:47 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:47 pv power: 3330W
Oct 17 16:01:47 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:47 battery soc: 53%
Oct 17 16:01:47 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:47 battery power: 2670W
Oct 17 16:01:47 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:47 grid power: -16W
Oct 17 16:01:47 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:47 site power: 2854W
Oct 17 16:01:57 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:57 ----
Oct 17 16:01:57 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:57 pv power: 3118W
Oct 17 16:01:57 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:57 battery soc: 53%
Oct 17 16:01:57 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:57 battery power: 2725W
Oct 17 16:01:57 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:57 grid power: 24W
Oct 17 16:01:57 raspi4 evcc[30747]: [site  ] DEBUG 2023/10/17 16:01:57 site power: 2949W
```

### What type of operating system are you running?

Linux

### Version

0.121.0

jheinitz, 2023-10-17:
Hello!

I think your configuration is missing the definition of `mybat`. However, you could add a `scale: -1` to the definition in order to change the sign of the values. My battery definition looks like this:
```
  - name: storage1
    type: custom
    power:
      source: mqtt
      topic: modbus/power/sun2000/storage/1
      jq: .fields.storage_charge_discharge_power
      scale: -1
    soc:
      source: mqtt
      topic: modbus/power/sun2000/storage/1
      jq: .fields.storage_state_of_capacity
    capacity: 10
```
Hopefully it solves your problem.

Best regards

jens

Jambola77, 2023-10-17:
Hello Jens,
thanks for the quick answer. 
I will add scale -1 to my config.
Cheers, Pete

closed 2023-10-17 by Jambola77