evcc-io/evcc #10371: 2 Wallbox PulsarPlus möglich? Konnte nichts finden

copadewe, 2023-10-17:
Hallo, ich konnte dazu nichts finden.
Ich habe im Haus 2 Wallboxen PulsarPlus, nach lange hin und her habe ich die erste jetzt in evcc bekommen. OCPP läuft!
Bei der zweiten geht OCPP nicht die Verbindung bleibt rot.
Hat jemand dieselbe Situation und es gelöst?
Danke!
Und um Github genüge zu tun:
Hello, I could not find anything about this.
I have in the house 2 wallboxes PulsarPlus, after long back and forth I got the first now in evcc. OCPP runs!
With the second OCPP does not go the connection remains red.
Has anyone the same situation and solved it?
Thank you!

copadewe, 2023-10-17:
> Ich glaube in #10187 würde etwas in die Richtung eingebaut

Was ist denn das die urlsuffix?
add Port and URL urlsuffix in settings to fix ip:port//stationid problems with wallbox and simulator ws://<evcc>:8887/<urlsuffix><stationid>

copadewe, 2023-10-18:
> Ich konnte 2 PulsarPlus Boxen erfolgreich verbinden. Dazu in den Wallbox Einstellungen die ID vergeben, z.B. "1", und dann dort in der URL für den OCCP Server (also evcc) die ID hinten als Pfad anhängen, z.B. ws://192.168.178.100:8887/2
> 
> Und in der evcc.yaml die StationID natürlich auch entsprechend konfigurieren, also z.B.
> 
>     * type: template
>       template: pulsarplus
>       stationid: 1
>       name: wallbox1
> 
> 
> Nicht ganz optimal ist, dass immer beide Wallboxen eingeschaltet sein müssen, sonst kommt evcc nicht hoch (weil es ja auf beide konfigurierten Boxen wartet).

Und das geht bei dir?
Bei mir kommt dann:
[ocpp  ] DEBUG 2023/10/18 08:37:23 charge point connected: 1
[ocpp  ] DEBUG 2023/10/18 08:37:44 charge point connected: 2
[main  ] FATAL 2023/10/18 08:37:49 cannot create charger 'Wallbox2': cannot create charger 'template': cannot create charger 'ocpp': connector 2 exceeds max available connectors: 1

StefanSchoof, 2023-10-18:
Ich glaube, dass ist bisher nur im nightly

copadewe, 2023-10-18:
> nightly

Danke Stefan, aber was ist nightly? bin neu bei evccc... habe es über HomeAssistant als Addon

closed 2023-10-18 by premultiply

msc76, 2023-10-18:
> > Ich konnte 2 PulsarPlus Boxen erfolgreich verbinden. Dazu in den Wallbox Einstellungen die ID vergeben, z.B. "1", und dann dort in der URL für den OCCP Server (also evcc) die ID hinten als Pfad anhängen, z.B. ws://192.168.178.100:8887/2
> > Und in der evcc.yaml die StationID natürlich auch entsprechend konfigurieren, also z.B.
> > ```
> > * type: template
> >   template: pulsarplus
> >   stationid: 1
> >   name: wallbox1
> > ```
> > 
> > 
> > Nicht ganz optimal ist, dass immer beide Wallboxen eingeschaltet sein müssen, sonst kommt evcc nicht hoch (weil es ja auf beide konfigurierten Boxen wartet).
> 
> Und das geht bei dir? Bei mir kommt dann: [ocpp ] DEBUG 2023/10/18 08:37:23 charge point connected: 1 [ocpp ] DEBUG 2023/10/18 08:37:44 charge point connected: 2 [main ] FATAL 2023/10/18 08:37:49 cannot create charger 'Wallbox2': cannot create charger 'template': cannot create charger 'ocpp': connector 2 exceeds max available connectors: 1
> 
> Es sind aber beide Boxen lt. mywallbox.com mit OCPP verbunden.

Ja, klappt bei mir und zwar auch schon mit früheren Versionen und aktuell mit 0.120.3.

[image]

yaml: 
- type: template
  template: pulsarplus
  stationid: 2
  connecttimeout: 10m
  timeout: 2m
  name: wallbox2

- type: template
  template: pulsarplus
  stationid: 1
  connecttimeout: 10m
  timeout: 2m
  name: wallbox1

Und die Wallboxen:
ID 2 und URL ws://192.168.178.100:8887/2
ID1 und ws://192.168.178.100:8887/1

Log ganz unspektakulär:
Oct 18 10:13:11 pihole systemd[1]: Started evcc.
Oct 18 10:13:11 pihole evcc[675735]: [main  ] INFO 2023/10/18 10:13:11 evcc 0.120.3
Oct 18 10:13:11 pihole evcc[675735]: [main  ] INFO 2023/10/18 10:13:11 using config file: /etc/evcc.yaml
Oct 18 10:13:11 pihole evcc[675735]: [main  ] INFO 2023/10/18 10:13:11 starting ui and api at :7070
Oct 18 10:13:12 pihole evcc[675735]: [db    ] INFO 2023/10/18 10:13:12 using sqlite database: /var/lib/evcc/evcc.db
Oct 18 10:13:13 pihole evcc[675735]: [2     ] DEBUG 2023/10/18 10:13:13 waiting for chargepoint: 10m0s
Oct 18 10:13:13 pihole evcc[675735]: [1     ] DEBUG 2023/10/18 10:13:13 waiting for chargepoint: 10m0s
Oct 18 10:13:37 pihole evcc[675735]: [ocpp  ] DEBUG 2023/10/18 10:13:37 charge point connected: 2
Oct 18 10:13:55 pihole evcc[675735]: [ocpp  ] DEBUG 2023/10/18 10:13:55 charge point connected: 1
Oct 18 10:14:00 pihole evcc[675735]: [site  ] INFO 2023/10/18 10:14:00 site config:
Oct 18 10:14:01 pihole evcc[675735]: [site  ] INFO 2023/10/18 10:14:01   meters:      grid ✓ pv ✓ battery ✗
Oct 18 10:14:01 pihole evcc[675735]: [site  ] INFO 2023/10/18 10:14:01     grid:      power ✓ energy ✗ currents ✗
Oct 18 10:14:01 pihole evcc[675735]: [site  ] INFO 2023/10/18 10:14:01     pv 1:      power ✓ energy ✓ currents ✓
Oct 18 10:14:01 pihole evcc[675735]: [site  ] INFO 2023/10/18 10:14:01   vehicles:
Oct 18 10:14:01 pihole evcc[675735]: [site  ] INFO 2023/10/18 10:14:01     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
Oct 18 10:14:01 pihole evcc[675735]: [lp-1  ] INFO 2023/10/18 10:14:01 loadpoint 1:
Oct 18 10:14:01 pihole evcc[675735]: [lp-1  ] INFO 2023/10/18 10:14:01   mode:        pv
Oct 18 10:14:01 pihole evcc[675735]: [lp-1  ] INFO 2023/10/18 10:14:01   charger:     power ✗ energy ✓ currents ✗ phases ✗ wakeup ✗
Oct 18 10:14:01 pihole evcc[675735]: [lp-1  ] INFO 2023/10/18 10:14:01   meters:      charge ✗
Oct 18 10:14:01 pihole evcc[675735]: [lp-2  ] INFO 2023/10/18 10:14:01 loadpoint 2:
Oct 18 10:14:01 pihole evcc[675735]: [lp-2  ] INFO 2023/10/18 10:14:01   mode:        minpv
Oct 18 10:14:01 pihole evcc[675735]: [lp-2  ] INFO 2023/10/18 10:14:01   charger:     power ✗ energy ✓ currents ✗ phases ✗ wakeup ✗
Oct 18 10:14:01 pihole evcc[675735]: [lp-2  ] INFO 2023/10/18 10:14:01   meters:      charge ✗
Oct 18 10:14:01 pihole evcc[675735]: [lp-1  ] DEBUG 2023/10/18 10:14:01 phase timer inactive
Oct 18 10:14:01 pihole evcc[675735]: [lp-1  ] DEBUG 2023/10/18 10:14:01 pv timer inactive
Oct 18 10:14:01 pihole evcc[675735]: [lp-1  ] DEBUG 2023/10/18 10:14:01 guard timer inactive
Oct 18 10:14:01 pihole evcc[675735]: [lp-2  ] DEBUG 2023/10/18 10:14:01 phase timer inactive
Oct 18 10:14:01 pihole evcc[675735]: [lp-2  ] DEBUG 2023/10/18 10:14:01 pv timer inactive
Oct 18 10:14:01 pihole evcc[675735]: [lp-2  ] DEBUG 2023/10/18 10:14:01 guard timer inactive
Oct 18 10:14:01 pihole evcc[675735]: [site  ] WARN 2023/10/18 10:14:01 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
Oct 18 10:14:01 pihole evcc[675735]: [site  ] DEBUG 2023/10/18 10:14:01 ----

copadewe, 2023-10-18:
> Ja, klappt bei mir und zwar auch schon mit früheren Versionen und aktuell mit 0.120.3.
> [image]
>
> yaml:
>
>     * type: template
>       template: pulsarplus
>       stationid: 2
>       connecttimeout: [...]

Klappt nicht, für mich ist evcc damit vorbei.
[ocpp  ] DEBUG 2023/10/18 11:43:45 charge point connected: 1
[main  ] FATAL 2023/10/18 11:44:45 cannot create charger 'Wallbox1': cannot create charger 'template': cannot create charger 'ocpp': ocpp message (533059298): GenericError - Request timed out
[main  ] FATAL 2023/10/18 11:44:45 will attempt restart in: 5m0s