evcc-io/evcc #10376: Leistungswerte für Grid Import und Export verkehrt herum angezeigt.

Phil2204, 2023-10-17:
### Describe the bug

Leistungswerte für Grid Import und Export jeweils falsch herum angezeigt. Ich habe EVCC als Addon in HA installiert und Daten vom Inverter über MQTT eingebunden.
Dummy-Charger vorerst eingestellt.

[image]
 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistant.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 15s # control cycle interval
sponsortoken: 

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  mqtt: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
[... 241 lines cut ...]
[lp-1  ] DEBUG 2023/10/17 21:25:08 charger status: B
[lp-1  ] DEBUG 2023/10/17 21:25:08 charger: guard elapse
[lp-1  ] DEBUG 2023/10/17 21:25:08 guard timer inactive
[lp-1  ] DEBUG 2023/10/17 21:25:08 pv charge current: 0.138A = 0A + 0.138A (-95W @ 3p)
[site  ] DEBUG 2023/10/17 21:25:23 ----
[lp-1  ] DEBUG 2023/10/17 21:25:23 charge power: 0W
[site  ] DEBUG 2023/10/17 21:25:23 pv power: 0W
[site  ] DEBUG 2023/10/17 21:25:23 battery soc: 15%
[site  ] DEBUG 2023/10/17 21:25:23 battery power: 0W
[site  ] DEBUG 2023/10/17 21:25:23 grid power: -487W
[site  ] DEBUG 2023/10/17 21:25:23 site power: -387W
[lp-1  ] DEBUG 2023/10/17 21:25:23 charger status: B
[lp-1  ] WARN 2023/10/17 21:25:23 charger out of sync: expected enabled, got disabled
[lp-1  ] DEBUG 2023/10/17 21:25:23 pv charge current: 0.561A = 0A + 0.561A (-387W @ 3p)
[lp-1  ] DEBUG 2023/10/17 21:25:23 charger enable
[lp-1  ] DEBUG 2023/10/17 21:25:23 wake-up timer: start
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.121.0

premultiply, 2023-10-17:
Dann einfach die Daten richtig übergeben (oder passend skalieren)
oder einfach die integrierten Templates statt Custom-Meters verwenden.

closed 2023-10-17 by premultiply

Phil2204, 2023-10-18:
Leider funktioniert das einbinden des Meters nur über mqtt. Demnach muss ich doch über Custom gehen oder nicht? Skalieren habe ich bereits ohne Erfolg versucht.

Phil2204, 2023-10-18:
Zudem kommen die Werte korrekt mit Vorzeichen im HA und im mqtt Explorer an.[image]

premultiply, 2023-10-18:
Grid scheint schon mal falsch herum zu sein.
Bei den anderen sieht man es nicht bei 0...