evcc-io/evcc #10377: Entsoe pricing forecast till mignight

kebui, 2023-10-17:
### Describe the bug

Since last week I'm using the entsoe integration (region Netherlands). However, the planner only seems to fetch prices up untill midnight. At midnight it shows prices for the rest of the day. For example now (21:00) it only shows prices untill 00:00 as shown below. Whereas entsoe, has prices available up until 00:00 the next day.

[image]

### Steps to reproduce

1.Configure entsoe integration
2.Configure region BZN|NL
3.Switch to PV mode
4. Select departure time
5. Select departure tab

...

### Configuration details

```yaml
# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # or variable tariffs
    type: entsoe
    domain: BZN|NL
    securitytoken: 83xxxx
    #tax: 0.21
    charges: 0.1724
```

### Log details

```text
[entsoe] ERROR 2023/10/17 21:27:26 EOF
[main  ] ERROR 2023/10/17 21:27:26 failed configuring grid tariff: cannot create tariff 'entsoe': EOF
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.121.0

andig, 2023-10-18:
Logfile does not match error description?

andig, 2023-10-18:
Afaikt we are requesting +24 hours:

		"PeriodEnd":    {now.Add(duration).Format(numericDateFormat)},

If that does not work Entsoe probably doesn't return the data. 

Closing due to missing logfile that could demonstrate the problem.

LarsTh86, 2023-10-20:
the same for me

andig, 2023-10-20:
> Closing due to missing logfile that could demonstrate the problem.

"the same for me" does not change that in any way :(

LarsTh86, 2023-10-21:
Were can I get the log file?

Ginoooke, 2023-10-22:
I have the same issue.
EVCC is only showing pricing untill midnight (no errors in log file).
If I run the api call separately I do get the full results including the next day (see attached file).

API call results in a dataset containing 2 TimeSeries (mRID 1 & 2).
I believe only the first set of TimeSeries is captured by EVCC.

[api result.pdf](https://github.com/evcc-io/evcc/files/13063777/api.result.pdf)

andig, 2023-10-22:
Fixed in https://github.com/evcc-io/evcc/commit/8f82e1a55b8f0ecb20fdea66cceee3b6cb0222af