evcc-io/evcc #10387: Keba Wallbox stoppt Ladevorgang nicht

stevienull, 2023-10-18:
### Describe the bug

Hallo.

ich habe eine Keba P30 C-Serie mit S10 Phasenumschalter im Einsatz.

Ich habe aktuell das Problem, dass der Ladevorgang nicht gestoppt wird. Die Phasenumschaltung klappt (3<>1) allerdings wird im PV Modus nicht abgeschaltet wenn (abends) die Sonne weg ist. Die Keba läd dann mit 1,4 kW weiter. 

Selbes Verhalten sehe ich, wenn ich bei zu wenig

 Sonne auf "Schnell" stelle => Fahrzeug läd. Stelle ich dann auf PV, würde ich erwarten, dass er nach x min ausschaltet. Auch hier läd das Fahrzeug weiter.

Evtl. habe ich auch in der Konfig noch einen Fehler drin. Vielleicht kann mal jemand drüber schauen.

Sponsor Token ist eingetragen. 

Gruß
Stefan

[log.txt](https://github.com/evcc-io/evcc/files/13031712/log.txt)

### Steps to reproduce

1. Fahrzeug laden (PV Modus)
2. warten bis die Sonne untergeht

oder

1. Fahrzeug laden (Schnell Modus)
2. auf PV Modus umstellen

### Configuration details

```yaml
# open evcc at http://evcc.local:7070

sponsortoken:  <token>

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: ERROR
levels:
  cache: error
  wallbox5: trace
  site: debug

# unique installation id
plant: <plantid>

interval: 10s # control cycle interval

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.1.46  
  name: grid1
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.1.160  
  port: 502  
  capacity: 7.2  
  name: battery3
- name: sb10
  type: template
  template: sma-inverter
  usage: pv
  host: 192.168.1.176 # IP-Adresse oder Hostname
  password: <pass>
- name: sb5
  type: template
  template: sma-inverter
  usage: pv
  host: 192.168.1.177 # IP-Adresse oder Hostname
  password: <pass>

chargers:
- type: template
  template: keba-modbus
  name: wallbox5
  # Modbus TCP
  modbus: tcpip
  id: 255
  host: 192.168.1.98 # Hostname
  port: 502 # Port 

vehicles:
- type: template
  template: enyaq 
  user: <mail>  
  password: <pass>  
  capacity: 58  
  icon: car  
  cache: 15m  
  mode: PV  
  targetSoc: 80  
  name: Enyaq

loadpoints:
- title: Carport
  charger: wallbox5
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  enable: # pv mode enable behavior
    delay: 300s # threshold must be exceeded for this long
    threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable:
    threshold: 200 # maximum import power (W)
    delay: 360s
  guardDuration: 5m # switch charger contactor not more often than this (default 5m)
site:
  title: Home
  bufferSoc: 80 # Hausbatterie wird oberhalb Soc 80% als Puffer genutzt
  residualPower: 100
  meters:
    grid: grid1
    pvs:
    - sb10
    - sb5
    batteries:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.30 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.15 # [currency]/kWh
```

### Log details

```text
Siehe Anhang
```

### What type of operating system are you running?

Linux

### Version

0.120.0 / 0.121.0

VolkerK62, 2023-10-18:
```
  disable:
    threshold: 200 # maximum import power (W)
    delay: 360s
```
Du sagst hier, dass 6 Minuten lang 200W aus dem Netz bezogen werden können, bevor abgeschaltet wird.
In Verbindung mit `residualpower: 100` reduziert sich dieser Wert auf 100W Netzbezug am Gridmeter.
Diese werden aber nie realisiert, weil der Hausspeicher das ausgleicht. Wenn der leer ist, würde abgeschaltet werden.
Wenn du `threshold: 0` einstellst, sollte dein Problem behoben sein.

VolkerK62, 2023-10-18:
Deshalb auch der Individualisierungshinweis in der Doku  https://docs.evcc.io/docs/guides/setup

stevienull, 2023-10-18:
Danke schonmal für die schnelle Rückmeldung 

Das ist ein guter Hinweis. Ich werde das morgen einmal testen und Feedback geben.

VolkerK62, 2023-10-18:
Ich habe jetzt erst gesehen, dass da ein LOG dabei war. Demnach wird die Batterie gar nicht genutzt.
Allerdings fehlen darin entscheidende Angaben.
Ändere in der yaml mal `log: ERROR` in `log: debug`, dann sind für die meisten Fälle ausreichende Informationen im Log verfügbar.

andig, 2023-10-18:
Kein Log...

stevienull, 2023-10-19:
Hallo zusammen,

@VolkerK62 
Ich habe gerade nochmal getestet mit Loglevel DEBUG
Leider war nicht genug Überschuss - also erst auf MIN+PV und dann umgestellt auf PV. Hat aber auch mit geänderter Einstellung 
` disable:
    threshold: 0`
auch nicht abgeschaltet (habs bestimmt ne viertel Stunde laufen lassen).

Hier das neue Log:
[evcc_20231019.log](https://github.com/evcc-io/evcc/files/13044481/evcc_20231019.log)

@andig auch oben hing schon ein Log dran.

Gruß
Stefan

andig, 2023-10-19:
> auch oben hing schon ein Log dran.

Wo?

andig, 2023-10-19:
```
Oct 19 16:44:10 openhabianpi evcc[28004]: [lp-1  ] DEBUG 2023/10/19 16:44:10 set charge mode: pv
Oct 19 16:44:10 openhabianpi evcc[28004]: [site  ] DEBUG 2023/10/19 16:44:10 ----
Oct 19 16:44:10 openhabianpi evcc[28004]: [lp-1  ] DEBUG 2023/10/19 16:44:10 charge power: 1370W
Oct 19 16:44:10 openhabianpi evcc[28004]: [site  ] DEBUG 2023/10/19 16:44:10 pv power: 637W
Oct 19 16:44:10 openhabianpi evcc[28004]: [site  ] DEBUG 2023/10/19 16:44:10 battery soc: 100%
Oct 19 16:44:10 openhabianpi evcc[28004]: [site  ] DEBUG 2023/10/19 16:44:10 battery power: 0W
Oct 19 16:44:10 openhabianpi evcc[28004]: [site  ] DEBUG 2023/10/19 16:44:10 grid power: 1148W
Oct 19 16:44:10 openhabianpi evcc[28004]: [site  ] DEBUG 2023/10/19 16:44:10 grid powers: [1078 89 -19]W
Oct 19 16:44:10 openhabianpi evcc[28004]: [site  ] DEBUG 2023/10/19 16:44:10 grid currents: [4.85 0.762 -0.9]A
Oct 19 16:44:10 openhabianpi evcc[28004]: [site  ] DEBUG 2023/10/19 16:44:10 site power: 1248W
Oct 19 16:44:10 openhabianpi evcc[28004]: [lp-1  ] DEBUG 2023/10/19 16:44:10 charge currents: [6.04 0 0]A
Oct 19 16:44:10 openhabianpi evcc[28004]: [lp-1  ] DEBUG 2023/10/19 16:44:10 detected active phases: 1p
Oct 19 16:44:10 openhabianpi evcc[28004]: [lp-1  ] DEBUG 2023/10/19 16:44:10 charge total import: 1779.434kWh
Oct 19 16:44:10 openhabianpi evcc[28004]: [lp-1  ] DEBUG 2023/10/19 16:44:10 charger status: C
Oct 19 16:44:10 openhabianpi evcc[28004]: [lp-1  ] DEBUG 2023/10/19 16:44:10 pv charge current: 0.576A = 6A + -5.42A (1248W @ 1p)
```

Lädt tatsächlich weiter. Liegt vmtl. an den Batterieeinstellungen? Was hast Du da im Gui?

VolkerK62, 2023-10-19:
Was in beiden Logs auffällt:
```
DEBUG 2023/10/19 16:58:19 battery soc: 100%
DEBUG 2023/10/19 16:58:19 battery power: 0W
```
Batterie ist immer bei 100% und wird nicht verwendet.

stevienull, 2023-10-20:
Scheint tatsächlich an der Batterie zu liegen. Habe die Batterie jetzt aus der Konfig genommen und getestet. (Batterie händisch ausgeschaltet). Erst auf MIN+PV => Auto läd. Dann auf PV => zu wenig Überschuss. Pausiere in x min. Danach wird der Ladevorgang Ordnungsgemäß gestoppt.

Was kann ich noch testen, um die Batterie wieder mit rein zu bekommen?

Als Info: Ich habe vorher die KEBA UDP Implementierung genutzt. Da ging auch die Batterie. Dann habe ich auf KEBA Modbus gewechselt (wg. der Phasenumschaltung). Danach hatte ich das beschriebene Verhalten.

VolkerK62, 2023-10-20:
>Ich habe vorher die KEBA UDP Implementierung genutzt. Da ging auch die Batterie. Dann habe ich auf KEBA Modbus gewechselt (wg. der Phasenumschaltung). Danach hatte ich das beschriebene Verhalten.

Ich kann mir nicht vorstellen, dass das eine mit dem anderen zu tun hat.

Ganz vage Vermutung (ohne von der Materie Ahnung zu haben):
Du hast SHM, evtl spielt der da mit rein.
Oft findet sich in Konfigurationen mit SHM noch ein `hems` Eintrag in der Konfig
```
hems:
  type: sma
  allowcontrol: false
```

VolkerK62, 2023-10-20:
Eigentlich kannst du die batterie wieder rein nehmen.
Musst nur die Batterieeinstellungen alle auf 100% stellen, dann sollte das Laden ganz normal funktionieren.
Klick auf die Batterie-% im UI, dann kommst du in die Einstellungen.

stevienull, 2023-10-31:
Hab die Batterie gemäß den letzten 2 Posts wieder rein genommen. Jetzt funktionierts wieder wie es soll. Danke