evcc-io/evcc #10388: Phoenix charger (phoenix-em-eth) throws i/o timeout after upgrade to 0.121.0

mclane, 2023-10-18:
### Describe the bug

I just upgraded from 0.119.5 to 0.121.0. After the upgrade, I get i/o timeout errors for the charger. evcc.yaml was untouched. The Phoenix controller does not control any meter; so I wonder why the debug command for the charger apparently generates meter-related messages. The charge meter is connected separately through a proteus serial to WLAN adapter. Up to 0.119.5 this configuration worked withut problems.

### Steps to reproduce

1. Upgrade from 0.119.5 to 0.121.0
2. Restart evcc service

### Configuration details

```yaml
The charger is configured like this:

meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.159
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.159
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.159
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
- type: template
  template: sunspec-inverter
  id: 2
  host: 192.168.0.40
  port: 502
  usage: pv
  modbus: tcpip
  name: pv4
- name: charge
  type: modbus
  model: sdm72
  rtu: false
  id: 3
  uri: 192.168.0.144:9999
  power: Power
  energy: Sum

chargers:
- type: template
  template: phoenix-em-eth
  host: 192.168.0.225
  port: 502
  name: wallbox6

```
```

### Log details

```text
uli@UrsaMajor:~$ evcc -l debug charger
[main  ] INFO 2023/10/18 19:28:22 evcc 0.121.0
[main  ] INFO 2023/10/18 19:28:22 using config file: /etc/evcc.yaml
[db    ] INFO 2023/10/18 19:28:22 using sqlite database: /home/uli/.evcc/evcc.db
[mqtt  ] INFO 2023/10/18 19:28:22 connecting evcc-1900239493 at tcp://192.168.0.38:1883
[mqtt  ] DEBUG 2023/10/18 19:28:22 tcp://192.168.0.38:1883 connected
Power:          read tcp 192.168.0.110:51130->192.168.0.225:502: i/o timeout
Energy:         read tcp 192.168.0.110:37928->192.168.0.225:502: i/o timeout
Current L1..L3: read tcp 192.168.0.110:41466->192.168.0.225:502: i/o timeout
Charge status:  read tcp 192.168.0.110:54130->192.168.0.225:502: i/o timeout
Enabled:        read tcp 192.168.0.110:47848->192.168.0.225:502: i/o timeout
Duration:       read tcp 192.168.0.110:35410->192.168.0.225:502: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

0.121.0

StefanSchoof, 2023-10-18:
Klingt ähnlich wie https://github.com/evcc-io/evcc/issues/10101

andig, 2023-10-18:
@mclane wir bräuchten bitte trace log für aktuelle und letzte funktionierende Version.

mclane, 2023-10-19:
Anbei die Traces:
[trace119.txt](https://github.com/evcc-io/evcc/files/13038951/trace119.txt)
[trace121.txt](https://github.com/evcc-io/evcc/files/13038952/trace121.txt)

andig, 2023-10-19:
Die Kommunikation ist identisch:

119:

```
[em-eth] TRACE 2023/10/19 08:29:02 modbus: send 00 01 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2023/10/19 08:29:02 modbus: recv 00 01 00 00 00 04 b4 01 01 00
```

vs 121:

```
[em-eth] TRACE 2023/10/19 08:35:02 modbus: send 00 01 00 00 00 06 01 01 01 90 00 01
[lp-1  ] ERROR 2023/10/19 08:35:12 charger: read tcp 192.168.0.110:51342->192.168.0.225:502: i/o timeout
```

In Deiner Config ist aber keine Slave ID konfiguriert. 

@premultiply ich hatte per Chat gefragt, warum die ID 180 (=B4) aus den Templates entfernt und durch 255 ersetzt wurde. Mit B4 funktioniert es, mit 255 nicht. Könntest Du Dir das bitte anschauen?

mclane, 2023-10-19:
Ich hatte vor ein paar Wochenmal ein Problem mit meiner Server-SSD (Dateisystem korrupt, nicht mehr lauffähig wiederherstellbar) und danach habe ich evcc neu aufgesetzt und dabei die Konfigurationshilfe verwendet. Die entstandene evcc.yaml habe ich dann noch bzgl. der Loadpoint-Konfiguration angepasst. In meiner alten Config (die ich noch auslesen konnte) war der Phoenix noch ohne das Template konfiguriert und dort war tatsächlich die ID auf 180 gesetzt. Dies habe ich nun in der neuen Config ergänzt und damit läuft die 0.121 dann auch. Danke für die Analyse.

PR #10247 (2023-10-21): phoenix-em-eth: autodetect meter (BC)
- Adds full meter autodetection
- Removes all manual meter config properties (breaking change if manually set)
- Fixes fixed modbus id 180 (fixes #10388)

Note: I'm still not sure if reading multiple registers for phase values ​​is possible on this controller. Individual value readings will be retained for the time being.
closed 2023-10-21 by andig