evcc-io/evcc #10395: Seit HA Update auf 2023.10 Startet EVCC ständig neu, Fahrzeigen werden nicht erkannt, überschußladen nicht geregelt.

killerwahl, 2023-10-19:
### Describe the bug

Hallo,
seit update von HA auf 2023.10 läuft EVCC nicht mehr richtig. EVCC startet ständig neu. 
Instaliert sind zwei Wallboxen KEBA und Pulsar Plus, Pulsar Plus per OCPP.
Bei Pulsar Plus werden keine Fahrzeige mehr erkannt,Ladung auch nicht gestartet,schalteich OCPP aus kann ich Ladung manuell Starten. Bei Keba werden Fahrzeigen zwar Erkannt aber z.B. bei 5kw überschuß ladet es nur mit 1.4kw und gehtauch nicht hoch.
Bis zu Update vom HA hat alles wunderbar funktioniert. Update auf 0.121.1 hatleiderauch keine abhilfe geschafft.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistant.local:8123
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
Sponsortoken: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true
[... 250 lines cut ...]
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/10/19 11:16:39 evcc 0.121.1
[main  ] INFO 2023/10/19 11:16:39 using config file: /config/evcc.yaml
[main  ] INFO 2023/10/19 11:16:39 starting ui and api at :7070
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.121.1

killerwahl, 2023-10-19:
Jetzt wurde Fahrzeig an KEBA erkannt, an Pulsar nicht, OCPP Verbindung steht.
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/10/19 11:38:52 evcc 0.121.1
[main  ] INFO 2023/10/19 11:38:52 using config file: /config/evcc.yaml
[main  ] INFO 2023/10/19 11:38:52 starting ui and api at :7070
[ocpp  ] INFO 2023/10/19 11:39:46 charge point connected, registering: 264611
[site  ] WARN 2023/10/19 11:39:53 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2023/10/19 11:39:53 site config:
[site  ] INFO 2023/10/19 11:39:53   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/19 11:39:53     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/10/19 11:39:53     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/19 11:39:53     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/10/19 11:39:53   vehicles:
[site  ] INFO 2023/10/19 11:39:53     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2023/10/19 11:39:53     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/10/19 11:39:53 loadpoint 1:
[lp-1  ] INFO 2023/10/19 11:39:53   mode:        pv
[lp-1  ] INFO 2023/10/19 11:39:53   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/19 11:39:53   meters:      charge ✓
[lp-1  ] INFO 2023/10/19 11:39:53     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2023/10/19 11:39:53 loadpoint 2:
[lp-2  ] INFO 2023/10/19 11:39:53   mode:        pv
[lp-2  ] INFO 2023/10/19 11:39:53   charger:     power ✗ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2023/10/19 11:39:53   meters:      charge ✗
[lp-1  ] DEBUG 2023/10/19 11:39:53 phase timer inactive
[lp-1  ] DEBUG 2023/10/19 11:39:53 pv timer inactive
[lp-1  ] DEBUG 2023/10/19 11:39:53 guard timer inactive
[lp-1  ] INFO 2023/10/19 11:39:53 vehicle updated: unknown -> Zoe
[lp-1  ] DEBUG 2023/10/19 11:39:53 set charge mode: pv
[lp-1  ] DEBUG 2023/10/19 11:39:53 set target soc: 95
[lp-2  ] DEBUG 2023/10/19 11:39:54 phase timer inactive
[lp-2  ] DEBUG 2023/10/19 11:39:54 pv timer inactive
[lp-2  ] DEBUG 2023/10/19 11:39:54 guard timer inactive
[lp-2  ] INFO 2023/10/19 11:39:54 vehicle updated: unknown -> Tucson
[site  ] DEBUG 2023/10/19 11:39:54 ----
[lp-1  ] DEBUG 2023/10/19 11:39:57 charge power: 0W
[lp-2  ] DEBUG 2023/10/19 11:39:57 charge power: 0W
[site  ] DEBUG 2023/10/19 11:39:59 pv power: 195W
[site  ] DEBUG 2023/10/19 11:40:04 battery soc: 11%
[site  ] DEBUG 2023/10/19 11:40:04 battery power: 217W
[site  ] DEBUG 2023/10/19 11:40:06 grid power: 23W
[site  ] DEBUG 2023/10/19 11:40:12 grid currents: [-0.53 0.64 0.74]A
[site  ] DEBUG 2023/10/19 11:40:13 site power: 240W
[lp-1  ] DEBUG 2023/10/19 11:40:21 vehicle odometer: 14165km
[lp-1  ] DEBUG 2023/10/19 11:40:24 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/19 11:40:29 charge total import: 2365.183kWh
[lp-1  ] DEBUG 2023/10/19 11:40:29 charger status: B
[lp-1  ] DEBUG 2023/10/19 11:40:29 charge start energy: 2.37e+03kWh
[lp-1  ] INFO 2023/10/19 11:40:29 car connected
[lp-1  ] DEBUG 2023/10/19 11:40:29 pv timer elapse
[lp-1  ] DEBUG 2023/10/19 11:40:29 pv timer inactive
[lp-1  ] DEBUG 2023/10/19 11:40:29 charger: guard elapse
[lp-1  ] DEBUG 2023/10/19 11:40:29 guard timer inactive
[lp-1  ] DEBUG 2023/10/19 11:40:29 charge total import: 2365.183kWh
[lp-1  ] DEBUG 2023/10/19 11:40:34 vehicle status: B
[lp-1  ] DEBUG 2023/10/19 11:40:34 vehicle soc: 88%
[lp-1  ] DEBUG 2023/10/19 11:40:34 vehicle range: 271km
[lp-1  ] DEBUG 2023/10/19 11:40:34 pv charge current: 0A = 0A + -0.348A (240W @ 3p)
[lp-1  ] DEBUG 2023/10/19 11:40:34 pv enable timer reset
[lp-1  ] DEBUG 2023/10/19 11:40:34 pv timer inactive
[site  ] DEBUG 2023/10/19 11:40:34 ----
[lp-1  ] DEBUG 2023/10/19 11:40:34 charge power: 0W
[lp-2  ] DEBUG 2023/10/19 11:40:34 charge power: 0W
[site  ] DEBUG 2023/10/19 11:40:36 pv power: 212W
[site  ] DEBUG 2023/10/19 11:40:38 battery soc: 10%
[site  ] DEBUG 2023/10/19 11:40:38 battery power: 356W
[site  ] DEBUG 2023/10/19 11:40:39 grid power: 49W
[site  ] DEBUG 2023/10/19 11:40:41 grid currents: [-0.75 -0.69 0.87]A
[site  ] DEBUG 2023/10/19 11:40:42 site power: 405W
[lp-2  ] DEBUG 2023/10/19 11:40:43 vehicle odometer: 25081km

tobiasbayer, 2023-10-19:
Ständige Neustarts bei Nutzung des HA-Addons [hatten wir schon mal](https://github.com/evcc-io/evcc/issues/7446#issuecomment-1508560734). Evtl. wieder die gleiche Ursache?

killerwahl, 2023-10-19:
Ich probiere Watchdog auszuschalten, immoment ist wieder abgesturzt.
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/10/19 11:44:55 evcc 0.121.1
[main  ] INFO 2023/10/19 11:44:55 using config file: /config/evcc.yaml
[main  ] INFO 2023/10/19 11:44:55 starting ui and api at :7070
[ocpp  ] INFO 2023/10/19 11:46:35 charge point connected, registering: 264611
[main  ] FATAL 2023/10/19 11:47:35 cannot create charger 'my_charger_2': cannot create charger 'template': cannot create charger 'ocpp': ocpp message (2824873902): GenericError - Request timed out
[main  ] FATAL 2023/10/19 11:47:35 will attempt restart in: 5m0s

killerwahl, 2023-10-19:
So, jetzt stürzt zwar EVCC nicht mehr ab, aber Regelttrotzdem nichts.
KEBA Fahrzeug erkannt, Laden auf Netzladen umgestelt, Ladung startet trotzdem nicht.
Pulsar Plus Fahrzeug wird immer nocht nicht erkannt.

[lp-2  ] DEBUG 2023/10/19 12:05:03 set charge mode: now
[lp-1  ] DEBUG 2023/10/19 12:05:52 set charge mode: now

guhlandreas, 2023-10-19:
Hallo,
ich habe heute auch ein ähnliches Problem bei mir bemerkt. Ich weiß allerdings nicht genau, seit wann oder nach welchem Update oder Änderung das Problem bei mir aufgetreten ist.
EVCC läuft bei mir im Home Assistant.
Ich verwende die aktuelle HA (2023.10.3) und EVCC (0.121.1) version.

EVCC startet nicht mehr.
Und dem log file kann ich entnehmen:

```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/10/19 20:49:59 evcc 0.121.1
[main  ] INFO 2023/10/19 20:49:59 using config file: /config/evcc.yaml
[main  ] INFO 2023/10/19 20:49:59 starting ui and api at :7070
[db    ] INFO 2023/10/19 20:49:59 using sqlite database: /data/evcc.db
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT * FROM `settings` 0 <nil>
[mqtt  ] INFO 2023/10/19 20:49:59 connecting evcc-605935984 at tcp://localhost:1883
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
```
Hat jemand eine Idee?

Ich bin ratlos und würde mich über Hilfe freuen!

Beste Grüße

Andreas

killerwahl, 2023-10-20:
Andreas, schalte Watchdog aus dann startet EVCC nicht mehr ständig neu. Bei mir läuft zwar EVCC seit gestern ohne neustart aber funktionieren tut es immer noch nicht.
An Pulsat über OCPP wird kein Fahrzeug erkannt und geladen.
An Keba werden zwar Fahrzeugen erkannt aber wenn ich z.B.volle Netzladung auswähle fängt Auto mit nur 1,2 kw anstatt 7,4kw zu Laden, in EVCC wird Ladung aber nicht angezeigt nur in Keba APP.
Letzte 3 Monaten hat alles wunderbar funktiomiert, jetzt haben wir zwar keine Sonne aber das nichts richtig funktioniert ist ärgerlich.

guhlandreas, 2023-10-20:
hallo @killerwahl,
danke für Deine Nachricht. Immerhin scheine ich nicht alleine mit meinem Problem zu sein ;-)
Ja, Watchdog ist aus, mit dem Effekt, dass EVCC nun nach dem start hängen bleibt und nicht läuft.
Mir ist der Fehler rätselhaft und ich weiss garnicht in welche Richtung ich suchen soll. Datenbank?
Viele Grüße
Andreas

killerwahl, 2023-10-20:
@guhlandreas, habe leider auch keine Ahnung. EVCC zeigt mir auch vollig falsche Werten an, die verändern sich seit gestern nicht.
Laut EVCC ist meine Batterie zu 9% voll und vom Dach nur 200W leistung kommen, obwohl Batterie leer ist und vom Dach jetzt 500w kommen.

killerwahl, 2023-10-20:
Update: habe Pulsar Plus aus EVCC gelöscht und EVCC neu gestartet. Jetzt funktioniert mindestens Energie Anzeige richtig. Vor kurzem hat Pulsar Plus auch ein Update bekommen, vielleicht haben die am OCPP Protokol was geändert. Habe beim Pulsar jetzt OCPP ausgeschaltet.
Werde am KEBA gleich testen ob Netzladen geht.

andig, 2023-10-20:
> im moment ist wieder abgesturzt.

Es ist völlig unklar, was das heissen soll. Bitte mal Logfile des konkreten Fehlers, sonst würde ich hier schließen. Wir bearbeiten keine Issues ohne Logfile!

guhlandreas, 2023-10-20:
@andig: Danke für das feedback in meine Richtung.
Unter https://github.com/evcc-io/evcc/issues/10407 habe ich ein neues Issue aufgemacht.
Nun bin ich neugierig auf Deine Idee ;-)

andig, 2023-10-20:
> Update: Fehler lag bei Pulsar Plus und OCPP... Werde Support von EVCC anschreiben

Das hier ist der Support von evcc ;) Ich mach mal zu bis es ein aussagekräftiges Logfile gibt.

closed 2023-10-20 by andig

killerwahl, 2023-10-21:
@andig wenn du mir sagen kannst wo Logfile bei HA versteckt sind kann ich die hier selbstvertändlich posten. Im Moment komme ich nur an Protokol ran. Was mir aufgefallen ist, ist diese Fehlermeldung im Protokol:
[ocpp  ] INFO 2023/10/21 09:05:36 charge point connected, registering: 264611
[site  ] WARN 2023/10/21 09:05:56 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
Wenn ich Pulsar aus EVCC lösche habe ich die Batterie Warnmeldung nicht.

andig, 2023-10-21:
Ich hab leider kein HA. Im Zweifel musst du das ohne testen.

killerwahl, 2023-10-21:
Im Moment läuft 120.03 stabil. Wenn ich mehr Zeit habe installiere ich EVCC auf PC und Teste. 
Schönes und Arbeitsfreies Wochenende.

killerwahl, 2023-10-21:
Nach Update von HA auf 2023.10.4 läuft EVCC auch als 0.121.1 wieder Einwandfrei. Morgen beobachte ich noch, wenn genug Sonne da ist.