evcc-io/evcc #10399: ocpp  FormationViolation - parsing time Error

blauberger, 2023-10-19:
### Describe the bug

Since Version 0.121.1 when a car is connected to the charger the log fills up with:

evcc[609]: [ocpp  ] ERROR 2023/10/19 17:03:59 ocpp message (288115): FormationViolation - parsing time "2023-10-19T15:28:07" as "2006-01-02T15:04:05Z07:00": cannot parse "" as "Z07:00" ([2,"288115","StartTransaction",{"connectorId":1,"idTag":"evcc","meterStart":0,"reservationId":0,"timestamp":"2023-10-19T15:28:07"}])

The charger is a Growatt THOR EV Charger.
It seems as the parser fails at the time zone info "Z07:00"

### Steps to reproduce

1. start evcc
2. wait until charge point is connected
3. log fills up with the error message

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 686862346864f324623874

interval: 30s # control cycle interval

meters:
- name: grid
  type: custom
  power: #Power # default value, optionally override # Leistung (W)
    source: mqtt
    topic: hichi/netzbezug
    timeout: 30s
- name: battery1
  type: custom
  power: # Leistung (W)
    source: mqtt
    topic: growatt/battleistung
    timeout: 30s
    scale: 1000
  soc: # Batterie SOC (%)
    source: mqtt
    topic: growatt/battSOC
    scale: 1
    timeout: 30s
  capacity: 9.8 # optional Batterie Kapazität (kWh)
- name: pv_growatt
  type: custom
  power: # Leistung (W)
    source: mqtt
    topic: growatt/pvleistung
    scale: 1000
    timeout: 30s
  energy: # Energiemenge heute (kWh)
    source: mqtt
    topic: growatt/pvenergieheute
    timeout: 30s

chargers:
- type: template
  template: ocpp
  stationid: XKCV772374673286578
  connector: 1
  connecttimeout: 5m
  timeout: 2m
  getconfiguration: true
  bootnotification: true
  name: wallbox1

vehicles:
- type: template
  template: carwings
  title: Leaf
  icon: car
  user: hg@jhjhjkh.com
  password: secrettop
  vin: 1
  capacity: 30
  phases: 1
  cache: 15m
  mode: pv
  targetSoc: 100
  minCurrent: 6
  maxCurrent: 16
  name: ev3

loadpoints:
- title: Garage
  charger: wallbox1
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: My home
  residualPower: 100
  meters:
    grid: grid
    pv:
    - pv_growatt
    battery:
    - battery1

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.4521 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.00 # [currency]/kWh

mqtt:
  broker: 192.168.100.247:1883
  topic: evcc
  user: jhdhlsf
  password: kjkjgfds
```

### Log details

```text
[main  ] INFO 2023/10/19 17:51:55 evcc 0.121.1
[main  ] INFO 2023/10/19 17:51:55 using config file: /home/pi/evcc.yaml
[main  ] INFO 2023/10/19 17:51:55 starting ui and api at :7070
[db    ] INFO 2023/10/19 17:51:58 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/10/19 17:52:01 connecting evcc-1588265221 at tcp://192.168.10   0.247:1883
[mqtt  ] DEBUG 2023/10/19 17:52:01 tcp://192.168.100.247:1883 connected
[XKCV772374673286578-1] DEBUG 2023/10/19 17:52:01 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2023/10/19 17:52:14 charge point connected: XKCV772374673286578
[ocpp  ] ERROR 2023/10/19 17:52:15 ocpp message (347852): FormationViolation - p   arsing time "2023-10-19T15:28:07" as "2006-01-02T15:04:05Z07:00": cannot parse "   " as "Z07:00" ([2,"347852","StartTransaction",{"connectorId":1,"idTag":"evcc","m   eterStart":0,"reservationId":0,"timestamp":"2023-10-19T15:28:07"}])
[ocpp  ] ERROR 2023/10/19 17:52:15 ocpp message (347854): FormationViolation - p   arsing time "2023-10-19T15:28:07" as "2006-01-02T15:04:05Z07:00": cannot parse "   " as "Z07:00" ([2,"347854","StartTransaction",{"connectorId":1,"idTag":"evcc","m   eterStart":0,"reservationId":0,"timestamp":"2023-10-19T15:28:07"}])
[ocpp  ] ERROR 2023/10/19 17:52:15 ocpp message (347856): FormationViolation - p   arsing time "2023-10-19T15:28:07" as "2006-01-02T15:04:05Z07:00": cannot parse "   " as "Z07:00" ([2,"347856","StartTransaction",{"connectorId":1,"idTag":"evcc","m   eterStart":0,"reservationId":0,"timestamp":"2023-10-19T15:28:07"}])
[ocpp  ] ERROR 2023/10/19 17:52:15 ocpp message (347857): FormationViolation - p   arsing time "2023-10-19T15:28:07" as "2006-01-02T15:04:05Z07:00": cannot parse "   " as "Z07:00" ([2,"347857","StartTransaction",{"connectorId":1,"idTag":"evcc","m   eterStart":0,"reservationId":0,"timestamp":"2023-10-19T15:28:07"}])
[ocpp  ] ERROR 2023/10/19 17:52:15 ocpp message (347858): FormationViolation - p   arsing time "2023-10-19T15:28:07" as "2006-01-02T15:04:05Z07:00": cannot parse "   " as "Z07:00" ([2,"347858","StartTransaction",{"connectorId":1,"idTag":"evcc","m   eterStart":0,"reservationId":0,"timestamp":"2023-10-19T15:28:07"}])
```

### What type of operating system are you running?

Linux

### Version

0.121.1

andig, 2023-10-19:
Da fehlt die Zeitzone im Timestamp. Aus der OCPP Spec (ich finde den "mandatory") Teil grad nicht. Verwendet wird RFC3339:

> For example, a BootNotification response could look like this:
[3,
   "19223201",
   {"status":"Accepted", "currentTime":"2013-02-01T20:53:32.486Z", "heartbeatInterval":300}
]

Man beachte das Z am Ende. Bitte Ticket beim Hersteller aufmachen.

@tvand zur Info. Ein weiterer Fall von "man könnte das ja mal eben anders machen"...

tvand, 2023-10-19:
@andig Ich fürchte, das kannst Du nicht gewinnen. Gerade im Industriebereich werden Spezifikationen oft als Empfehlungen gesehen. In der Branche ist Softwareentwicklung oft eine Nebentätigkeit. Entsprechend sind oft Skills und Qualität. Been there, done that. Noch'n Beispiel: https://developer.easee.cloud/docs/ocpp-smart-charging

andig, 2023-10-19:
Ändert ja nichts daran, dass jemand gezwungen ist, den Schrott zu unterstützen oder zu kaufen. Bei Easee ist mir nix offensichtliches aufgefallen- die haben es ja sogar geschafft, zu dokumentieren, welchen Teil sie nicht haben.

blauberger, 2023-10-20:
Thank you for the very quick reply.
I have one more comment.
In the specs of ocpp 1.6 I can read
..
3.15. Time notations
This section is normative.
Implementations MUST use ISO 8601 date time notation. Message receivers must be able to handle fractional
seconds and time zone offsets (another implementation might use them). Message senders MAY save data
usage by omitting insignificant fractions of seconds.
..
and as mentioned here
[https://en.wikipedia.org/wiki/ISO_8601](https://en.wikipedia.org/wiki/ISO_8601)
..
If the time is in UTC, add a Z directly after the time without a space. Z is the zone designator for the zero UTC offset.
..
and some lines above
..
If no UTC relation information is given with a time representation, the time is assumed to be in local time.
..
Could this not also be understood as that the "Z" or the other time zone strings for example "+02:00" are optional?
Or is this a requirement of evcc to have all chargers and, may be, all other devices at UTC?

premultiply, 2023-10-20:
Die Box sendet hier eine Lokalzeit.
Wo kommt denn dieser Fehler her?

andig, 2023-10-20:
Depends on https://github.com/lorenzodonini/ocpp-go/issues/237

tvand, 2023-10-20:
@andig Naja, Theorie und Praxis, ya know. Meist ist es so, dass man als Kunde zum Kaufzeitpunkt gar keine Chance hat, herauszufinden, was alles nicht geht. Zumal man ja in der Regel nicht mal weiß, wonach man suchen muss.

andig, 2023-10-20:
Daraus lassen sich leider keine Ansprüche gegen Dritte ableiten. Da etwas OT- ich blende die Kommentare mal aus.

andig, 2023-10-20:
Library has a global data format setting that could be used- but only as long as you have a single format for all charge points...

closed 2023-10-20 by andig

andig, 2023-10-31:
@blauberger please retry with tomorrow's nightly. Upstream enhancement has been included. Let's see if that is capable to parse the timestamps and if your box can understand evcc requests.

blauberger, 2023-11-02:
With the nightly build, the box now also works with the time zone set. The error messages have disappeared. Very good work! Thank you very much.