evcc-io/evcc #10405: NRGKick Bluetooth error?

redi78, 2023-10-20:
### Describe the bug

I configured NRGKick charger as follows:

`chargers:
- type: template
  template: nrgkick-bluetooth 
  mac: A8:03:2A:30:B8:66
  pin: 1234
  name: wallbox4`

and get the following error message:

`sudo evcc -l debug charger
[main  ] INFO 2023/10/20 11:16:48 evcc 0.121.0
[main  ] INFO 2023/10/20 11:16:48 using config file: /etc/evcc.yaml
[db    ] INFO 2023/10/20 11:16:49 using sqlite database: /root/.evcc/evcc.db
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
`

Did i something wrong?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: c0372a0548285409713f5052e40dcaf8688b1eba68b04ccb0f350f5db9470610

interval: 10s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.1.119  
  port: 12000  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.1.119
  port: 12000  
  usage: pv  
  modbus: tcpip  
  name: pv2

#meters:
#  - name: my_pv
#    type: template
#    template: huawei-dongle
#    usage: pv

    # Modbus TCP
#    modbus: tcpip
#    id: 1
#    host: 192.168.1.40 # Hostname
#    port: 502 # Port 

chargers:
- type: template
  template: nrgkick-bluetooth 
  mac: A8:03:2A:30:B8:66
  pin: 1234
  name: wallbox4

vehicles:
- type: template
  template: renault 
  title: Zoe  
  user: rene.diem@gmx.at  
  password: '!*3WiHCdqMR4X@'  
  capacity: 50  
  name: ev3

loadpoints:
- title: Carport
  charger: wallbox4
  vehicle: ev3
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
      - pv2
```

### Log details

```text
evcc --log debug
[main  ] INFO 2023/10/20 11:55:40 evcc 0.121.0
[main  ] INFO 2023/10/20 11:55:40 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/20 11:55:40 starting ui and api at :7070
[db    ] INFO 2023/10/20 11:55:41 using sqlite database: /home/redi78/.evcc/evcc.db
[site  ] INFO 2023/10/20 11:55:43 site config:
[site  ] INFO 2023/10/20 11:55:43   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/10/20 11:55:43     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/10/20 11:55:43     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/20 11:55:43   vehicles:
[site  ] INFO 2023/10/20 11:55:43     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2023/10/20 11:55:43 loadpoint 1:
[lp-1  ] INFO 2023/10/20 11:55:43   mode:        pv
[lp-1  ] INFO 2023/10/20 11:55:43   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/20 11:55:43   meters:      charge ✓
[lp-1  ] INFO 2023/10/20 11:55:43     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/10/20 11:55:43 phase timer inactive
[lp-1  ] DEBUG 2023/10/20 11:55:43 pv timer inactive
[lp-1  ] DEBUG 2023/10/20 11:55:43 guard timer inactive
[lp-1  ] INFO 2023/10/20 11:55:43 vehicle updated: unknown -> Zoe
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0003] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0004] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
WARN[0004] Recovering from panic: send on closed channel 
WARN[0004] MapToStruct: invalid field detected *device.Device1Properties.Bonded 
[lp-1  ] ERROR 2023/10/20 11:55:44 charger: Pair failed: Authentication Canceled
[server] FATAL 2023/10/20 11:55:44 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.0

andig, 2023-10-20:
This is typically a mismatch with the BlueZ version. Did this work before in an older evcc release? Could you upgrade your OS?

redi78, 2023-10-20:
no - this is the first time i try to use evcc

andig, 2023-10-20:
Could you please check your bluez version?

redi78, 2023-10-20:
`bluetoothd -v
5.66
`

redi78, 2023-10-20:
seems that manual paring works on terminal

`bluetoothctl pair A8:03:2A:30:B8:66
Attempting to pair with A8:03:2A:30:B8:66
[CHG] Device A8:03:2A:30:B8:66 Connected: yes
[CHG] Device A8:03:2A:30:B8:66 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device A8:03:2A:30:B8:66 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device A8:03:2A:30:B8:66 UUIDs: e792361a-2c8a-4373-b365-e22c18dc1144
[CHG] Device A8:03:2A:30:B8:66 ServicesResolved: yes
[CHG] Device A8:03:2A:30:B8:66 Paired: yes
[NEW] Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_A8_03_2A_30_B8_66/service0006
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_A8_03_2A_30_B8_66/service0006/char0007
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
[NEW] Descriptor (Handle 0x0000)
	/org/bluez/hci0/dev_A8_03_2A_30_B8_66/service0006/char0007/desc0009
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_A8_03_2A_30_B8_66/service000a
	e792361a-2c8a-4373-b365-e22c18dc1144
	Vendor specific
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_A8_03_2A_30_B8_66/service000a/char000b
	fc24f07e-8256-4a18-b0bc-fb1aa922c7de
	Vendor specific
[NEW] Descriptor (Handle 0x0000)
	/org/bluez/hci0/dev_A8_03_2A_30_B8_66/service000a/char000b/desc000d
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_A8_03_2A_30_B8_66/service000a/char000e
	ecbb79c5-6972-47ce-abbe-787a003c4f5d
	Vendor specific
Pairing successful
`

andig, 2023-10-20:
> should i install a previous version?

5.65 should work, see related issue.

redi78, 2023-10-20:
i think i need little help. do you have command for downgrade to this version for raspberry pi os lite?

andig, 2023-10-20:
Closing here until support for 5.66 is merged upstream.

closed 2023-10-20 by andig

redi78, 2023-10-20:
HI, installed bluez Version 5.55 and same error:
 
 `[lp-1  ] INFO 2023/10/20 20:39:27   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/20 20:39:27   meters:      charge ✓
[lp-1  ] INFO 2023/10/20 20:39:27     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/10/20 20:39:27 phase timer inactive
[lp-1  ] DEBUG 2023/10/20 20:39:27 pv timer inactive
[lp-1  ] DEBUG 2023/10/20 20:39:27 guard timer inactive
[lp-1  ] INFO 2023/10/20 20:39:27 vehicle updated: unknown -> Zoe
[lp-1  ] ERROR 2023/10/20 20:39:30 charger: characteristic not found
[site  ] DEBUG 2023/10/20 20:39:30 ----
[lp-1  ] ERROR 2023/10/20 20:39:45 charge meter: characteristic not found
[site  ] DEBUG 2023/10/20 20:39:47 pv power: 0W
[site  ] DEBUG 2023/10/20 20:39:48 grid power: 707W
[site  ] DEBUG 2023/10/20 20:39:50 grid currents: [0.84 0.28 2.97]A
[site  ] DEBUG 2023/10/20 20:39:51 site power: 707W
[lp-1  ] DEBUG 2023/10/20 20:39:51 vehicle odometer: 25833km
WARN[0028] Recovering from panic: send on closed channel 
[lp-1  ] ERROR 2023/10/20 20:39:53 charge meter: characteristic not found
[lp-1  ] ERROR 2023/10/20 20:40:06 charge total import: find device: discovery timeout exceeded
[lp-1  ] ERROR 2023/10/20 20:40:08 charger: characteristic not found
[site  ] DEBUG 2023/10/20 20:40:08 ----
[lp-1  ] ERROR 2023/10/20 20:40:35 charge meter: find device: discovery timeout exceeded
`

premultiply, 2023-10-20:
Das ist eine völlig andere Fehlermeldung als oben!

redi78, 2023-10-20:
`sudo evcc charger --log trace
[main  ] INFO 2023/10/20 21:32:17 evcc 0.121.1
[main  ] INFO 2023/10/20 21:32:17 using config file: /etc/evcc.yaml
[db    ] INFO 2023/10/20 21:32:18 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/10/20 21:32:18 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT * FROM `settings` 6 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2023/10/20 21:32:18 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>

`

rpoisel, 2023-11-01:
How old is the NRGkick cable your are using? Newer models (2022 onwards) are known **not** to work with evcc.

Further info: [Link](https://github.com/evcc-io/evcc/discussions/4048)