evcc-io/evcc #10407: EVCC startet nicht mehr unter Home Assistant

guhlandreas, 2023-10-20:
### Describe the bug

Hallo,
ich habe gerade bemerkt, dass mein EVCC nicht mehr läuft. Wenn ich es (wieder) starte stürzt es ab und bleibt hängen. Ich weiß allerdings nicht genau, seit wann oder nach welchem Update oder Änderung das Problem bei mir aufgetreten ist.

EVCC läuft bei mir im Home Assistant.
Ich verwende die aktuelle HA (2023.10.3) und EVCC (0.121.1) version.

Hat jemand eine Idee?

Ich bin ratlos und würde mich über Hilfe freuen!

Beste Grüße

Andreas

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: https
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.7.xxx
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xxxx

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug #trace
  lp-1: debug #trace
  lp-2: debug #trace
  cache: error #trace
  db: error #trace

[... 210 lines cut ...]
[db    ] TRACE 2023/10/19 20:49:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT * FROM `settings` 0 <nil>
[mqtt  ] INFO 2023/10/19 20:49:59 connecting evcc-605935984 at tcp://localhost:1883
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2023/10/19 20:49:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.121.1

andig, 2023-10-20:
Probier bitte mal den Tibber Pulse neu zu starten- das scheint jetzt mehrfach zu Hängern geführt zu haben.

guhlandreas, 2023-10-20:
@andig: Du bist der Beste!!
1000 Dank! Das war's!
Da wäre ich alleine nie darauf gekommen.
Vielen Dank nochmals für die schnelle Hilfe!
Andreas

andig, 2023-10-20:
Frag mich bitte nicht warum- das ist leider maximal blöde :O.

thebrainkafka, 2023-10-20:
Sorry offtopic @guhlandreas .
Wie rebootet man den tibber pulse?

rene-mueller, 2023-10-20:
> Sorry offtopic @guhlandreas .
> 
> Wie rebootet man den tibber pulse?

Batterie raus und wieder rein

smichaelis01, 2023-10-21:
trotz Neustart vom Puls geht nichts

smichaelis01, 2023-10-21:
Reset WB und Neustart aller Systeme hat geholfen

DANKE

closed 2023-10-23 by andig