evcc-io/evcc #10414: Datenfehler mit Influx

tp49, 2023-10-21:
Hallo zusammen,
seit einigen EVCC updates (habe etwas gewartet, ob Fehler im neuen Update behoben ist) wurde ein Fehler bei der Statusabfrage von evcc in Verbindung mit influx angezeigt. Meine Versionen evcc 0.121.1; InfluxDB v2.7.3
Dieser lautete: "evcc[xxxxxx]: [influx] ERROR 'xxxx.xx.xx xx.xx.xx' write failed (attempts 1): internal error: unexpected error writing points to database: field type conflict"
Es betrifft wohl den Parameter SmartCostType. Wenn ich diesen dann mit dem Data Explorer abfrage erscheint folgende Fehlermeldung: "unsupported input type for mean aggregate: string"
Gibt es eine Lösung?

andig, 2023-10-21:
Guter Hinweis, danke. Es scheint als sollten wir Strings raus filtern.

andig, 2023-10-21:
Closed in https://github.com/evcc-io/evcc/commit/fa0fb907ff588ad323bfa8a3142479b9b562ca37

closed 2023-10-21 by andig