evcc-io/evcc #10430: BYD Atto 3

Meikefer, 2023-10-22:
Könnt ihr den BYD Atto 3 einbinden?

andig, 2023-10-22:
Gibts eine Schnittstelle?

premultiply, 2023-10-22:
Fall das ein reiner Batterieklotz ist (DC) wird der wie immer über den zugehörigen Wechselrichter eingebunden und spielt selbst keine Rolle.
Falls es ein Kombigerät mit integriertem Wechselrichter ist, braucht es wie jeder Wechselrichter eine vorhandene und dokumentierte Schnittstelle, über die man an die Messdaten herankommt.

andig, 2023-10-22:
Das ist ein Auto :)

premultiply, 2023-10-22:
Woher soll ich das wissen? 😅

Meikefer, 2023-10-22:
Naja, "das" Atto würde sich aber auch komisch anhören. Es ist ja auch nicht "das" Golf 😆
Aber klar, bisher ist BYD in Europa hauptsächlich für Speicher bekannt.

Leider finde ich keine Infos zur Schnittstelle 🙁

andig, 2023-10-22:
Ohne ist nix zu machen :(

closed 2023-10-22 by andig

premultiply, 2023-10-22:
Geht bei Autos aber natürlich auch ohne als "Offline Fahrzeug".