evcc-io/evcc #10431: Beginnt zu laden ohne dass die Haus-Batterie voll ist

dennisN75, 2023-10-22:
### Describe the bug

Seit einigen Tagen beginnt die Wallbox das Auto zu laden, obwohl der Ladestand meiner Hausbatterie die 95% noch nicht erreicht hat. Ich habe da eigentlich nichts umgestellt und die ganze Zeit lief das ohne Probleme.
Aber jetzt beginnt das Auto zu laden, obwohl die Sonnenbatterie erst 20% hat.
Was kann ich tun?

### Steps to reproduce

Neuste Version ist installiert

### Configuration details

```yaml
Ich habe einen KIA EV6
Einen Sonnenbatterie und einen g-eCharger
```

### Log details

```text
Hier mal meine config:

meters:
- type: template
  template: sonnenbatterie
  usage: grid
  host: 192.168.2.122
  port: 80
  name: grid1
- type: template
  template: sonnenbatterie
  usage: pv
  host: 192.168.2.122
  port: 80
  name: pv2
- type: template
  template: sonnenbatterie
  usage: battery
  host: 192.168.2.122
  port: 80
  name: battery3

chargers:
- type: template
  template: go-e-v3
  host: 192.168.2.186
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

vehicles:
- name: my_car
  type: template
  template: kia
  title: Kia EV6 # Wird in der Benutzeroberfläche angezeigt (Optional)
  user: dennis@nizard.de # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
  password: Sigula63! # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
  vin: KNAC481AFN5004716 # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (Optional)
  capacity: 77.4 # Akkukapazität in kWh (Optional)
  language: de # 'de' für Deutsch und 'en' für Englisch (Optional)

site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  bufferSoc: 100
  prioritySoc: 95
```

### What type of operating system are you running?

Linux

### Version

0.121.1

GrimmiMeloni, 2023-10-22:
@dennisN75 zur Sicherheit: Welche Werte für bufferSoc und prioritySoc sieht Du in der UI? (Die Werte in der Konfiguration werden nur einmal initial verwendet, danach gilt die UI.)

dennisN75, 2023-10-22:
Ich sehe dort die 100% und die 95%.
Aber im Text wird mir angezeigt, "Fahrzeug zuerst, dann das Haus"

VolkerK62, 2023-10-22:
>Aber im Text wird mir angezeigt, "Fahrzeug zuerst, dann das Haus"

das ist nur die Erklärung für die Symbole

dennisN75, 2023-10-22:
ok danke.

Habe jetzt in evcc auf AUS gestellt. Sobald ich auf PV stelle, beginnt er das Auto zu laden, obwohl die Haus-Batterie erst 30% hat. Keine Ahnung was ich machen kann.

VolkerK62, 2023-10-22:
Nur zur Klarstellung. Auch wenn die Hausbatterie den gewünschte Wert noch nicht erreicht hat, beginnt die Autoladung, wenn über die Hausbatterieladung hinaus noch genug PV-Überschuss (Einspeisung) vorhanden ist.

dennisN75, 2023-10-22:
Ja genau, Das Auto beginnt zu laden auch wenn die Hausbatterie die 95% noch nicht erreicht hat.
Ist der Überschuss so groß, dass die Hausbatterie mit 4kw geladen wird und noch genaug für das Auto übrig bleibt, ist das ja ok.
Aber aktuell ist es so, dass das Laden der Batterie auf beispielsweise 0,8kw runtergefahren wird und das Auto mit voller Leistung geladen wird.
Das möchte ich nicht. Haus und Hausbatterie sollen immer Vorrang haben.

Logfile liefere ich gleich nach

premultiply, 2023-10-22:
`residualPower` richtig gesetzt?

closed 2023-10-22 by premultiply

dennisN75, 2023-10-22:
Hier mal ein Auszug aus dem Log. Hoffe das ist das richtige

Oct 22 11:36:40 homebridge evcc[565]: [lp-1  ] DEBUG 2023/10/22 11:36:40 charge power: 0W
Oct 22 11:36:41 homebridge evcc[565]: [site  ] DEBUG 2023/10/22 11:36:41 pv power: 2199W
Oct 22 11:36:41 homebridge evcc[565]: [site  ] DEBUG 2023/10/22 11:36:41 battery soc: 51%
Oct 22 11:36:41 homebridge evcc[565]: [site  ] DEBUG 2023/10/22 11:36:41 battery power: -1476W
Oct 22 11:36:41 homebridge evcc[565]: [site  ] DEBUG 2023/10/22 11:36:41 grid power: 115W
Oct 22 11:36:41 homebridge evcc[565]: [site  ] DEBUG 2023/10/22 11:36:41 giving priority to battery charging at soc: 51%
Oct 22 11:36:41 homebridge evcc[565]: [site  ] DEBUG 2023/10/22 11:36:41 site power: 115W
Oct 22 11:36:41 homebridge evcc[565]: [lp-1  ] DEBUG 2023/10/22 11:36:41 charge currents: [0 0 0]A
Oct 22 11:36:41 homebridge evcc[565]: [lp-1  ] DEBUG 2023/10/22 11:36:41 charge total import: 332.095kWh
Oct 22 11:36:41 homebridge evcc[565]: [lp-1  ] DEBUG 2023/10/22 11:36:41 charger status: B

GrimmiMeloni, 2023-10-22:
Auszüge helfen nicht. Im Beispiel oben lädt Deine Hausbatterie wie gewünscht.

VolkerK62, 2023-10-22:
wie @premultiply  schon geschrieben hat, ergänze erstmal in der `site` den Eintrag `residualpower: 100`.
https://docs.evcc.io/docs/reference/configuration/site#residualpower
Dann fällt das als mögliche Fehlerquelle weg.

dennisN75, 2023-10-22:
Das hinzufügen von residualpower: 100 scheint es gewesen zu sein. Jetzt hat die Hausbetterie wieder Prio und nur wenn genügend übrig ist, wird das Auto geladen.