evcc-io/evcc #10436: Charged Counter gets stuck when charging is paused over night in PV mode

thomas-wolf, 2023-10-22:
### Describe the bug

I observed multiple times that the charged counter that displays how many kWh have been charged gets stuck when the charging is paused for some hours or overnight.

For example, I started loading my vehicle yesterday in PV mode. Throughout the day, 9.3 kWh have been sent to the car's battery. I did not stop the charging process overnight, but just let evcc manage it. As there was no surplus, the loading stopped at some time yesterday afternoon. Today around noon, when some surplus was available, the charging resumed. Even though the car state of charge display shows progress, the "charged" counter gets stuck at yesterdays 9.3 kWh. 

### Steps to reproduce

1. Start Loading in PV mode
2. After sun, check the charged counter 
3. Wait until there is some surplus available the next day 
4. The charge counter will not be updated

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xyz

interval: 10s # control cycle interval

sponsortoken: token

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solaredge-hybrid
  id: 1
  host: ip
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: solaredge-hybrid
  id: 1
  host: ip
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: solaredge-hybrid
  id: 1
  host: ip
  port: 1502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:
- type: template
  template: go-e-v3
  host: ip
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Home
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  bufferSoc: 100
  prioritySoc: 75
  residualPower: 0   

vehicles:
  - name: born
    type: template
    template: cupra
    title: Cupra Born # Wird in der Benutzeroberfläche angezeigt (Optional)
    user: email # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: pw # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    #vin: W... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (Optional)
    #capacity: 50 # Akkukapazität in kWh (Optional) 
  - name: my_car
    type: template
    template: kia
    title: ev6 # Wird in der Benutzeroberfläche angezeigt (optional)
    user: email # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: pw # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    # vin:# Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 77.4  # Akkukapazität in kWh (optional)
    language: de # 'de' für Deutsch und 'en' für Englisch (optional)
```

### Log details

```text
[site  ] DEBUG 2023/10/22 14:10:53 ----
[lp-1  ] DEBUG 2023/10/22 14:10:53 charge power: 1228W
[site  ] DEBUG 2023/10/22 14:10:53 pv power: 549W
[site  ] DEBUG 2023/10/22 14:10:53 battery soc: 83%
[site  ] DEBUG 2023/10/22 14:10:53 battery power: 1076W
[site  ] DEBUG 2023/10/22 14:10:53 grid power: 24W
[site  ] DEBUG 2023/10/22 14:10:53 site power: 750W
[lp-1  ] DEBUG 2023/10/22 14:10:53 charge currents: [5.55 0 0]A
[lp-1  ] DEBUG 2023/10/22 14:10:53 detected active phases: 1p
[lp-1  ] DEBUG 2023/10/22 14:10:53 charge total import: 622.457kWh
[lp-1  ] DEBUG 2023/10/22 14:10:53 charger status: C
[main  ] DEBUG 2023/10/22 14:10:53 vehicle status: A (Cupra Born)
[lp-1  ] DEBUG 2023/10/22 14:10:53 phase timer inactive
[lp-1  ] DEBUG 2023/10/22 14:10:53 pv charge current: 2.74A = 6A + -3.26A (750W @ 1p)
[lp-1  ] DEBUG 2023/10/22 14:10:53 site power 750W >= 0W disable threshold
[lp-1  ] DEBUG 2023/10/22 14:10:53 pv disable timer start: 3m0s
[lp-1  ] DEBUG 2023/10/22 14:10:53 pv disable in 3m0s
[site  ] DEBUG 2023/10/22 14:11:03 ----
[lp-1  ] DEBUG 2023/10/22 14:11:03 charge power: 1230W
[site  ] DEBUG 2023/10/22 14:11:03 pv power: 542W
[site  ] DEBUG 2023/10/22 14:11:03 battery soc: 83%
[site  ] DEBUG 2023/10/22 14:11:03 battery power: 1075W
[site  ] DEBUG 2023/10/22 14:11:03 grid power: 32W
[site  ] DEBUG 2023/10/22 14:11:03 site power: 757W
[lp-1  ] DEBUG 2023/10/22 14:11:03 charge currents: [5.59 0 0]A
[lp-1  ] DEBUG 2023/10/22 14:11:03 detected active phases: 1p
[lp-1  ] DEBUG 2023/10/22 14:11:03 charge total import: 622.460kWh
[lp-1  ] DEBUG 2023/10/22 14:11:03 charger status: C
[main  ] DEBUG 2023/10/22 14:11:03 vehicle status: A (Cupra Born)
[lp-1  ] DEBUG 2023/10/22 14:11:03 pv charge current: 2.71A = 6A + -3.29A (757W @ 1p)
[lp-1  ] DEBUG 2023/10/22 14:11:03 site power 757W >= 0W disable threshold
[lp-1  ] DEBUG 2023/10/22 14:11:03 pv disable in 2m50s
[lp-1  ] DEBUG 2023/10/22 14:11:03 pv disable timer remaining: 2m50s
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.0

andig, 2023-10-22:
Please reopen and add

    evcc meter

when this happens again.

closed 2023-10-22 by andig

thomas-wolf, 2023-10-23:
pi@phoscon:~ $ evcc meter
[main  ] INFO 2023/10/23 11:30:13 evcc 0.121.1
[main  ] INFO 2023/10/23 11:30:13 using config file: /etc/evcc.yaml
[db    ] INFO 2023/10/23 11:30:14 using sqlite database: /home/pi/.evcc/evcc.db
grid1
-----
Power: 65W

pv2
---
Power: 2451W

battery3
--------
Power: 523W
Soc:   85%