evcc-io/evcc #10444: Nach Update auf 0.121.1 geht nichts mehr

wwshaui, 2023-10-23:
### Describe the bug

Guten Morgen,
habe auf 0.121.1 updatet, jetzt wird nichts mehr hell.
WEB GUI läuft, aber ohne Werte.

Lt. Log sind einige Sachen veraltet!?

Danke für die Hilfe im voraus.

Gruß
Stefan

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
root@orangepiplus2e:~# sudo systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-10-23 06:35:23 CEST; 8min ago
   Main PID: 5371 (evcc)
      Tasks: 9 (limit: 4496)
     Memory: 20.9M
        CPU: 5.387s
     CGroup: /system.slice/evcc.service
             └─5371 /usr/bin/evcc

Oct 23 06:35:28 orangepiplus2e evcc[5371]: [lp-1  ] INFO 2023/10/23 06:35:28   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Oct 23 06:35:28 orangepiplus2e evcc[5371]: [lp-1  ] INFO 2023/10/23 06:35:28   meters:      charge ✓
Oct 23 06:35:28 orangepiplus2e evcc[5371]: [lp-1  ] INFO 2023/10/23 06:35:28     charge:    power ✓ energy ✓ currents ✓
Oct 23 06:35:28 orangepiplus2e evcc[5371]: [site  ] DEBUG 2023/10/23 06:35:28 ----
Oct 23 06:38:44 orangepiplus2e evcc[5371]: [site  ] ERROR 2023/10/23 06:38:44 pv 1 power: read failed: EOF
Oct 23 06:38:44 orangepiplus2e evcc[5371]: [site  ] DEBUG 2023/10/23 06:38:44 pv power: 0W
Oct 23 06:41:59 orangepiplus2e evcc[5371]: [site  ] ERROR 2023/10/23 06:41:59 battery 1 power: read failed: EOF
Oct 23 06:43:04 orangepiplus2e evcc[5371]: [site  ] ERROR 2023/10/23 06:43:04 battery 1 soc: read failed: EOF
Oct 23 06:43:04 orangepiplus2e evcc[5371]: [site  ] DEBUG 2023/10/23 06:43:04 battery soc: 0%
Oct 23 06:43:04 orangepiplus2e evcc[5371]: [site  ] DEBUG 2023/10/23 06:43:04 battery power: 0W
root@orangepiplus2e:
```

### Log details

```text
Oct 23 06:17:08 orangepiplus2e systemd[1]: Started evcc.
Oct 23 06:17:13 orangepiplus2e evcc[843]: [main  ] INFO 2023/10/23 06:17:13 evcc 0.121.1
Oct 23 06:17:13 orangepiplus2e evcc[843]: [main  ] INFO 2023/10/23 06:17:13 using config file: /etc/evcc.yaml
Oct 23 06:17:13 orangepiplus2e evcc[843]: [main  ] INFO 2023/10/23 06:17:13 starting ui and api at :7070
Oct 23 06:17:14 orangepiplus2e evcc[843]: [db    ] INFO 2023/10/23 06:17:14 using sqlite database: /var/lib/evcc/evcc.db
Oct 23 06:17:14 orangepiplus2e evcc[843]: [mqtt  ] INFO 2023/10/23 06:17:14 connecting evcc-1582592606 at tcp://localhost:1883
Oct 23 06:17:16 orangepiplus2e evcc[843]: [lp-1  ] WARN 2023/10/23 06:17:16 vehicles option is deprecated
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] WARN 2023/10/23 06:17:16 deprecated: use 'pv' instead of 'pvs'
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] WARN 2023/10/23 06:17:16 deprecated: use 'battery' instead of 'batteries'
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] INFO 2023/10/23 06:17:16 site config:
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] INFO 2023/10/23 06:17:16   meters:      grid ✓ pv ✓ battery ✓
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] INFO 2023/10/23 06:17:16     grid:      power ✓ energy ✓ currents ✓
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] INFO 2023/10/23 06:17:16     pv 1:      power ✓ energy ✓ currents ✗
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] INFO 2023/10/23 06:17:16     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] INFO 2023/10/23 06:17:16   vehicles:
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] INFO 2023/10/23 06:17:16     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] INFO 2023/10/23 06:17:16     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
Oct 23 06:17:16 orangepiplus2e evcc[843]: [lp-1  ] INFO 2023/10/23 06:17:16 loadpoint 1:
Oct 23 06:17:16 orangepiplus2e evcc[843]: [lp-1  ] INFO 2023/10/23 06:17:16   mode:        pv
Oct 23 06:17:16 orangepiplus2e evcc[843]: [lp-1  ] INFO 2023/10/23 06:17:16   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Oct 23 06:17:16 orangepiplus2e evcc[843]: [lp-1  ] INFO 2023/10/23 06:17:16   meters:      charge ✓
Oct 23 06:17:16 orangepiplus2e evcc[843]: [lp-1  ] INFO 2023/10/23 06:17:16     charge:    power ✓ energy ✓ currents ✓
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] WARN 2023/10/23 06:17:16 interval <30s can lead to unexpected behavior, see https://doc>
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] DEBUG 2023/10/23 06:17:16 ----
Oct 23 06:22:02 orangepiplus2e evcc[843]: [site  ] ERROR 2023/10/23 06:22:02 pv 1 power: read failed: EOF
Oct 23 06:22:02 orangepiplus2e evcc[843]: [site  ] DEBUG 2023/10/23 06:22:02 pv power: 0W
Oct 23 06:25:17 orangepiplus2e evcc[843]: [site  ] ERROR 2023/10/23 06:25:17 battery 1 power: read failed: EOF
Oct 23 06:26:21 orangepiplus2e systemd[1]: Stopping evcc...
Oct 23 06:26:21 orangepiplus2e systemd[1]: evcc.service: Succeeded.
Oct 23 06:26:21 orangepiplus2e systemd[1]: Stopped evcc.
```

### What type of operating system are you running?

Linux

### Version

0.121.1

Sillium007, 2023-10-23:
Moin,

die Frage ob Sachen veraltet sind scheint das Log ja schon zu beantworten:
```
Oct 23 06:17:16 orangepiplus2e evcc[843]: [lp-1  ] WARN 2023/10/23 06:17:16 vehicles option is deprecated
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] WARN 2023/10/23 06:17:16 deprecated: use 'pv' instead of 'pvs'
Oct 23 06:17:16 orangepiplus2e evcc[843]: [site  ] WARN 2023/10/23 06:17:16 deprecated: use 'battery' instead of 'batteries'
```

Und Verbindungen scheinen keine möglich zu sein:
```
Oct 23 06:22:02 orangepiplus2e evcc[843]: [site  ] ERROR 2023/10/23 06:22:02 pv 1 power: read failed: EOF
Oct 23 06:22:02 orangepiplus2e evcc[843]: [site  ] DEBUG 2023/10/23 06:22:02 pv power: 0W
Oct 23 06:25:17 orangepiplus2e evcc[843]: [site  ] ERROR 2023/10/23 06:25:17 battery 1 power: read failed: EOF
```

andig, 2023-10-23:
Config fehlt

closed 2023-10-23 by andig

wwshaui, 2023-10-23:
Ok läuft wieder.
Habe die Benennungen pv und battery geändert, dann lief wieder alles.
Danke und Gruß

Toller Arbeit die Ihr macht!
Hab das System schon einige male weiterempfohlen.