evcc-io/evcc #10446: Dadapower - ignoring inconsistent phases: 1p < 3p observed active

davidwuest, 2023-10-23:
### Describe the bug

Hallo, ich habe auch mit der Aktuellen Version den Fehler: ignoring inconsistent phases: 1p < 3p observed active

Ladestation: Dadapower Premium Wallbox

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
site:
    title: xxxxxxxxx
    meters:
      grid: gridmeter # grid meter reference
      pv: pvmeter
  
chargers:
  - name: ladestation_links
    type: template
    template: dadapower
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 10.12.1.70 # Hostname
    port: 502 # Port 
  - name: ladestation_rechts
    type: template
    template: dadapower
    # Modbus TCP
    modbus: tcpip
    id: 2
    host: 10.12.1.70 # Hostname
    port: 502 # Port 

loadpoints:
  - title: Links # display name for UI
    charger: ladestation_links # Wallbox Referenz
    mode: off # charge mode (off, now, minpv, pv)
  - title: Rechts # display name for UI
    charger: ladestation_rechts # Wallbox Referenz
    mode: off # charge mode (off, now, minpv, pv)

meters:
  - name: gridmeter
    type: template
    template: shelly-pro-3em
    usage: grid
    host: 10.12.1.76 # IP-Adresse oder Hostname
    user: # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.) # Optional
    password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) # Optional 
  - name: pvmeter
    type: template
    template: shelly-pro-3em
    usage: pv
    host: 10.12.1.78 # IP-Adresse oder Hostname
    user: # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.) # Optional
    password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) # Optional 

mqtt:
  broker: mqtt.home:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: xxxxxxxx
  password: xxxxxx

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.374 # [currency]/kWh

vehicles:
  - name: xxxxx
    type: template
    template: id
    title: "xxxxx ID4" # Wird in der Benutzeroberfläche angezeigt (Optional)
    user: xxxxxxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxxxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: xxxxxxxxxxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (Optional)
    capacity: 77 # Akkukapazität in kWh (Optional) 
    identifiers:
      - xxxxxxxx # RFID card
    mode: minpv # enable PV-charging when vehicle is identified
    minSoc: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
    targetSoc: 80 # limit charge to 90%

  - name: xxxxx
    type: template
    template: id
    title: "xxxxxx ID4" # Wird in der Benutzeroberfläche angezeigt (Optional)
    user: xxxxxxxxxxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.).
    password: xxxxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: xxxxxxxxxxxxxxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (Optional)
    capacity: 77 # Akkukapazität in kWh (Optional) 
    mode: minpv # enable PV-charging when vehicle is identified
    minSoc: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
    targetSoc: 80 # limit charge to 90%
    identifiers:
      - xxxxxxxxxxxxxxxx # RFID card
```

### Log details

```text
[lp-1  ] WARN 2023/10/23 09:24:02 ignoring inconsistent phases: 1p < 3p observed active
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.121.1

andig, 2023-10-23:
Anscheinend gibts derzeit einen Bug mit der Firmware- bitte Hersteller kontaktieren.

closed 2023-10-23 by andig

davidwuest, 2023-10-24:
Ich habe Dadapower bereits angeschrieben.

An-Wi, 2023-10-24:
> > charge currents: [5.64 5.64 5.64]A
> 
> Na die Box schickt auf 3 Phasen Ströme! Das kann bei 1p Betrieb ja nicht der Fall sein. Entweder Schütz nicht geschaltet oder Meßwerte stimmen nicht- weiterer Fehler in der Box?

Nein! Das tut sich nicht - hier ein Foto des geeichten Zählers der Wallbox selbst:
[image]

Der anderen Fehler in meiner "Box" ist nur das Problem mit "Ariya: vehicle range: not available" aus einem anderen Issue. Die beiden Fehler "pushen" sich jetzt gegenseitig und machen die Liste in der Meldungen-Box lang und etwas unübersichtlich (Darstellung in diesem Fall könnte man schöner machen, ist aber in meinen Augen kein eigenes Issue wert):
[image]

An-Wi, 2023-10-24:
Ich bin mir ziemlich sicher, dass es vor 2-3 Wochen im Front-End schon mal richtig beim 1-phasigem Laden angezeigt wurde. Aber ich meine, die Fehlermeldung "ignoring inconsistent phases" war parallel dazu auch schon da?!? 🤷‍♂️

@andig Du kannst mir gern mal die Modbus-Beschreibung von DaDaPower per Mail schicken - dann kann ich mich besser in das Trace-Log eindenken. Danke.

An-Wi, 2023-10-25:
@andig Hatte leider das 3p-Trace-Log gestern vergessen, hier einzufügen.
[evcc_charger_trace__3p.txt](https://github.com/evcc-io/evcc/files/13163284/evcc_charger_trace__3p.txt)

Da mir seit ein paar Minuten die Dadapower-Modbus-Register-Beschreibung vorliegt, sieht es für mich so aus, dass momentan immer nur L1current (Reg. 1006) abgefragt wird, egal ob 1p oder 3p eingestellt ist. 
Warum werden Reg. 1007 (L2current) und 1008 (L3current) nicht ausgelesen? Ein Bug?

An-Wi, 2023-10-25:
dadapower.go - Codezeile 200 ist relativ neu uns sieht "unlogisch" aus:
`res[i] = float64(binary.BigEndian.Uint16(b)) / 100`
**b** wird in der Schleife nicht mit **i** inkrementiert!?!

andig, 2023-10-25:
Autsch! Moment…

An-Wi, 2023-10-25:
> Autsch! Moment… fixed!

👍 Sieht jetzt erstmal logischer aus. Teste es dann mit dem nächsten Release.

-----
Hab nun auch Feedback vom DadaPower-Techniker: Problem wird morgen intern geprüft.

Meine _Erkenntnisse_ nach einigen polls mit einem Modbus-Scanner - scheinbar liegt ein grundlegendes Problem in der Modbus-Registerabfrage bei DadaPower vor...
Fragt man die Register 1007 und 1008 einzeln ab (sie wie es bis zum release 0.120.1(?) noch gemacht wurde), dann liefert der response richtige Werte. Fragt man die Register 1006-1008 in einem request ab, dann liefert nur das 1. Register "neue" Wert. Ab Byte 3 des Datenteils werden immer die "alten" Werte (aus den letzten Einzelabfragen) zurückgeliefert... sehr "strange"!

Das ist auch der Grund, warum es bis vor Kurzem in evcc noch funktioniert hat, als im Code noch einzelne requests implementiert waren.
Ich würde dieses Problem jetzt erstmal hier auf Eis legen - sobald die DadaPower-Modbus-Schnittstelle "erwartungsgemäß" funktioniert, sollte auch der neue Code reibungslos arbeiten.

An-Wi, 2023-11-08:
Ein neues Firmware-Update von Dadapower ist heute auf meine Wallbox gekommen: **1.2.7**
Damit ist nun bei 3-phasigem Laden die Anzeige der Werte pro Phase wieder korrekt:
`Current: L1..L3: 15.2A 15.1A 15.2A`

Dieses Problem ist damit abgeschlossen.

---

Zwei soeben entdeckte Logeinträge geben mir allerdings noch Rätsel in dieser Thematik auf:
```
[lp-1  ] DEBUG 2023/11/08 19:42:18 !! scalePhasesIfAvailable: 3p -> 3p
[lp-1  ] DEBUG 2023/11/08 19:42:18 !! scalePhases: GetPhases 3p <> 3p wanted
```
Die Einträge sehe ich, seit ich auf Lademodus "Schnell" umgestellt habe.
Phaseneinstellung steht bei mir aktuell auf "3-phasig" (nicht "automatischer Wechsel").

Hier nochmal der vollständige Logauszug eines Rundlaufs:
```
[site  ] DEBUG 2023/11/08 19:42:18 ----
[lp-1  ] DEBUG 2023/11/08 19:42:18 charge power: 10807W
[site  ] DEBUG 2023/11/08 19:42:18 pv power: 0W
[site  ] DEBUG 2023/11/08 19:42:18 grid power: 11570W
[site  ] DEBUG 2023/11/08 19:42:18 grid currents: [16.1 15.4 16.9]A
[site  ] DEBUG 2023/11/08 19:42:18 site power: 11370W
[lp-1  ] DEBUG 2023/11/08 19:42:18 charge currents: [15.2 15.1 15.2]A
[lp-1  ] DEBUG 2023/11/08 19:42:18 detected active phases: 3p
[lp-1  ] DEBUG 2023/11/08 19:42:18 charge total import: 1076.908kWh
[lp-1  ] DEBUG 2023/11/08 19:42:18 charger status: C
[lp-1  ] DEBUG 2023/11/08 19:42:18 soc estimated: 86.74% (vehicle: 86.00%)
[lp-1  ] DEBUG 2023/11/08 19:42:18 vehicle soc: 87%
[lp-1  ] ERROR 2023/11/08 19:42:18 vehicle range: not available
[lp-1  ] DEBUG 2023/11/08 19:42:18 !! scalePhasesIfAvailable: 3p -> 3p
[lp-1  ] DEBUG 2023/11/08 19:42:18 !! scalePhases: GetPhases 3p <> 3p wanted
[site  ] DEBUG 2023/11/08 19:42:38 ----
```

Was sagen diese beiden Logeinträge mit "!!" aus?