evcc-io/evcc #10448: Vestel WB Fehlermeldung "charger out of sync: expected disabled, got enabled" nach Update auf EVCC 0.121.1

ogheinz77, 2023-10-23:
### Discussed in https://github.com/evcc-io/evcc/discussions/10409

<div type='discussions-op-text'>

<sup>Originally posted by **ogheinz77** October 20, 2023</sup>
Hallo,

ich bekomme seit dem Update auf EVCC 0.121.1 wieder folgenden Fehler:

Oct 20 17:09:21 evcc evcc[11928]: [lp-1  ] WARN 2023/10/20 17:09:21 charger out of sync: expected disabled, got enabled
Oct 20 17:11:09 evcc evcc[11928]: [lp-1  ] WARN 2023/10/20 17:11:09 charger out of sync: expected disabled, got enabled
Oct 20 17:12:57 evcc evcc[11928]: [lp-1  ] WARN 2023/10/20 17:12:57 charger out of sync: expected disabled, got enabled

Die Meldungen kommen vorwiegend wenn kein Auto dran ist.

Hatte den Fehler schon mal, war dann allerdings nicht mehr da als ich EVCC Intervall auf 18s gestellt habe.
Heute habe ich auf EVCC 0.121.1 aktualisiert und der Fehler kommt alle 2 min und müllt mir leider das Log voll.
Mit dem EVCC Intervall weiter runter zugehen bringt kein Erfolg. 
Bei der EVCC 0.121.0 habe den Fehler auch.

**Wenn ich ein Downgrade auf EVCC 0.120.03 mache bekomme ich den Fehler nicht!**

Für die Vestel WB gibt es zurzeit keine neue Firmware.

Info:
Vestel Firmware: v3.118.0
EVCC: 0.121.1

Hatte das auch schon hier rein geschrieben:
(https://github.com/evcc-io/evcc/discussions/5920)

</div>

andig, 2023-10-23:
Logfile fehlt

closed 2023-10-23 by andig

ogheinz77, 2023-10-23:
Liefere ich nach

LE4711, 2023-10-26:
Bei mir tritt der Fehler auch auf seit EVCC 0.120.2 (vorher 0.120.0)

LE4711, 2023-10-26:
[image]

LE4711, 2023-10-26:
Nach evcc-Restart:
[image]

andig, 2023-10-26:
Leider nicht aussagekräftig. Bitte 

    vestel: trace

und dann Log für alte und neue Version.

LE4711, 2023-10-26:
Ich habe mir die Logs nochmals angesehen:
bis einschl. 0.120.1 gab es die Meldungen nicht.

Ich bin dann von 0.120.1 auf 0.121.0 (und später auf 0.121.2) und seit 0.121.0 kommen die Meldungen, wenn das Auto nicht angeschlossen ist

Ich habe jetzt einen downgrade auf **0.120.3** gemacht und damit sind die Meldungen weg

andig, 2023-10-26:
> Bei mir tritt der Fehler auch auf seit EVCC 0.121.2 (vorher 0.121.0)

Sorry, mit dem Durcheinander kann ich nichts anfagen. Logfile beider Versionen, sonst gehts hier leider nicht weiter.

LE4711, 2023-10-26:
[evcc_0121_2.log](https://github.com/evcc-io/evcc/files/13179844/evcc_0121_2.log)
[evcc_0120_3.log](https://github.com/evcc-io/evcc/files/13179845/evcc_0120_3.log)

andig, 2023-10-26:
Beides Logs ohne Fehler. Es tut mir leid, aber für solche Suchspiele fehlt mir die Zeit.

LE4711, 2023-10-26:
?????
[image]

Oct 26 16:52:32 raspberrypi evcc[21796]: [lp-1  ] WARN 2023/10/26 16:52:32 charger out of sync: expected disabled, got enabled

LE4711, 2023-10-26:
Ich habe übrigens wieder einen downgrade auf 0.120.3 gemacht