evcc-io/evcc #10452: Charged Counter gets stuck when charging is paused over night in PV mode

thomas-wolf, 2023-10-23:
### Describe the bug

I observed multiple times that the charged counter that displays how many kWh have been charged gets stuck when the charging is paused for some hours or overnight.

For example, I started loading my vehicle yesterday in PV mode. Throughout the day, 9.3 kWh have been sent to the car's battery. I did not stop the charging process overnight, but just let evcc manage it. As there was no surplus, the loading stopped at some time yesterday afternoon. Today around noon, when some surplus was available, the charging resumed. Even though the car state of charge display shows progress, the "charged" counter gets stuck at yesterdays 9.3 kWh.

The screenshot below shows inconsistent numbers:

[image]

### Steps to reproduce

1. Start Loading in PV mode
2. After sunset, check the charged counter
3. Wait until there is some surplus available the next day
4. The charge counter will not be updated

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xyz

interval: 10s # control cycle interval

sponsortoken: token

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solaredge-hybrid
  id: 1
  host: ip
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: solaredge-hybrid
  id: 1
  host: ip
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: solaredge-hybrid
  id: 1
  host: ip
  port: 1502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:
- type: template
  template: go-e-v3
  host: ip
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Home
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  bufferSoc: 100
  prioritySoc: 75
  residualPower: 0   

vehicles:
  - name: born
    type: template
    template: cupra
    title: Cupra Born # Wird in der Benutzeroberfläche angezeigt (Optional)
    user: email # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: pw # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    #vin: W... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (Optional)
    #capacity: 50 # Akkukapazität in kWh (Optional) 
  - name: my_car
    type: template
    template: kia
    title: ev6 # Wird in der Benutzeroberfläche angezeigt (optional)
    user: email # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: pw # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    # vin:# Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 77.4  # Akkukapazität in kWh (optional)
    language: de # 'de' für Deutsch und 'en' für Englisch (optional)
```

### Log details

```text
evcc meter
[main  ] INFO 2023/10/23 11:30:13 evcc 0.121.1
[main  ] INFO 2023/10/23 11:30:13 using config file: /etc/evcc.yaml
[db    ] INFO 2023/10/23 11:30:14 using sqlite database: /home/pi/.evcc/evcc.db
grid1
-----
Power: 65W

pv2
---
Power: 2451W

battery3
--------
Power: 523W
Soc:   85%
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.1

andig, 2023-10-23:
Wie passt deine Beschreibung zum Screenshot?

thomas-wolf, 2023-10-23:
Schau Dir den Zählerstand an: da sind in diesem Ladevorgang ca. 20 kWh ins Fahrzeug geflossen. Der Geladen-Zähler zeigt hingegen 10,4 kWh an. Die Ladeverluste sind glücklicherweise keine 50% - der App des Fahrzeug zufolge sind die knapp 20 kWh tatsächlich ins Fahrzeug geflossen.

andig, 2023-10-23:
Die 9,3kW aus Deinem Post haben mich verwirrt- eigentlich sind die für die Problembeschreibung irrelevant. Bitte vollständiges Logfile mit

    evcc —log debug,db:trace

Damit wir sehen was da für Werte kommen. Evcc meter zeigt leider nicht die Werte des Ladezählers :(

thomas-wolf, 2023-10-23:
ok, habe das Log Level entsprechend gesetzt und melde mich mit dem Log File, wenn es wieder auftritt.

thomas-wolf, 2023-10-23:
gerade ist der charged_kwh Zähler auf 306 Wh stehen geblieben. Hierzu folgende Logs - ich kenne natürlich nicht die Interna, aber mir fällt auf, dass meter_end_kwh beim Update erhöht wird, der Wert für charged_kwh aber konstant bleibt:

Oct 23 13:23:52 phoscon evcc[24934]: [db    ] TRACE 2023/10/23 13:23:52 UPDATE `sessions` SET `created`="2023-10-23 12:54:02.258",`finished`="2023-10-23 13:23:52.295",`loadpoint`="Garage",`identifier`="",`vehicle`="ev6",`odometer`=1005.9,`meter_start_kwh`=631.51,`meter_end_kwh`=632.325,`charged_kwh`=0.30559599999999953,`charge_duration`="24m40s",`solar_percentage`=98.11232159489889,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 72 1 <nil>
Oct 23 13:43:22 phoscon evcc[24934]: [db    ] TRACE 2023/10/23 13:43:22 UPDATE `sessions` SET `created`="2023-10-23 12:54:02.258",`finished`="2023-10-23 13:23:52.295",`loadpoint`="Garage",`identifier`="",`vehicle`="ev6",`odometer`=1005.9,`meter_start_kwh`=631.51,`meter_end_kwh`=632.325,`charged_kwh`=0.30559599999999953,`charge_duration`="24m40s",`solar_percentage`=98.11232159489889,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 72 1 <nil>
Oct 23 13:51:02 phoscon evcc[24934]: [db    ] TRACE 2023/10/23 13:51:02 UPDATE `sessions` SET `created`="2023-10-23 12:54:02.258",`finished`="2023-10-23 13:51:02.271",`loadpoint`="Garage",`identifier`="",`vehicle`="ev6",`odometer`=1005.9,`meter_start_kwh`=631.51,`meter_end_kwh`=632.603,`charged_kwh`=0.30559599999999953,`charge_duration`="32m20s",`solar_percentage`=98.11232159489889,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 72 1 <nil>
Oct 23 13:51:12 phoscon evcc[24934]: [db    ] TRACE 2023/10/23 13:51:12 UPDATE `sessions` SET `created`="2023-10-23 12:54:02.258",`finished`="2023-10-23 13:51:02.271",`loadpoint`="Garage",`identifier`="",`vehicle`="ev6",`odometer`=1005.9,`meter_start_kwh`=631.51,`meter_end_kwh`=632.603,`charged_kwh`=0.30559599999999953,`charge_duration`="32m20s",`solar_percentage`=98.11232159489889,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 72 1 <nil>

andig, 2023-10-23:
Bitte das komplette Log, nur Bruckstücke helfen uns nicht. Ich mache wieder auf wenn aussagekräftiges Log vorliegt.

thomas-wolf, 2023-10-23:
`Oct 23 14:01:02 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:02 ----
Oct 23 14:01:02 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:02 charge power: 0W
Oct 23 14:01:02 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:02 pv power: 3836W
Oct 23 14:01:02 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:02 battery soc: 91%
Oct 23 14:01:02 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:02 battery power: -3015W
Oct 23 14:01:02 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:02 grid power: -522W
Oct 23 14:01:02 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:02 site power: -3537W
Oct 23 14:01:02 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:02 charge currents: [0 0 0]A
Oct 23 14:01:02 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:02 charge total import: 633.420kWh
Oct 23 14:01:02 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:02 charger status: B
Oct 23 14:01:02 phoscon evcc[24934]: [lp-1  ] INFO 2023/10/23 14:01:02 stop charging <-
Oct 23 14:01:02 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:02 wake-up timer: start
Oct 23 14:01:02 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:02 pv timer reset
Oct 23 14:01:02 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:02 pv timer inactive
Oct 23 14:01:02 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:02 charge total import: 633.420kWh
Oct 23 14:01:02 phoscon evcc[24934]: [db    ] TRACE 2023/10/23 14:01:02 UPDATE `sessions` SET `created`="2023-10-23 12:54:02.258",`finished`="2023-10-23 14:01:02.191",`loadpoint`="Garage",`identifier`="",`vehicle`="ev6",`odometer`=1005.9,`meter_start_kwh`=631.51,`meter_end_kwh`=633.42,`charged_kwh`=0.30559599999999953,`charge_duration`="42m10s",`solar_percentage`=98.11232159489889,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 72 1 <nil>
Oct 23 14:01:07 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:07 vehicle soc: 56%
Oct 23 14:01:07 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:07 vehicle soc limit: 100%
Oct 23 14:01:07 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:07 vehicle range: 267km
Oct 23 14:01:07 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:07 pv charge current: 15.4A = 0A + 15.4A (-3537W @ 1p)
Oct 23 14:01:07 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:07 site power -3537W <= 0W enable threshold
Oct 23 14:01:07 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:07 pv enable timer start: 1m0s
Oct 23 14:01:07 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:07 pv enable in 1m0s
Oct 23 14:01:12 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:12 ----
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 charge power: -0W
Oct 23 14:01:12 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:12 pv power: 4202W
Oct 23 14:01:12 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:12 battery soc: 91%
Oct 23 14:01:12 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:12 battery power: -3796W
Oct 23 14:01:12 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:12 grid power: -94W
Oct 23 14:01:12 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:12 site power: -3890W
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 charge currents: [0 0 0]A
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 charge total import: 633.420kWh
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 charger status: C
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] INFO 2023/10/23 14:01:12 start charging ->
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 wake-up timer: stop
Oct 23 14:01:12 phoscon evcc[24934]: [db    ] TRACE 2023/10/23 14:01:12 UPDATE `sessions` SET `created`="2023-10-23 12:54:02.258",`finished`="2023-10-23 14:01:02.191",`loadpoint`="Garage",`identifier`="",`vehicle`="ev6",`odometer`=1005.9,`meter_start_kwh`=631.51,`meter_end_kwh`=633.42,`charged_kwh`=0.30559599999999953,`charge_duration`="42m10s",`solar_percentage`=98.11232159489889,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 72 1 <nil>
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 vehicle soc: 56%
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 vehicle soc limit: 100%
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 vehicle range: 267km
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 pv charge current: 18.9A = 2A + 16.9A (-3890W @ 1p)
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 pv timer reset
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 pv timer inactive
Oct 23 14:01:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:12 max charge current: 16A
Oct 23 14:01:22 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:22 ----
Oct 23 14:01:22 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:22 charge power: 3517W
Oct 23 14:01:22 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:22 pv power: 4095W
Oct 23 14:01:22 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:22 battery soc: 91%
Oct 23 14:01:22 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:22 battery power: -1284W
Oct 23 14:01:22 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:22 grid power: 1067W
Oct 23 14:01:22 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:01:22 site power: -217W
Oct 23 14:01:22 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:01:22 charge currents: [15.5 0 0]A
[... cut ...]
Oct 23 14:22:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:22:12 vehicle range: 274km
Oct 23 14:22:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:22:12 pv charge current: 13A = 14A + -1A (231W @ 1p)
Oct 23 14:22:12 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:22:12 max charge current: 12A
Oct 23 14:22:22 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:22:22 ----
Oct 23 14:22:22 phoscon evcc[24934]: [lp-1  ] DEBUG 2023/10/23 14:22:22 charge power: 2587W
Oct 23 14:22:22 phoscon evcc[24934]: [site  ] DEBUG 2023/10/23 14:22:22 pv power: 33

andig, 2023-10-23:
Das ist jetzt nicht das Log zum Fall von oben? Wo wäre hier der Fehler?

thomas-wolf, 2023-10-23:
wo der Fehler steckt, kann ich Dir nicht sagen. Aber schau Dir mal den Eintrag um 14:06:52 an:

Oct 23 14:06:52 phoscon evcc[24934]: [db    ] TRACE 2023/10/23 14:06:52 UPDATE `sessions` SET `created`="2023-10-23 12:54:02.258",`finished`="2023-10-23 14:06:52.193",`loadpoint`="Garage",`identifier`="",`vehicle`="ev6",`odometer`=1005.9,`meter_start_kwh`=631.51,`meter_end_kwh`=633.75,`charged_kwh`=0.30559599999999953,`charge_duration`="47m40s",`solar_percentage`=98.11232159489889,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 72 1 <nil>

Meter End - Meter Start >= 2 kWh, charged_kwh bleibt bei ca. 0.31 , was auch der Wert ist aus dem Screenshot aus einem meiner letzten Kommentare. Ich kenne mich natürlich nicht mit der Applikationslogik aus, aber hätte hier nicht ein höherer charged_kwh -Wert in die DB geschrieben werden müssen?

andig, 2023-10-23:
Wenn die Session endet (was sie hier tut) machen wir das:

	meterStop := lp.chargeMeterTotal()
	if meterStop > 0 {
		s.MeterStop = &meterStop
	}

	if chargedEnergy := lp.getChargedEnergy() / 1e3; chargedEnergy > s.ChargedEnergy {
		lp.sessionEnergy.Update(chargedEnergy)
	}

`getChargedEnergy` liest 

	return lp.sessionEnergy.TotalWh()

@naltatis hier bin ich ein bisschen irritiert. Hier hätte ich eigentlich einen check auf `api.ChargeRater` erwartet, es passiert ab etwas anderes, ggf. aufgrund https://github.com/evcc-io/evcc/pull/7380. Könntest Du da mal einen Blick rein werfen?

naltatis, 2023-10-28:
Meine Vermutung ist, dass das mit dem Go-e Fix hier zusammenhängt:

https://github.com/evcc-io/evcc/blob/c856da153f8c5b165cdc46a22d06cb6f2ca7e4b5/core/loadpoint.go#L1346-L1350

Das, was vor dem Schreiben in die DB gemacht wird enthält den ` - lp.chargedAtStartup` Workaround nicht.

```
if chargedEnergy := lp.getChargedEnergy() / 1e3; chargedEnergy > s.ChargedEnergy {
	lp.sessionEnergy.Update(chargedEnergy)
}
```

Ich glaube, dass wir diesen Code aber sowieso nicht (mehr) brauchen.

naltatis, 2023-10-28:
Ganz Durchblicke ich den Go-e Workaround https://github.com/evcc-io/evcc/issues/5092 noch nicht. Aber hier ist ja auch dieser Charger im Spiel.

naltatis, 2023-10-28:
@thomas-wolf Kannst du das `trace` mal auf die ganze Anwendung ausweiten? Interessant wären hier die `sessionEnergy ` Einträge:

```
[cache ] TRACE 2023/10/28 15:08:06 lp-1/sessionEnergy: 325.0676617611556
```

Sieht ja so aus, als ob die nicht wieder aktualisiert werden.

thomas-wolf, 2023-10-28:
bin die nächsten Tage unterwegs und habe kein VPN, d.h. ich habe erst gegen Samstag / Sonntag wieder Zugriff auf evcc, dann aber gerne.

Ein Freund, der nicht ganz zufällig eine sehr ähnliche Konfiguration hat - selbe Wallbox, selbe evcc-Version - meinte, er sehe auch in der go-e App dasselbe Problem, so dass die Ursache vielleicht in einem der letzten Firmwareupdates zu suchen ist... wir haben beide die Firmware v 055.7 installiert.

thomas-wolf, 2023-10-28:
hier ein Datenexport von der WB für den fraglichen Zeitraum , der zu den von mir zuletzt geposteten Logs passt - das sieht mMn valide aus, ist aber natürlich eine retrospektive Betrachtung, die möglicherweise von go-e schon konsolidiert wurde:

Start TS | End TS | Dauer gesamt | Ladezeit | Max. Leistung | Max. Strom | Energie | Zählerstand Start | Zählerstand Ende
20.10.2023 17:35:23 | 24.10.2023 09:45:15 | 3d 16:09:52 | 13:16:01 | 8,259 kW | 15,9 A | 33,939 kWh | 602,809 kWh | 636,748 kWh

andig, 2023-10-30:
Ich habs mal lesbar formatiert...

premultiply, 2023-11-01:
Wollen wir chargedEnergy nicht einfach komplett ignorieren wenn wir richtige Zählerstände haben?

Eine Alternative wäre immer den größeren Wert in der Art ChargedEnergy := max("TotalJetzt"- "TotalStart", api.ChargedEnergy) zu nehmen.
Das würde auch unterschiedliche Genauigkeiten, Fehlimplementierungen (Ladephase statt Session), verspätete Updates (Easee) automatisch ausgleichen.

andig, 2023-11-01:
> Das würde auch unterschiedliche Genauigkeiten, Fehlimplementierungen (Ladephase statt Session), verspätete Updates (Easee) automatisch ausgleichen.

Eigentlich nicht- denn dank "max" passen die Werte weiter nicht zusammen. Wir vergleichen uns ja _auch_ immer mit irgendwelchen Herstellerapps. Ich würde zuerst gerne mal verstehen, welche der Zahlen denn die "richtige" wäre. Das erscheint mit hier noch nicht klar.

premultiply, 2023-11-01:
Der Zählerstand.
Die geladene Energie des Chargers kann jederzeit zurückgesetzt werden.
Hier wird dann nicht die Energie der Session sauber über mehrere Phasen zwischen an- und abstecken addiert sondern beliebig zwischendrin zurückgesetzt. Am Anfang jeder Ladephase oder beim Tageswechsel etc.

Oben genannte Formel sorgt dabei für stets korrekte Daten, denn fehlerhafte Werte sind stets zu klein.

andig, 2023-11-01:
D.h. dein Vorschlag wäre also Änderung ohne hier weiter root cause Analyse zu machen? Für mich auch ok :)

naltatis, 2023-11-01:
> verspätete Updates (Easee) automatisch ausgleichen.

Das Problem würde Umstellen auf Zähler leider nicht lösen. Der Zählerstand kommt bei Easee genau so unregelmäßig wie die Sessionenergie.

> Wollen wir chargedEnergy nicht einfach komplett ignorieren wenn wir richtige Zählerstände haben?

Find ich generell, ein gutes Vorgehen:

1. Wahl: Zählerdifferenz
2. Wahl: Sessionenergie vom Charger
3. Wahl: Interpolation der Leistung

Wichtig ist nur, dass wir weiterhin regelmäßige Inkremente bekommen (`lp.chargeRater.ChargedEnergy()`) um bspw. Anteil Sonnenenergie und CO2/Preisberechnung sauber machen zu können.

andig, 2023-11-01:
@naltatis was meinst Du mit Wahl genau? Wenn es Zähler gibt den Wert nehmen?

Verstehen wir denn was im Fall oben das Problem ist?

premultiply, 2023-11-01:
> Das Problem würde Umstellen auf Zähler leider nicht lösen. Der Zählerstand kommt bei Easee genau so unregelmäßig wie die Sessionenergie.

Jain. Durch die Formel oben wird aber die Updaterate verdoppelt und somit die Chance maximiert trotz vereinzelt fehlender Updates einen korrekten aktuellen/letzten Zählerstand zu erhalten.

Man weiß ja bei den einzelnen Chargern nie wo der Hund begraben liegt. Irgendwo haben sie alle ihre Macken und wir können nur aus den angebotenen Daten möglichst Präzise ableiten.

Die Fehler entstehen hier eigentlich nur dadurch dass wir stellenweise den Werten vom Charger zu viel vertrauen.
Wenn wir das um ein paar Logikprüfungen erweitern sollte das nahezu narrensicher werden. Quasi die Rohdaten erstmal durch einen "Legalizer" jagen und gegeineinander (soweit möglich) auf Plausibilität prüfen. ;)

premultiply, 2023-11-01:
> Verstehen wir denn was im Fall oben das Problem ist?

ChargedEnergy wird nicht aktualisiert oder zurückgesetzt.

andig, 2023-11-01:
@premultiply dann passt Dein Vorschlag, also 1+2 aus https://github.com/evcc-io/evcc/issues/10452#issuecomment-1788760856. Sollten wir ausprobieren.

naltatis, 2023-11-01:
> also 1+2 aus https://github.com/evcc-io/evcc/issues/10452#issuecomment-1788760856. Sollten wir ausprobieren.

Ja, 3. gibt es ja heute schon und würde unverändert bleiben. -> `chargerater.go`

andig, 2023-11-01:
Siehe auch https://github.com/evcc-io/evcc/discussions/10239, da wird Phasenumschaltung als Ursache vermutet...

dubtec, 2023-11-02:
Hallo zusammen,

Ich bin mir nicht sicher, ob eine „Ich auch!“-Meldung erwünscht, aber ich beobachte den selben Fehler: Zählerstand unterscheidet sich von der
Lademenge. Auch bei mir ist die Differenz der Zählerstände der vermeintlich korrekte Wert. 

[image]

Zu meinem Set-up:
- go-e Gemini flex
- Installation in docker
- Phasenumschaltung aktiviert
- Version 0.121.5
- PV-Überschussladen als Standard

Wenn mich nicht alles täuscht, war die Lademenge in einer Vorgänger-Version noch richtig. Ich meine, dass ich 0.121.4 davor hatte.

andig, 2023-11-02:
> Ich bin mir nicht sicher, ob eine „Ich auch!“-Meldung erwünscht

Sie hilft nicht ;). Hilfreich könnte sein, rauszufinden ob die Phasenumschaltung wirklich die Ursache ist. Dafür im Logfile auf Aktualisierung der Ladeenergie achten. ...und im Zweifel Ticket bei Go-E aufmachen.

thomas-wolf, 2023-11-05:
Habe das Log Level auf Trace geändert, aber heute gab es leider zu wenig Sonne, um auf 3p zu switchen. Morgen soll es besser werden, das warte ich mal noch ab. Falls die Sonnenstunden nicht ausreichen: Ich denke, es geht ja vor Allem um die Ergebnisse des Calls gegen die go-e API nach einem Phasenwechsel, d.h. ich könnte die Situation mit gleicher Aussagekraft auch im Schnelllademodus durch einen erzwungenen Phasenswitch über die UI provozieren, richtig?

thomas-wolf, 2023-11-11:
Anfangs der Woche gab es einige Phasenwechsel, aber der charged counter wurde brav aktualisiert. Habe weder auf Wallbox- noch auf evcc Seite ein Update gemacht. Ist also aktuell leider nicht mehr reproduzierbar für mich. 

Vermutlich ein Bug in der go-e App, dass die Abfrage des charged Wertes hin und wieder falsche Werte liefert? 

Sobald das Problem wieder auftritt, poste ich die Logs.

andig, 2023-11-18:
Gibts hier noch etwas zu tun? Sonst würde ich schließen.

phrozen77, 2023-11-29:
> Anfangs der Woche gab es einige Phasenwechsel, aber der charged counter wurde brav aktualisiert. Habe weder auf Wallbox- noch auf evcc Seite ein Update gemacht. Ist also aktuell leider nicht mehr [...]

Frage an der Stelle:

Funktioniert das jetzt zufällig seit dem du den Loglevel erhöht hast?

Genau so ist es nämlich bei mir - seit ich Loadpoint und Tesla auf "trace" gestellt habe, ist das Verhalten nicht wieder aufgetreten.

Dauert da nun plötzlich irgendwas länger und der Bug tritt dadurch nicht mehr auf?

closed 2023-11-29 by andig

thomas-wolf, 2023-11-29:
> Frage an der Stelle:
>
> Funktioniert das jetzt zufällig seit dem du den Loglevel erhöht hast?
>
> Genau so ist es nämlich bei mir - seit ich Loadpoint und Tesla auf "trace" gestellt habe, ist das [...]

Interessant, auf diese Verbindung bin ich noch nicht gekommen. Aber ja, könnte sein: Seit ich auf Trace umgestellt habe, ists nicht mehr aufgetreten. Hatte allerdings seitdem nur noch wenige Switches von 1p auf 3p und es ist ja auch zuvor m.E. nicht immer aufgetreten.

thomas-wolf, 2023-12-18:
Heute ists wieder soweit, just im Moment tritt das Problem wieder auf. Ladevorgang wurde gestern gestartet, pausierte über Nacht. Auto war die ganze Zeit angesteckt. Jetzt steht der Counter auf 13 kwh und bewegt sich nicht mehr:

[image]

Die go-e API liefert einen unplausiblen Wert: 

`Dec 18 13:34:45 phoscon evcc[27006]: [go-e  ] TRACE 2023/12/18 13:34:45 {"alw":true,"eto":1129278,"car":2,"**wh":3868.02**,"trx":null,"nrg":[227.23,228.16,230.02,2.48,13.528,0,0,3042.146,0,0,0,3042.146,98.96482,0,0,100],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}`

Gesamtes Log:

```
Dec 18 13:40:26 phoscon evcc[27006]: [site  ] DEBUG 2023/12/18 13:40:26 grid power: -172W
Dec 18 13:40:26 phoscon evcc[27006]: [site  ] DEBUG 2023/12/18 13:40:26 site power: -172W
Dec 18 13:40:26 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:40:26 charge currents: [12.4 0 0]A
Dec 18 13:40:26 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:40:26 detected active phases: 1p
Dec 18 13:40:26 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:40:26 charge total import: 1129.551kWh
Dec 18 13:40:26 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:40:26 charger status: C
Dec 18 13:40:26 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:40:26 vehicle soc: 34%
Dec 18 13:40:26 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:40:26 vehicle soc limit: 100%
Dec 18 13:40:26 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:40:26 vehicle range: 121km
Dec 18 13:40:26 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:40:26 pv charge current: 13.7A = 13A + 0.748A (-172W @ 1p)
Dec 18 13:40:26 phoscon evcc[27006]: [site  ] TRACE 2023/12/18 13:40:26 telemetry: charge: Δ31/31Wh @ 2778W
Dec 18 13:40:26 phoscon evcc[27006]: [site  ] TRACE 2023/12/18 13:40:26 POST https://api.evcc.io/v1/charge
Dec 18 13:40:26 phoscon evcc[27006]: [site  ] TRACE 2023/12/18 13:40:26 {"instanceId":"a9ae483fa129fd5f0791afbf5ac00f84fa2e77974a6bb3d9dc82ba83d86fda61","chargePower":2777.664,"greenPower":2777.664,"chargeEnergy":0.030904995514345876,"greenEnergy":0.030904995514345876}
Dec 18 13:40:26 phoscon evcc[27006]: --
Dec 18 13:40:26 phoscon evcc[27006]: {"status":"ok"}
Dec 18 13:40:35 phoscon evcc[27006]: [site  ] DEBUG 2023/12/18 13:40:35 ----
Dec 18 13:40:35 phoscon evcc[27006]: [go-e  ] TRACE 2023/12/18 13:40:35 GET http://192.168.178.114/api/status?filter=alw,car,eto,nrg,wh,trx,cards
Dec 18 13:40:35 phoscon evcc[27006]: [go-e  ] TRACE 2023/12/18 13:40:35 {"alw":true,"eto":1129559,"car":2,"wh":4150.675,"trx":null,"nrg":[226.61,227.54,229.4,2.48,12.426,0,0,2778.732,0,0,0,2778.732,98.68165,0,0,100],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}
Dec 18 13:40:35 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:40:35 charge power: 2779W
Dec 18 13:40:35 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:35 modbus: send f3 e0 00 00 00 06 01 03 9c 87 00 32
Dec 18 13:40:36 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:36 modbus: recv f3 e0 00 00 00 67 01 03 64 05 ad 01 e5 01 e5 01 e2 ff fe 0f 8c 0f 94 0f 92 08 f3 08 fd 09 06 ff ff 0d 0f 00 00 13 81 ff fe 0d 15 00 00 ab a0 ff fe d9 06 ff fe 01 60 73 60 00 00 11 62 ff fd 1d c8 ff ff 0d 41 00 00 80 00 10 84 80 00 80 00 ff fe 00 04 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00
Dec 18 13:40:36 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:36 DCPower: 3393
Dec 18 13:40:36 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:36 modbus: send f3 e1 00 00 00 06 01 03 f5 74 00 02
Dec 18 13:40:36 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:36 modbus: recv f3 e1 00 00 00 07 01 03 04 00 00 00 00
Dec 18 13:40:36 phoscon evcc[27006]: [site  ] DEBUG 2023/12/18 13:40:36 pv power: 3393W
Dec 18 13:40:36 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:36 modbus: send f3 e2 00 00 00 06 01 03 f5 74 00 02
Dec 18 13:40:36 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:36 modbus: recv f3 e2 00 00 00 07 01 03 04 00 00 00 00
Dec 18 13:40:36 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:36 modbus: send f3 e3 00 00 00 06 01 03 f5 84 00 02
Dec 18 13:40:36 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:36 modbus: recv f3 e3 00 00 00 07 01 03 04 00 00 42 c8
Dec 18 13:40:36 phoscon evcc[27006]: [site  ] DEBUG 2023/12/18 13:40:36 battery soc: 100%
Dec 18 13:40:36 phoscon evcc[27006]: [site  ] DEBUG 2023/12/18 13:40:36 battery power: 0W
Dec 18 13:40:36 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:36 modbus: send f3 e4 00 00 00 06 01 03 9c fe 00 69
Dec 18 13:40:36 phoscon evcc[27006]: [modbus] TRACE 2023/12/18 13:40:36 modbus: recv f3 e4 00 00 00 d5 01 03 d2 00 af 00 55 00 2e 00 2b ff ff 59 4e 59 4e 59 d0 59 f4 00 00 00 00 00 00 00 00 ff fe 13 7f ff fe 00 a4 f8 91 04 31 03 e3 00 00 00 a7 07 6f 04 31 03 e3 00 00 ff e4 ff f7 ff da 00 12 00 00 f2  [... cut ...]
Dec 18 13:46:16 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:46:16 charge total import: 1129.822kWh
Dec 18 13:46:16 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:46:16 charger status: C
Dec 18 13:46:16 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:46:16 vehicle soc: 34%
Dec 18 13:46:16 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:46:16 vehicle soc limit: 100%
Dec 18 13:46:16 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:46:16 vehicle range: 121km
Dec 18 13:46:16 phoscon evcc[27006]: [lp-1  ] DEBUG 2023/12/18 13:46:16 pv charge current: 13.2A = 13A + 0.235A (-54W @ 1p)

```

andig, 2023-12-18:
Laut API:

```
[go-e  ] TRACE 2023/12/18 13:46:15 {"alw":true,"eto":1129822,"car":2,"wh":4414.066,"trx":null,"nrg":[227.54,228.16,230.02,2.48,12.426,0,0,2792.963,0,0,0,2792.963,98.78161,0,0,100],"cards":...}
```

Es ist wie Du sagst:

> Die go-e API liefert einen unplausiblen Wert:

Soweit ich sehe bekommen wir 4,4kWh von der Go-E geschickt. Wenn das nicht stimmt dann schickt Go-E das Falsche. Da hilft nur ein Ticket bei Go-E. Gerne Ergebnis hier verlinken.

@thomas-wolf gerne `modbus` aus dem Trace raus nehmen- das machts nur unübersichtlicher.