evcc-io/evcc #10456: Critical Bug: Max current in yml ignored when vehicle is plugged in

davidgiga1993, 2023-10-23:
### Describe the bug

EVCC resets / overrides the maxCurrent defined in the yml file as soon as a vehicle has been detected.

This causes a lot of issues as it also overrides the maximum current configured on the charger (PMCC) causing the breaker to flip on overload.

### Steps to reproduce

1. Use specified evcc.yaml
2. Start evcc, the max current should be correct in the UI
3. Connect vehicle, the max current now resets to the maximum value

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: https
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: some-host.domain.com
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken: 

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: debug
levels:
  eebus: info
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

# meter definitions
[... 282 lines cut ...]
evcc  | [site  ] DEBUG 2023/10/23 17:57:07 site power: 230W
evcc  | [lp-1  ] DEBUG 2023/10/23 17:57:07 charge currents: [32.1 31.8 32.5]A
evcc  | [lp-1  ] DEBUG 2023/10/23 17:57:07 detected active phases: 3p
evcc  | [lp-1  ] DEBUG 2023/10/23 17:57:07 set max current: 32
evcc  | [lp-1  ] DEBUG 2023/10/23 17:57:07 charger status: C
evcc  | [lp-1  ] DEBUG 2023/10/23 17:57:07 forced charging at vehicle soc 30% (< 40% min soc)
evcc  | [site  ] DEBUG 2023/10/23 17:57:37 ----
evcc  | [lp-1  ] DEBUG 2023/10/23 17:57:37 charge power: 0W
evcc  | [lp-1  ] DEBUG 2023/10/23 17:57:37 forced charging at vehicle soc 30% (< 40% min soc)
evcc  | [site  ] DEBUG 2023/10/23 17:57:37 pv power: 28W
evcc  | [site  ] DEBUG 2023/10/23 17:57:37 grid power: 567W
evcc  | [site  ] DEBUG 2023/10/23 17:57:37 grid powers: [230 266 72]W
evcc  | [site  ] DEBUG 2023/10/23 17:57:37 grid currents: [1.72 1.65 0.794]A
evcc  | [site  ] DEBUG 2023/10/23 17:57:37 site power: 177W
evcc  | [lp-1  ] DEBUG 2023/10/23 17:57:37 charge currents: [0 0 0]A
evcc  | [lp-1  ] ERROR 2023/10/23 17:57:37 charger: timeout
```

### What type of operating system are you running?

Docker container

### Version

0.121.0 - 0.121.2

davidgiga1993, 2023-10-23:
Some further analysis:
The charger reports 16A as max current, which is correct given that the charger limit is set to 16A. The [eebus code](https://github.com/evcc-io/evcc/blob/70b28f2d08097c1f928208220ad9668365dc8405/charger/eebus.go#L183) updates the max current to 16A. I would open a discussion here that the max current of the charger should not be able to override the max current of the loadpoint if `LP.maxCurrent < charger.maxCurrent`. If you agree with this I can create an MR for that.

It's still unclear to me where the 32A came from, either the charger reported wrong values or something else in EVCC sets this to 32A.

andig, 2023-10-23:
> The logic I mentioned above should be revisited.

Works as designed. The EEBus charger overrides the currents. We will change this with https://github.com/evcc-io/evcc/pull/10335. See precedent rules there.

> The PMCC seems to report wrong max current values from time to time.

/cc @DerAndereAndi

closed 2023-10-23 by andig

davidgiga1993, 2023-10-23:
Thanks @andig! One question: The new logic has the min/maxcurrent priority vehicle, charger, lp.
From a pure abstract perspective I would see the LP value as more important than the charger because the loadpoint in my mind includes the electrical installation. Or is the idea that the values from the charger are always seen as "more correct"?

Would it make sense to always use the min value of all the 3?

andig, 2023-10-23:
> because the loadpoint in my mind includes the electrical installation

Fully acknowledged. We treat the charger as logically controlled by evcc. We never override any of the "physical" layer limits in the charger. That is always afaik a separate "physical limits" settings.

davidgiga1993, 2023-10-23:
But shouldn't then the loadpoint settings have higher priority than the one reported from the charger itself? Or is the charger seen as "source of truth" about the physical layer?

andig, 2023-10-23:
Not sure if you're referring to the current implementation or the PR? Point is: whatever we write to the charger, the chargers "this is the physical limit" setting will remain untouched and still be the upper boundary. That way safety is in hardware (or rather: in the charger's software), but not in evcc!

andig, 2023-10-23:
Evcc should set the „current“ but never (!) change the chargers min/max. Evcc will still send current within the min/max limits it is awar of. If these are outside cher limits, charger will clip/clamp them. 

/cc @DerAndereAndi

davidgiga1993, 2023-10-23:
Yes exactly I agree, but the PMCC doesn't use it's own limits if evcc sends a "charge current" value. Therefore in this case evcc is the source of truth for the global charge current limiter :-)

That's why my idea above was that the LP limit should have higher priority than the values from the charger itself (as the PMCC also apparently reports the max value sometimes instead of the actual set limit).
Or maybe it's just a bug in the PMCC communication. In that case the PR logic will be fine as long as the limits received from the PMCC are correct.

andig, 2023-10-24:
> I agree, but the PMCC doesn't use it's own limits if evcc sends a "charge current" value. 

@DerAndereAndi is this due to the way we're using EEbus or is that a defect in the PMCC implementation?

davidgiga1993, 2023-10-24:
> lets please take one step back to better understand the context: how much current does your house installation allow and how is your PMCC connected? Reason for asking: physical connection (CEE16 vs CEE32) _must_ match allowed physical limits.

In this instance the power connection was made using CEE16, the limit on the PMCC was set to 15A (16A in the test above for testing purposes).

DerAndereAndi, 2023-10-24:
The PMCC is connected using CEE or Schuko plugs and has its limits set by the used cables automatically. Using EEBUS it assumes that overload protection is done dynamically with the connected HEMS, which evcc does not support as of now. This is usually the main feature of compatible EEBUS HEMS systems.

The algorithm is added in the EEBUS charger because more and more EEBUS chargers (including PMCC, Spelsberg, and more) also support ISO15118-2 connections over AC to the EV. And the min limits are lower than using PWM. And also an ISO connection can always falls back to PWM. And therefore the limits can always change and should not be hardcoded in any configuration. So the logic is not because of EEBUS, but because of corresponding EVSEs that also support ISO15118-2 and cars that do so (e.g. Porsche Taycan, Cayenne Hybrid, MEB based EVs, Mercedes, some BMW, and more).

If the EVSE doesn't have charging authorization turned on, the EV will charge immediately and evcc may take a few seconds to lower the charging limits due to the way EEBUS and ISO work. The time depends on how long the evcc interface is set up and when within that intervall the EV is plugged in. So even by setting up a loadpoint limit to lower than what the cable can provide, you might have the same effects! A change in logic and evcc implementation will not guarantee that it works for you.

Trying to set the max limit on the loadpoint imho is a workaround for your specific setup but falls short because of the described needs. The suggestion discussed in the ticket and the new PR would not solve that either as it defined as vehicles takes precedence over charger over loadpoint. To make this work in an optimal case the loadpoint should define outer limits (with minCurrent being 0!), the charger might only reduce these limits, and the vehicle further reducing these.

A workaround for now may be to let the charger implementation only overwrite the minCurrent limit, as mostly the maxCurrent does not change with the communication protocol.

davidgiga1993, 2023-10-24:
Thanks for confirming that! It would also be enough imho to just use the smalles value of the 3 sources (vehicle, loadpoint, charger) and use that for the maxCurrent value. Should be easy enough to do

andig, 2023-10-24:
> In this instance the power connection was made using CEE16, the limit on the PMCC was set to 15A (16A in the test above for testing purposes).

Which means 16A ist the limit that physics allow. Evcc can never go above this limit since PMCC would enforce 16A. If it does not do that (whatever setting evcc supplies) then that is a bad PMCC bug.

Evcc will not respect the 15A, partly because it's not even aware of that.

davidgiga1993, 2023-10-24:
> Which means 16A ist the limit that physics allow. Evcc can never go above this limit since PMCC would enforce 16A. If it does not do that (whatever setting evcc supplies) then that is a bad PMCC [...]

No, the PMCC allowed 32A, causing the breaker to trip

andig, 2023-10-24:
That is a bad issue in the PMCC imho!

davidgiga1993, 2023-10-24:
> That is a bad issue in the PMCC imho, because it would allow software (HEMS) to override hardware (cable) safety limits. I would suggest to immediately raise this with Porsche. Evcc is not designed to ensure physical safety- that is a hardware task!

I will try, but in my experience with software issues on cars and their accessories things like these will take years to get acknowledges and even longer to get fixed.

But if evcc should not include a workaround/solution in the meantime I'll probably just run a fork with my patch until something gets fixed :-)

andig, 2023-10-24:
Absolutel! But this one is really gross. ❤️‍🩹

DerAndereAndi, 2023-10-24:
> Probably because it resets the reported limit when receiving a message from the HEMS.

This is absolutely not what is happening guys!

if the PMCC is connected to a CEE32 plug, it allows 32A per phase. And thus this is reported to the HEMS.

If the HEMS sets a lower charging limit, then it does it for this charging session only. Any higher values than reported by the plug connection will be rejected.

The suggested PR does NOT change the behvaiour or "fix" anything. I already mentioned what is going on and what the options are.

davidgiga1993, 2023-10-24:
> The suggested PR does NOT change the behvaiour or "fix" anything. I already mentioned what is going on and what the options are.

Do you mean my PR or the other one? Because mine definitely fixes it :-) I'm using it right now and the max current used is 15A (as specified in the LP yml definition). Even when the vehicle is plugged in and evcc takes a bit to respond the pmcc first uses it's own limit setting (which is 15A) and then evcc sends 15A as max value due to the min clamping (evcc basically ignores the 32A maxCurrent reported by the pmcc).
This can even be easily verified by unit testing the entire logic

DerAndereAndi, 2023-10-24:
Your PR. It breaks other behaviour.

davidgiga1993, 2023-10-24:
Not sure I understand why because you mentioned
> To make this work in an optimal case the loadpoint should define outer limits (with minCurrent being 0!), the charger might only reduce these limits, and the vehicle further reducing these.

Which is basically what's happening in the end since the minimum value for the "maxCurrent" wins, or am I overlooking something here?
Not trying to argue, just trying to understand

DerAndereAndi, 2023-10-24:
I provided a PR which would solve your issue and not break others.

In general: You would also get the same result without any code changes by setting the max current of the connected vehicle to your desired max.

andig, 2023-10-24:
> if the PMCC is connected to a CEE32 plug, it allows 32A per phase. And thus this is reported to the HEMS.

@DerAndereAndi @davidgiga1993 said its connected via CEE16

DerAndereAndi, 2023-10-24:
No, the only reference I find is:

> In this instance the power connection was made using CEE16, the limit on the PMCC was set to 15A (16A in the test above for testing purposes).

The limit in the PMCC is ONLY used, if no HEMS is connected. As it is stated in the screenshot mentioned in https://github.com/evcc-io/evcc/issues/10456#issuecomment-1775844020

I do not see a reference anywhere in here where it is stated that using a CEE16 cable a 32A limit is reported.

DerAndereAndi, 2023-10-24:
Maybe it would help to get a clarification of the setup and what actually happens :)

davidgiga1993, 2023-10-24:
To clarify: The pmcc cable is 32A, the wall plug CEE16. So from a pmcc side it detects a 32A cable. Thus the realization when the manual limit behavior didn't work as expected.
Nevertheless, even with a CEE32 I wouldn't want to use the full 32A for charging.
And yes the 16A pmcc cable would probably be better but it's not on stock and I still need to charge somehow until it gets shipped. But then again, I would like evcc to honor my settings, so #10478 should solve that.

DerAndereAndi, 2023-10-24:
@davidgiga1993 So this all is expected behaviour from the PMCC side.

andig, 2023-10-26:
> The pmcc cable is 32A, the wall plug CEE16. So from a pmcc side it detects a 32A cable. 

That is a physical security issue. Your setup cannot guarantee that the hardware can maintain the physical limits. The PMCC current limit is _only_ a fallback limit for when a HEMS is not configured.