evcc-io/evcc #10459: override language in config

samsmooth, 2023-10-23:
In Tesla's browser you cannot select a language - select boxes just do not work - In the Tesla browser you simply can't get the dropdown options. So evcc uses browser default settings which will be Dutch here in Germany...

Would it be possible to set the language in the yaml configuration as a override and display the select box in case it's not set in config?

naltatis, 2023-12-07:
Currently UI language preference is only held in the Browser (default auto-detect). Instead of adding a lang-overwrite option I think about persisting the setting to the server. This would also introduce cross device lang sync. Any arguments against this?

The only issue I can think about is having multiple people with different lang-preferences using the same evcc instance. In most cases I'd assume this is not a problem. Agreed?

RTTTC, 2024-01-03:
> The only issue I can think about is having multiple people with different lang-preferences using the same evcc instance. In most cases I'd assume this is not a problem. Agreed?

This would be slight inconvenience for me, since I do prefer EN, and my family can only use their native language.
But I will cope with that just fine, as it solves other problems for out community members :)
Thank you guys for all the improvements - it’s like Christmas every time!!🎁

PR #11867 (2024-01-26): Header-based auto-language detection
Fix #11789, fix #10459

Use `Accept-Language` header (Go) instead of `navigator.language(s)` (JS) for "Language: automatic" setting in UI.
closed 2024-01-26 by andig