evcc-io/evcc #10463: Charge Energy Overview ist falsch

henne49, 2023-10-24:
### Describe the bug

Nach einem Update auf 0.121.2 sind die Gesamt Lade Daten leider falsch. Siehe Screenshots.

Lasst mich wissen wie ich helfen kann, dass die Daten stimmen?

### Steps to reproduce

1.Using Version 0.121.1 richtige Daten
[image]

2.Using Version 0.121.2 falsche Daten
[image]

Nach Einspielen des Backups 0.121.1 sind die Daten wieder korrekt. 

### Configuration details

```yaml
welchen Teil der evcc.yaml braucht ihr? Würde ungern meine komplette yaml teilen, wegen privaten Details.
```

### Log details

```text
Ich nutze den home-assistant container, ich wüsste leider nicht, wie ich das tun kann.
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.121.2

naltatis, 2023-10-24:
Hi @henne49,

wir haben in https://github.com/evcc-io/evcc/pull/10341 die Berechnung umgestellt. Bislang wurden in dem Dialog Werte angezeigt, die auf Basis von Leistungsdaten und Zeit errechnet wurden. Das war ungenau und hat auch in der Vergangenheit immer wieder zu Inkonsistenzen geführt. In der neuen Version verwenden wir die Daten aus den Ladevorgängen. Diese sind genauer, da hier, sofern vorhanden, auf Zählerwerte zurückgegriffen wird und wir nicht die Leistung interpolieren. Zudem hat das den Vorteil, dass wir jetzt auch Zeiträume (30d, 1y, ...) betrachten können.

Hast du einen konkreten Wert, der dir unplausibel vorkommt? Bitte vergleiche das einmal mit deinen Daten aus den Ladevorgängen.

RTTTC, 2023-10-24:
>  In the new version, we use the data from the charging sessions.

And the charging sessions were implemented relatively not that long ago, so data that now is seen is since that time.
This is my interpretation of the change.

henne49, 2023-10-24:
exactly, this changes misses out old data, I guess that is where the different views/data come from. 

If that is the reason, OK I go with that.

Thank you for the great project!

andig, 2023-10-24:
@naltatis wäre das ein kurzes Update in https://github.com/evcc-io/evcc/discussions/2576 wert?

closed 2023-10-24 by andig

naltatis, 2023-10-24:
@andig I've added a description to the release note:
https://github.com/evcc-io/evcc/releases/tag/0.121.2

docolli, 2023-10-27:
Schaut ihr bitte auch auf die Daten, die per MQTT übertragen werden? Da gibt es bei mir jetzt auch deutliche Unterschiede.

andig, 2023-10-27:
...nur wenn Du ein neues Issue mit entsprechender Beschreibung draus machst ;)