evcc-io/evcc #10471: Auswertung Ladeenergie in 0.121.2 zeigt geringere Verbräuche

teilemann, 2023-10-24:
### Describe the bug

Seit dem Update auf Version 0.121.2 zeigt die Auswertung der Ladeenergie geringere Gesamtverbräuche für Sonne und Netz. Trotz Umstellung auf gesamten Zeitraum bleiben die Werte falsch, also zu gering.

[image]

In 0.121.1 sah das noch so aus:

[image]

Beim Downgrade werden die korrekten Werte wiederhergestellt.

### Steps to reproduce

In der Hauptseite des Programms unten rechts auf ..% Sonnenenergie klicken.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 25s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3NzkwMjA0MDAsImlhdCI6MTY4NDQxMjQwMCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6InRlaWxlbWFubiJ9.iVHM71rK5oOjYiE0EpQQyMhi4FOhvc8wqk4jxH3fElw # your token from sponsor.evcc.io

vehicles:
  - name: aedrian
    type: template
    template: tesla
    title: Ædrian # Wird in der Benutzeroberfläche angezeigt # Optional
    accessToken: XXXXXXXXXXXXXXXXXX
    # Siehe https://docs.evcc.io/docs/devices/vehicles#tesla
    refreshToken: XXXXXXXXXXXXXXX
    # Siehe https://docs.evcc.io/docs/devices/vehicles#tesla
    vin: LRW3E7XXXXXXXXXX # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 60 # Akku-Kapazität in kWh # Optional 

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
# grid custom
- name: grid
  type: movingaverage
  decay: 1
  meter:
    type: custom
    power:
      source: mqtt
      topic: solarview/D0/DPAC
      scale: -1
    energy:
      source: calc
      add:
      - source: mqtt
        topic: solarview/D0/BKDY
      - source: mqtt
        topic: solarview/D0/GKDY
        scale: -1
        timeout: 30s

# grid generated
#- type: template
#  template: mqtt-meter 
#  usage: grid  
#  power: solarview/D0/DPAC  
#  scalepower: -1  
#  energy: solarview/D0/GKDY  
#  scaleenergy: 1  
#  timeout: 30s  
#  name: grid

# pv custom
- name: pv
  type: custom
  power:
    source: mqtt
    topic: solarview/WR0/PAC
  energy:
    source: mqtt
    topic: solarview/WR0/KDY
    timeout: 30s
  currents: # optional currents (A)
    - source: mqtt # L1 plugin type
      topic: solarview/WR1/IL1
    - source: mqtt # L2 plugin type
      topic: solarview/WR2/IL1
    - source: mqtt # L3 plugin type
      topic: solarview/WR0/IL1 

# pv generated 
#- type: template
#  template: mqtt-meter 
#  usage: pv  
#  power: solarview/WR0/PAC  
#  scalepower: 1  
#  energy: solarview/WR0/KDY  
#  scaleenergy: 1  
#  currentp1: solarview/WR1/IL1  
#  currentp2: solarview/WR2/IL1  
#  currentp3: solarview/WR0/IL1  
#  timeout: 30s  
#  name: pv

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/evcc-io/config#chargers
chargers:
- type: template
  template: go-e-v3 
  host: go-eCharger.fritz.box  
  name: wallbox

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: go-e Charger
  charger: wallbox
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  mode: pv
  vehicle: aedrian
  enable:
    threshold: 0 # 0 bedeutet Export >= min. LAdeleistung
    delay: 5m
  disable:
    threshold: 250 # maximum import power (W)
    delay: 15m

# site describes the EVU connection, PV and home battery
site:
  title: Mein Zuhause
  meters:
    grid: grid
    pvs:
    - pv

# tariffs are the fixed or variable tariffs
# cheap can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.3388 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.195517 # [currency]/kWh    

# mqtt message broker
mqtt:
  broker: raspi.fritz.box:1883
  topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:
  
# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
```

### Log details

```text
[main  ] INFO 2023/10/24 17:17:06 evcc 0.121.1
[main  ] INFO 2023/10/24 17:17:06 using config file: /home/pi/evcc.yaml
[main  ] INFO 2023/10/24 17:17:06 starting ui and api at :7070
[db    ] INFO 2023/10/24 17:17:06 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/10/24 17:17:06 connecting evcc-834253258 at tcp://raspi.fritz.box:1883
[mqtt  ] DEBUG 2023/10/24 17:17:06 tcp://raspi.fritz.box:1883 connected
[lp-1  ] WARN 2023/10/24 17:17:07 locking phase config to 1p for switchable charger
[site  ] WARN 2023/10/24 17:17:07 deprecated: use 'pv' instead of 'pvs'
[site  ] INFO 2023/10/24 17:17:07 site config:
[site  ] INFO 2023/10/24 17:17:07   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/10/24 17:17:07     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/24 17:17:07     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/10/24 17:17:07   vehicles:
[site  ] INFO 2023/10/24 17:17:07     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/10/24 17:17:07 loadpoint 1:
[lp-1  ] INFO 2023/10/24 17:17:07   mode:        pv
[lp-1  ] INFO 2023/10/24 17:17:07   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/10/24 17:17:07   meters:      charge ✓
[lp-1  ] INFO 2023/10/24 17:17:07     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/10/24 17:17:07 phase timer inactive
[lp-1  ] DEBUG 2023/10/24 17:17:07 pv timer inactive
[lp-1  ] DEBUG 2023/10/24 17:17:07 guard timer inactive
[lp-1  ] INFO 2023/10/24 17:17:07 vehicle updated: unknown -> Ædrian
[site  ] WARN 2023/10/24 17:17:07 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/10/24 17:17:07 ----
[lp-1  ] DEBUG 2023/10/24 17:17:07 charge power: 0W
[mqtt  ] DEBUG 2023/10/24 17:17:07 solarview/WR0/PAC wait for initial value
[main  ] INFO 2023/10/24 17:17:08 new version available: 0.121.2
[mqtt  ] DEBUG 2023/10/24 17:17:14 solarview/WR0/KDY wait for initial value
[site  ] DEBUG 2023/10/24 17:17:15 pv power: 86W
[mqtt  ] DEBUG 2023/10/24 17:17:15 solarview/D0/DPAC wait for initial value
[site  ] DEBUG 2023/10/24 17:17:16 grid power: 198W
[site  ] DEBUG 2023/10/24 17:17:16 site power: 198W
[lp-1  ] ERROR 2023/10/24 17:17:16 vehicle odometer: asleep
[lp-1  ] DEBUG 2023/10/24 17:17:16 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/24 17:17:16 charge total import: 1442.928kWh
[lp-1  ] DEBUG 2023/10/24 17:17:16 charger status: A
[lp-1  ] INFO 2023/10/24 17:17:16 car disconnected
[lp-1  ] DEBUG 2023/10/24 17:17:16 set charge mode: pv
[lp-1  ] DEBUG 2023/10/24 17:17:16 set min current: 6
[lp-1  ] DEBUG 2023/10/24 17:17:16 set max current: 16
[lp-1  ] DEBUG 2023/10/24 17:17:16 set priority: 0
[site  ] DEBUG 2023/10/24 17:17:32 ----
[lp-1  ] DEBUG 2023/10/24 17:17:32 charge power: 0W
[site  ] DEBUG 2023/10/24 17:17:32 pv power: 89W
[site  ] DEBUG 2023/10/24 17:17:32 grid power: 205W
[site  ] DEBUG 2023/10/24 17:17:32 site power: 205W
[lp-1  ] DEBUG 2023/10/24 17:17:32 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/24 17:17:32 charge total import: 1442.928kWh
[lp-1  ] DEBUG 2023/10/24 17:17:32 charger status: A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.1

RTTTC, 2023-10-24:
https://github.com/evcc-io/evcc/issues/10463

naltatis, 2023-10-24:
Ja, wir haben die Berechnungsmethode umgestellt. Ich habe einen Hinweis darauf in den Releasenotes ergänzt:
https://github.com/evcc-io/evcc/releases/tag/0.121.2

closed 2023-10-24 by naltatis