evcc-io/evcc #10480: Victron Wallbox: charger: invalid status: G

jubudu, 2023-10-24:
### Describe the bug

Victron Wallbox reports 24 (0…23) different states.
If vehicle is connected and waits for charging in manual mode, it reports status 6 („6 - waiting for start“)
„6“ is the 7th status (0…6….23)
Obviously this is mapped to the 7th letter of the alphabet - G, which is an invalid status for evcc (only A…F).

Moreover the documenting of the victron wallbox is misleading:
evcc refers to the EVCS-Modbus-registers of the **GX-device** (i.e. the Cerbo) and not of the wallbox. 
Those GX-registers are 3800 … 3827, while the EVCS-wallbox registers start at 5000…5108.
So the IP must be the one of the cerbo GX-device and the ID of the wallbox used by the GX-device is 40 and not 100.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:

  - name: Victron-Wallbox
    type: template
    template: victron
    # Modbus TCP
    modbus: tcpip
    id: 40 # Modbus ID of the EVCS-Wallbox
    host: 192.168.xxx.xxx # IP of the GX-device
    port: 502 # Port
```

### Log details

```text
[lp-2  ] ERROR 2023/10/25 00:37:55 charger: invalid status: G
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.121.2

andig, 2023-10-25:
Ohja, da ist die Umsetzung von https://github.com/evcc-io/evcc/files/11727609/CCGX-Modbus-TCP-register-list.xlsx unvollständig.

jubudu, 2023-10-26:
Status G ist die Nr. 6 in der Victron-Zählweise, da Victron bei „0 - disconnected“ beginnt. 
0 -> A
1 -> B
2 –> C (von mir noch nicht überprüft)
3 -> D ( dto.)
4 ->  nur im Auto-Modus der Victron-Box möglich
5 -> B
6 -> B
7 -> nur im Auto-Modus der Victron-Box möglich

Evcc könnte den manuellen modus der Box erzwingen, so dass die Stati 4 und 7 nicht auftreten. Dies ginge über Schreiben des Modbus-Registers 3815 mit  dem Wert 0.

andig, 2023-10-26:
> Evcc könnte den manuellen modus der Box erzwingen, so dass die Stati 4 und 7 nicht auftreten. Dies ginge über Schreiben des Modbus-Registers 3815 mit dem Wert 0.

Das würde ich gerne dem Anwender überlassen- falls da noch ein Hinweis fehlt gerne PR für das Victron Template. Wenn wir die Statuus sehen weiter Fehler. Mapping passt, Lösung morgen im Nightly.

closed 2023-10-26 by andig