evcc-io/evcc #10482: Error with consumption statistics

der-peer, 2023-10-25:
### Describe the bug

On v0.121.1 I have other historic values for loading with grid/pv than with v0.121.2.
I had around 80% PV (~3600kW PV, 840kW grid), now it says 92% (2500/90). I did switch to "whole time" beforehand.

### Steps to reproduce

1. switch to v0.121.2
2. switch back to v0.121.1

### Configuration details

```yaml
no setting regarding this 🤷‍♂️
```

### Log details

```text
no log entries about this topic
```

### What type of operating system are you running?

Docker container

### Version

0.121.1 / 0.121.2

VolkerK62, 2023-10-25:
there was a change.
see release notes https://github.com/evcc-io/evcc/releases/tag/0.121.2

der-peer, 2023-10-25:
Thanks. Stupid me, just updated (it's so convenient using docker), without checking release notes before.

Nonetheless. Values are wrong. Statistics startet Nov. 2, 2022. The actual shown values are way too low. Especially for grid consumption.

karstenrothe, 2023-10-25:
Same here, Update from 0.120.0 to 0.121.2 and my "1.699 kwh Netz" since 7.11.2022 are gone.
My Old Values are 
55,2% Sonne
2.094 kwh Sonne 
1.699 kwh Netz
 after the Update
"Zeitraum 365 Tage"
92,2% Sonne
1.387kwh Sonne 
118kwh Netz

wedojan, 2023-10-25:
Ich habe das selbe Proble.
Der Netzbezug ist viel zu gering. (Er war zuletzt bei etwa 22% des Gesamtverbrauchs)

[image]

RTTTC, 2023-10-26:
It calculates using sessions data _since the addition of solar % to sessions_.

@andig - I think we should not close this issue for a few months, so people see it first. 
Otherwise someone will keep opening new issues with this question...

karstenrothe, 2023-10-26:
Danke für die Aufklärung.
Gut das die Berechnung jetzt genauer ist.
Trotzdem frustrierend, die alte Statistik mal eben um 7 Monate gekappt :'-(
Meine Statistik haute schon echt gut hin und ich fieberte dem 7.11.23 entgegen, damit meine 12 Monats Statistik fertig ist.
Habe meine alten Werte aufgeschrieben und rechne den Rest per Hand - und in 7 Monaten haben wir dann ja wieder das Jahr mit neuer Berechnung voll ;-)
Wirklich geniales Projekt was hervorragend läuft, dem Anwender hilft den Eigenverbrauch zu optimieren und damit Geld spart und die (Um)Welt damit auch noch ein wenig besser macht!
Vielen lieben Dank Euch, macht weiter so!!!!

andig, 2023-10-26:
> Trotzdem frustrierend, die alte Statistik mal eben um 7 Monate gekappt :'-(

@naltatis wie wäre es denn, wenn wir einmalig (woran würden wir das festmachen) einen einzelnen Ladevorgang "Restmenge" anlegen würden der die bisher nicht in Sessions gehaltenen Daten zusammen fasst?

@karstenrothe den kannst Du Dir natürlich auch händisch in die Tabelle der Ladevorgänge eintragen.

RTTTC, 2023-10-26:
No need for me, since this is only "cosmetics" anyway.
Next summer there will be full year data available again :)
So lets just keep it simple.

naltatis, 2023-10-26:
> "Restmenge" anlegen würden der die bisher nicht in Sessions gehaltenen Daten zusammen fasst?

Da steckt ne Menge Komplexität und ggf. Fehlerpotential drin. Was wir noch machen könnten, damit das transparenter wird ist den ersten Ladevorgang, also den "ab Datum" Wert mit in der UI anzuzeigen. Gerade bei "Gesamt" ist das sinnvoll.

karstenrothe, 2023-10-26:
Das würde sofort simpel und klar die Statistik erklären - auch für 365 die dann ja erst ab ?Juni23? läuft.
So hatte ich es aus den Release Notes auch nicht gelesen, aber jetzt dann auch verstanden ;-)
Mehr Aufwand würde ich auch nicht investieren, like RTTTC: "more important issues".

VolkerK62, 2023-10-26:
Wenn ich mir das "Gesamt CSV" herunterlade, habe ich bei Summe Energie = 1328 kWh
Im UI bekomme ich jedoch für "gesamt" nur 677 kWh Sonne + 73 kWh Netz = 750 kWh angezeigt.
Der Unterschied ist dann doch ziemlich groß.
Laut CSV habe ich die 750 kWh seit 4.5.2023 bis heute geladen.

andig, 2023-10-26:
> Wer lesen kann, ist eindeutig im Vorteil ;-)

Das nehme ich als "erledigt" :)

closed 2023-10-26 by andig