evcc-io/evcc #10484: PV mode stops controlling the Wallbox current

panzerdev, 2023-10-25:
### Describe the bug

evcc stops setting the current on the wallbox in PV mode. It resumes doing that after switching to "now" mode and back to "pv"

### Steps to reproduce

I can't reproduce it but it happens more often in the last days / weeks

### Configuration details

```yaml
provided if needed
```

### Log details

```text
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:27 ----
evcc_1  | [lp-1  ] DEBUG 2023/10/25 14:37:27 charge power: 0W
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:27 charge power: 3677W
evcc_1  | [lp-3  ] DEBUG 2023/10/25 14:37:27 charge power: 0W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 pv power: 3859W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 battery soc: 97%
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 battery power: 174W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 grid power: 5W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 site power: 279W
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:28 charge currents: [15.8 0 0]A
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:28 detected active phases: 1p
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:28 charge total import: 8274.717kWh
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:28 charger status: C
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:28 soc estimated: 62.16% (vehicle: 62.00%)
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:28 vehicle soc: 62%
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:28 vehicle soc limit: 80%
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:28 vehicle range: 203km
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:28 pv charge current: 4.79A = 6A + -1.21A (279W @ 1p)
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 ----
evcc_1  | [lp-1  ] DEBUG 2023/10/25 14:37:28 charge power: 0W
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:28 charge power: 3690W
evcc_1  | [lp-3  ] DEBUG 2023/10/25 14:37:28 charge power: 0W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 pv power: 3860W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 battery soc: 97%
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 battery power: 177W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 grid power: -0W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:28 site power: 277W
evcc_1  | [lp-3  ] DEBUG 2023/10/25 14:37:29 charge currents: [0 0 0]A
evcc_1  | [lp-3  ] DEBUG 2023/10/25 14:37:29 charger status: B
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:38 ----
evcc_1  | [lp-1  ] DEBUG 2023/10/25 14:37:38 charge power: 0W
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:38 charge power: 3695W
evcc_1  | [lp-3  ] DEBUG 2023/10/25 14:37:38 charge power: 0W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:39 pv power: 3803W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:39 battery soc: 97%
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:39 battery power: 194W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:39 grid power: -4W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:39 site power: 290W
evcc_1  | [lp-1  ] DEBUG 2023/10/25 14:37:39 charge currents: [0 0 0]A
evcc_1  | [lp-1  ] DEBUG 2023/10/25 14:37:39 charge total import: 3215.963kWh
evcc_1  | [lp-1  ] DEBUG 2023/10/25 14:37:39 charger status: A
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:48 ----
evcc_1  | [lp-1  ] DEBUG 2023/10/25 14:37:48 charge power: 0W
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:37:48 charge power: 3681W
evcc_1  | [lp-3  ] DEBUG 2023/10/25 14:37:48 charge power: 0W
evcc_1  | [site  ] DEBUG 2023/10/25 14:37:48 pv power: 3726W
evcc_1  | [site  ] DEBUG 2023/10/25 14:38:04 battery soc: 97%
evcc_1  | [site  ] DEBUG 2023/10/25 14:38:04 battery power: 229W
evcc_1  | [site  ] DEBUG 2023/10/25 14:38:04 grid power: -8W
evcc_1  | [site  ] DEBUG 2023/10/25 14:38:04 site power: 321W
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:38:04 charge currents: [15.8 0 0]A
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:38:04 detected active phases: 1p
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:38:04 charge total import: 8274.754kWh
[... 184 lines cut ...]
evcc_1  | [lp-3  ] DEBUG 2023/10/25 14:39:18 charge power: 0W
evcc_1  | [site  ] DEBUG 2023/10/25 14:39:19 pv power: 3680W
evcc_1  | [site  ] DEBUG 2023/10/25 14:39:19 battery soc: 97%
evcc_1  | [site  ] DEBUG 2023/10/25 14:39:19 battery power: -177W
evcc_1  | [site  ] DEBUG 2023/10/25 14:39:19 grid power: 120W
evcc_1  | [site  ] DEBUG 2023/10/25 14:39:19 site power: 43W
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 charge currents: [14.3 0 0]A
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 detected active phases: 1p
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 charge total import: 8274.829kWh
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 charger status: C
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 soc estimated: 62.34% (vehicle: 62.00%)
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 vehicle soc: 62%
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 vehicle soc limit: 80%
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 vehicle range: 203km
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 pv charge current: 14.2A = 14.4A + -0.187A (43W @ 1p)
evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 max charge current: 14.2A
```

### What type of operating system are you running?

Linux

### Version

0.121.2

VolkerK62, 2023-10-25:
`DEBUG 2023/10/25 14:37:28 pv charge current: 4.79A = 6A + -1.21A (279W @ 1p)`
evcc ist der Meinung, mit 6A zu laden. Was ist vorher im Log passiert?

panzerdev, 2023-10-25:
Ich muss schauen ob ich die Logs noch habe, aber ich frage mich mit wie er darauf kommt? Immerhin sieht man ja die anderen Werte.

VolkerK62, 2023-10-25:
afaik findet kein Abgleich zwischen der Stromvorgabe und dem tatsächlichen Ladestrom statt. Wenn kein Fehler gemeldet wird, wird davon ausgegangen, das die Wallbox das umsetzt, was vorgegeben wurden.

andig, 2023-10-26:
Bitte vollständiges Log/Config. Es ist nichtmal klar, um welche WB es geht :(

closed 2023-10-26 by andig

panzerdev, 2023-10-26:
Da mir dieses Verhalten bei der Heidelberg EC passiert ist und beim Heitzstab mit custom Charger passiert ist habe ich die config weggelassen. Hab sie hinzugefügt.

Der Log sollte doch genug sein? Es sind mehrere Regelschleifen drinnen oder gibt es einen bestimmten Zeitpunkt ab dem der anfangen soll? Application start ist einfach zu viel wenn evcc schon paar Tage läuft.

andig, 2023-10-26:
> evcc stops setting the current on the wallbox in PV mode

Woran machst Du das fest? Kann ich im Logfile nicht erkennen.

panzerdev, 2023-10-26:
> evcc_1  | [lp-2  ] DEBUG 2023/10/25 14:39:19 max charge current: 14.2A

Das fehlt oben in den Logs.

Die Wallboxen waren erreichbar aber haben keine Befehle bekommen.

andig, 2023-10-26:
> pv charge current: 4.6A = 6A + -1.4A (321W @ 1p)

Warum auch unter 6A? Da bin ich ja schon auf Minimum. Wie gesagt- Logfile ist King...

panzerdev, 2023-10-26:
Ich werde beim nächsten Mal das Logfile in gänze hochladen. Vielleicht kann ich das irgendwie doch reproduzieren, ich schau mal