evcc-io/evcc #10491: [heidel] ERROR heartbeat: modbus: response slave id '2' does not match request '1'

Mikka20, 2023-10-25:
### Describe the bug

Hallo zusammen,

an meinem Modubus-Adapter Protoss PE11 hängt die WallBox Heidelberg Energy Control. Damit funktioniert evcc ohne Probleme. Erweitere ich nun die Modbus-Linie um ein weiteres Gerät am Protoss PE11, erscheint die folgende Meldung im Log und in der evcc Gui:
[heidel] ERROR 2023/10/25 20:54:25 heartbeat: modbus: response slave id '2' does not match request '1'

Die WallBox hat die Modbus ID 1 und das weitere Gerät der Eastron DM72D-M die ID 2.

Die Suche hier bei den Issues bzw. der Doku von evcc hat mich leider nicht weiter gebracht.
Kann jemand mit der Meldung etwas anfangen und wie kann ich diese beseitigen?

VG,
Mikka

### Steps to reproduce

Modbus-Linie um ein weiteres Gerät (Eastron DM72D-M) erweitern.

Eastron DM72D-M <-- Busleitung J-Y(ST)-Y 2x2 x 0,8 --> Protoss PE11 <-- Busleitung J-Y(ST)-Y 2x2 x 0,8 --> WallBox Heidelberg
                                                                                                            

### Configuration details

```yaml
network:
  schema: http
  host: evcc.localdomain
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

sponsortoken: abc

log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
   - name: grid
     type: custom
     power: # Leistung (W)
       source: mqtt
       topic: Energie/Haus/Power-consumption-W-L1-L2-L3
       #scale: -1
     energy: # Energie (kWh) - Umrechnung von Wh -> kWh
       source: mqtt
       topic: Energie/Haus/Total-consumption-kWh
       #scale: 0.001
     currents:
       - source: mqtt
         topic: Energie/Haus/Power-consumption-W-L1
       - source: mqtt
         topic: Energie/Haus/Power-consumption-W-L2
       - source: mqtt
         topic: Energie/Haus/Power-consumption-W-L3
   - name: WallBoxMeter
     type: custom
     power: # Leistung (W)
       source: mqtt
       topic: Energie/WallBox/Power-consumption-W-sys
       #scale: -1
     energy: # Energie (kWh) - Umrechnung von Wh -> kWh
       source: mqtt
       topic: Energie/WallBox/Energy-returned-kWh-total
       #scale: 0.001
     currents:
       - source: mqtt
         topic: Energie/WallBox/Power-consumption-W-L1
       - source: mqtt
         topic: Energie/WallBox/Power-consumption-W-L2
       - source: mqtt
         topic: Energie/WallBox/Power-consumption-W-L3

chargers:
  - name: HeidelbergEnergieControl
    type: template
    template: heidelberg

    # RS485 via TCP/IP (Modbus RTU)
    modbus: rs485tcpip
    id: 1
    host: 192.168.178.143
    port: 8899 # Port

vehicles:
  - name: CooperSE
    type: template
    template: mini
    title: Cooper SE # Wird in der Benutzeroberfläche angezeigt (Optional)
    user: user # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: password # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: vin # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (Optional)
    capacity: 28.9 # Akkukapazität in kWh (Optional)

site:
  title: evcc # display name for UI
  meters:
    grid: grid # grid meter

loadpoints:
  - title: Garage # display name for UI
    charger: HeidelbergEnergieControl # charger
    meter: WallBoxMeter # charge meter
    mode: "off" # set default charge mode, use "off" to disable by default if charger is publicly available
    vehicle: CooperSE # set default vehicle (disables vehicle detection)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    phases: 3 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)

mqtt:
  broker: 192.168.178.20:1883
  topic: evcc # root topic for publishing, set empty to disable
  clientid: evcc
  user: evcc
  password: evcc
```

### Log details

```text
[heidel] ERROR 2023/10/25 21:25:10 heartbeat: modbus: response slave id '2' does not match request '1'
[heidel] ERROR 2023/10/25 21:25:25 heartbeat: modbus: response slave id '2' does not match request '1'
```

### What type of operating system are you running?

Docker container

### Version

0.121.2

Mikka20, 2023-10-25:
[lp-1  ] ERROR 2023/10/25 21:30:34 charger: modbus: response slave id '2' does not match request '1'
[heidel] ERROR 2023/10/25 21:30:48 heartbeat: modbus: response slave id '2' does not match request '1'
[heidel] ERROR 2023/10/25 21:31:03 heartbeat: modbus: response slave id '2' does not match request '1'

panzerdev, 2023-10-25:
Laut Doku für SDM72D-M

[image]

Und die Heidelberg EC hat

[image]

Ich würde sagen du kannst die nicht an gleichen Kabel betreiben

Mikka20, 2023-10-25:
Hi,

die Werte habe ich am SDM72D-M angepasst. In FHEM mit einem Modbus-Modul kann ich die Werte des Eanertex ohne Probleme auslesen.

Mikka20, 2023-10-25:
Die Doku Online zu dem Zähler und der kleine Beipackzettel scheinen auseinander zu driften :-)
Am Zähler selbst kann man 19200 einstellen. Im FHEM device wird auch die korrekte Baudrate angezeigt:
Modbus_Speed__baud | 19200

Modbus_Node_adr | 2 
Modbus_Parity_Stop | 1stop.bit_even.parity

Mikka20, 2023-10-25:
Die Doku passt besser:
https://xn--stromzhler-v5a.eu/media/pdf/93/17/d7/SDM72DM-V2.pdf

For Modbus RTU, the following RS485 communication parameters can be configured from the Set-up menu:
Baud rate 1200,2400, 4800, 9600, 19200 bps
Parity none (default)/odd/even
Stop bits 1 or 2
RS485 network address nnn – 3-digit number, 001 to 247
Modbus™ Word order Hi/Lo byte order is set automatically to normal or reverse. It cannot be configured from
the set-up menu.

panzerdev, 2023-10-25:
Du hast ja rs485tcpip eingestellt. Hast schon mal versucht dein Protos als Modbus TCP to RTU to betreiben und dann auf midbus: tcpip umzustellen? Habe selber 3 Heidelberg Wallboxen so an evcc und node red dran. Habe erst deinen Gerät eins von Waveshare. Der ordnet die Antworten auf den RTU dem TCP request zu

Mikka20, 2023-10-25:
Habe ich noch nicht, kann ich und werde ich probieren morgen. Danke
Das Auto muss erstmal laden :-)