evcc-io/evcc #10494: Fehler bei evcc soc Abfrage cupra Born openwb

mairj2345, 2023-10-26:
### Describe the bug

Seit gestern tritt folgender Fehler auf bei der Soc Abfrage per evcc in der openwb:

2023-10-26 08:37:40: PID: 7276: Lp1: Error from EVCC: 2023/10/26 08:37:40 rpc error: code = DeadlineExceeded desc = context deadline exceeded (LV0)

Auto ist ein cupra Born. Abfrage per cupra App geht. 

### Steps to reproduce

1. Abfrage manuell oder automatisch Soc openwb
2.
3.
...

### Configuration details

```yaml
An der Konfiguration würde nichts verändert.
```

### Log details

```text
2023-10-26 08:37:40: PID: 7276: Lp1: Error from EVCC: 2023/10/26 08:37:40 rpc error: code = DeadlineExceeded desc = context deadline exceeded (LV0)
```

### What type of operating system are you running?

Linux

andig, 2023-10-26:
Da ist nix zu machen- Cupra Server antwortet nicht. Wenn das Problem länge bestehen bleibt müssten wir den Default Timeout erhöhen.

closed 2023-10-26 by andig

mairj2345, 2023-10-26:
Danke für die Antwort. Problem besteht immer noch. Auch die App braucht lange, bis sie was anzeigt. Vielleicht wäre eine Verlängerung des timeouts keine verkehrte Idee.